espressif/esp-idf #5545: netif string to new IP structs

tonyp7, 2020-07-06:
**Description**

netif proposes two IP types: esp_ip4_addr_t and esp_ip6_addr_t.

However, the new library does not support native functions to have string to IP conversions.

**Proposed solution**

I would like to propose a PR to add these functionalities and would like to get feedback on if the PR would get accepted. Proposed solution would be:

```c
esp_err_t esp_netif_str_to_ip4(const char *src, esp_ip4_addr_t *dst);
esp_err_t esp_netif_str_to_ip6(const char *src, esp_ip6_addr_t *dst);
```

**Alternatives**

The alternative is to use inet_pton. This is not intended to use with netif new ip structs and feels very "hackish"

**Context**

Before I work on this PR, I'd like validation that this would be accepted.

david-cermak, 2020-09-09:
Hi @tonyp7 

Thanks for the proposal, this looks good to me.  Indeed, very useful to have these function at hand. Feel free to post a PR.

david-cermak, 2021-01-25:
Please note that we've already decided to implement these functions internally, just FYI if anyone wanted to post a PR.

@tonyp7 Thanks again for the suggested API.

closed 2021-02-10 by projectgus