espressif/esp-idf #5547: ESP BLE Mesh fast_prov_server例程无法实现功能

Beibixian, 2020-07-06:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v3.4-1
- Build System:         [idf.py]
- Compiler version ：xtensa-esp32-elf -gcc (crosstool-NG esp- -2019r2) 8. 2.0
- Operating System:     [Windows]
- (Windows only) environment type: [Command Prompt|PowerShell].
- Using an IDE?: [vscode]
- Power Supply:  [USB]

## Problem Description

//Detailed problem description goes here.

例程来源：https://github.com/espressif/esp-idf/tree/release/v4.1/examples/bluetooth/esp_ble_mesh/ble_mesh_fast_provision/fast_prov_server
在测试官网中ESP BLE Mesh fast_prov_server的例程中出现问题：
向3块相同的开发板烧录fast_prov_server的例程
使用安卓小米手机(APP：EspBLEMesh)，在配网中选择快速配网后，只能实现所选那一块开发板的配网（蓝灯熄灭），其它两块开发板仍处于未配网状态（蓝灯长亮同时可以被app检测到未配网）
无法实现官网提供实例视频：http://download.espressif.com/BLE_MESH/BLE_Mesh_Demo/V0.4_Demo_Fast_Provision/ESP32_BLE_Mesh_Fast_Provision.mp4 中的全体开发板自动快速配网功能
监视器log如下：
PS D:\esp32-4-1\workspace\fast_prov_server> D:\esp32-4-1\esp32-4-1-tools\python_env\idf4.0_py3.8_env\Scripts\python.exe D:\esp32-4-1\esp32-4-1-idf\tools\idf.py -p COM3 monitor
Checking Python dependencies...
Python requirements from D:\esp32-4-1\esp32-4-1-idf\requirements.txt are satisfied.
Executing action: monitor
Running idf_monitor in directory d:\esp32-4-1\workspace\fast_prov_server
Executing "D:\esp32-4-1\esp32-4-1-tools\python_env\idf4.0_py3.8_env\Scripts\python.exe D:\esp32-4-1\esp32-4-1-idf\tools/idf_monitor.py -p COM3 -b 115200 d:\esp32-4-1\workspace\fast_prov_server\build\fast_prov_server.elf -m 'D:\esp32-4-1\esp32-4-1-tools\python_env\idf4.0_py3.8_env\Scripts\python.exe' 'D:\esp32-4-1\esp32-4-1-idf\tools\idf.py' '-p' 
'COM3'"...
--- idf_monitor on COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6888
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (39) boot: compile time 15:08:01
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d38c (185228) map
I (176) esp_image: segment 1: paddr=0x0003d3b4 vaddr=0x3ffbdb60 size=0x02c5c ( 11356) load
I (181) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xa9b10 (695056) map
0x400d0018: _stext at ??:?

I (432) esp_image: segment 3: paddr=0x000e9b30 vaddr=0x3ffc07bc size=0x015f0 (  5616) load
I (434) esp_image: segment 4: paddr=0x000eb128 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/esp32-4-1/esp32-4-1-idf/components/freertos/xtensa_vectors.S:1778

I (439) esp_image: segment 5: paddr=0x000eb530 vaddr=0x40080400 size=0x12af8 ( 76536) load
I (491) boot: Loaded app from partition at offset 0x10000
I (492) boot: Disabling RNG early entropy source...
I (492) cpu_start: Pro cpu up.
I (496) cpu_start: Application information:
I (500) cpu_start: Project name:     fast_prov_server
I (506) cpu_start: App version:      1
I (511) cpu_start: Compile time:     Jul  6 2020 15:29:10
I (517) cpu_start: ELF file SHA256:  d77f59455da030dd...
I (523) cpu_start: ESP-IDF:          v4.0.1-dirty
I (528) cpu_start: Starting app cpu, entry point is 0x400812a8
0x400812a8: call_start_cpu1 at D:/esp32-4-1/esp32-4-1-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (539) heap_init: Initializing. RAM available for dynamic allocation:
I (545) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (551) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (557) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (564) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (570) heap_init: At 3FFCF9A0 len 00010660 (65 KiB): DRAM
I (576) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (582) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (589) heap_init: At 40092EF8 len 0000D108 (52 KiB): IRAM
I (595) cpu_start: Pro cpu start user code
I (613) spi_flash: detected chip: generic
I (614) spi_flash: flash io: dio
I (614) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (622) EXAMPLE: Initializing...
I (662) BTDM_INIT: BT controller compile version [c1cbe45]
I (662) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (772) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1832) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code: 0
I (1842) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code: 0
I (1842) EXAMPLE: BLE Mesh Fast Prov Node initialized
I (36482) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer: PB-GATT
I (36912) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer: PB-GATT
I (36912) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (36922) EXAMPLE: net_idx: 0x0001, unicast_addr: 0x0002
I (36922) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (39072) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013
[image]

在搜寻“W (39072) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013”相关的issues后了解到是外部设备发出了关闭的信号，但仍不明白如何解决问题
希望能够得到解答

Alvin1Zhang, 2020-07-06:
Thanks for reporting, we will look into.

Campou, 2020-07-06:
Hi @Beibixian 

After the EspBLEMesh App provisions the first device successfully, it will disconnect from the device. So the log above `W (39072) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013` is just for the disconnection.

Unfortunately the attached image can not be loaded. So if convenient, please provide the detailed logs of the 3 devices. Also the version of the App and the commit id of the idf.

Thanks.

Beibixian, 2020-07-07:
Thanks for your quick reply.
My EspBleMesh app version is 1.0.0(10) 
And the commit id of the idf is: v4.0.1
之前例程来源写错了，这个才是我所烧录的程序：
https://github.com/espressif/esp-idf/tree/v4.0.1/examples/bluetooth/esp_ble_mesh/ble_mesh_fast_provision

监视器得到另外两块开发板的logs完全相同，如下：
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6888
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (39) boot: compile time 09:53:23
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d33c (185148) map
I (176) esp_image: segment 1: paddr=0x0003d364 vaddr=0x3ffbdb60 size=0x02cac ( 11436) load
I (181) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xa9ae4 (695012) map
0x400d0018: _stext at ??:?

I (432) esp_image: segment 3: paddr=0x000e9b04 vaddr=0x3ffc080c size=0x015a0 (  5536) load
I (434) esp_image: segment 4: paddr=0x000eb0ac vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/esp32-4-1/esp32-4-1-idf/components/freertos/xtensa_vectors.S:1778

I (439) esp_image: segment 5: paddr=0x000eb4b4 vaddr=0x40080400 size=0x12af8 ( 76536) load
I (491) boot: Loaded app from partition at offset 0x10000
I (492) boot: Disabling RNG early entropy source...
I (492) cpu_start: Pro cpu up.
I (496) cpu_start: Application information:
I (500) cpu_start: Project name:     ble_mesh_fast_prov_server
I (507) cpu_start: App version:      1
I (511) cpu_start: Compile time:     Jul  7 2020 09:52:42
I (517) cpu_start: ELF file SHA256:  32b4ff33a591c6c0...
I (523) cpu_start: ESP-IDF:          v4.0.1-dirty
I (529) cpu_start: Starting app cpu, entry point is 0x400812a8
0x400812a8: call_start_cpu1 at D:/esp32-4-1/esp32-4-1-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (539) heap_init: Initializing. RAM available for dynamic allocation:
I (546) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (552) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (558) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (564) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (570) heap_init: At 3FFCF9A0 len 00010660 (65 KiB): DRAM
I (577) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (583) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (589) heap_init: At 40092EF8 len 0000D108 (52 KiB): IRAM
I (596) cpu_start: Pro cpu start user code
I (614) spi_flash: detected chip: generic
I (615) spi_flash: flash io: dio
I (615) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (623) FAST_PROV_SERVER_DEMO: Initializing...
I (663) BTDM_INIT: BT controller compile version [c1cbe45]
I (663) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (773) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1163) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code: 0
I (1173) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code: 0
I (1173) FAST_PROV_SERVER_DEMO: BLE Mesh Fast Prov Node initialized

这两块开发板的logs基本和app上选择配网那一块板子的logs相同（之前已附上）除了少了最后一句：
W (39072) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013

好像他们三块完全就是独立的个体，没有实现相互之间的通信，任选一块app快速配网并不会对其它两块板子的logs有任何新增影响......

Beibixian, 2020-07-07:
这另外两块板子的logs显示已配网的第一块并没有和他们产生任何通讯及影响，他们仍处于刚初始化完成未配网的阶段

Beibixian, 2020-07-07:
所有logs都已附上，未存在删减

Beibixian, 2020-07-07:
I see the same problem reported on: https://www.esp32.com/viewtopic.php?t=14488
The situation is just as same as mine, but after upgrading the app, the problem still exists on my boards......

Beibixian, 2020-07-07:
the same problem: https://esp32.com/viewtopic.php?f=2&t=11720&p=47201&hilit=provision+fast+provisioning&sid=754e1e3132f3a8febe224bfa0f779c62#p47201

Campou, 2020-07-08:
Hi @Beibixian 

The latest App is v1.1.0, you could download it from [here](https://github.com/EspressifApp/EspBLEMeshForAndroid/releases/tag/v1.1.0).

When using the App for fast provisioning, please input the number of the devices in the App (i.e. `需要快速配网节点个数`).

For the example, could you please have a try with the latest [release/v4.0 branch](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/bluetooth/esp_ble_mesh/ble_mesh_fast_provision)?

Thanks.

Beibixian, 2020-07-08:
> Hi @Beibixian
>
> The latest App is v1.1.0, you could download it from [here](https://github.com/EspressifApp/EspBLEMeshForAndroid/releases/tag/v1.1.0).
>
> When using the App for fast provisioning, [...]

build failed, here are logs:
Generated D:/esp32-4-1/workspace/fast_prov_server/build/bootloader/bootloader.bin
[1130/1142] Building C object esp-idf/fast_provisioning/CMakeFiles/__idf_fast_provisioning.dir/ble_mesh_fast_prov_client_model.c.obj
FAILED: esp-idf/fast_provisioning/CMakeFiles/__idf_fast_provisioning.dir/ble_mesh_fast_prov_client_model.c.obj
D:\esp32-4-1\esp32-4-1-tools\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -ID:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning -ID:/esp32-4-1/esp32-4-1-idf/components/newlib/platform_include -ID:/esp32-4-1/esp32-4-1-idf/components/freertos/include -ID:/esp32-4-1/esp32-4-1-idf/components/heap/include -ID:/esp32-4-1/esp32-4-1-idf/components/log/include -ID:/esp32-4-1/esp32-4-1-idf/components/soc/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/soc/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_rom/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_common/include -ID:/esp32-4-1/esp32-4-1-idf/components/xtensa/include -ID:/esp32-4-1/esp32-4-1-idf/components/xtensa/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/driver/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_ringbuf/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_event/include -ID:/esp32-4-1/esp32-4-1-idf/components/tcpip_adapter/include -ID:/esp32-4-1/esp32-4-1-idf/components/lwip/include/apps -ID:/esp32-4-1/esp32-4-1-idf/components/lwip/include/apps/sntp -ID:/esp32-4-1/esp32-4-1-idf/components/lwip/lwip/src/include -ID:/esp32-4-1/esp32-4-1-idf/components/lwip/port/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/lwip/port/esp32/include/arch -ID:/esp32-4-1/esp32-4-1-idf/components/vfs/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_wifi/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_wifi/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/esp_eth/include -ID:/esp32-4-1/esp32-4-1-idf/components/efuse/include -ID:/esp32-4-1/esp32-4-1-idf/components/efuse/esp32/include -ID:/esp32-4-1/esp32-4-1-idf/components/app_trace/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/common/osi/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/host/bluedroid/api/include/api -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_common/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_core -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_core/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_core/storage -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/btc/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_models/common/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_models/client/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/mesh_models/server/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/api/core/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/api/models/include -ID:/esp32-4-1/esp32-4-1-idf/components/bt/esp_ble_mesh/api -ID:/esp32-4-1/esp32-4-1-idf/components/nvs_flash/include -ID:/esp32-4-1/esp32-4-1-idf/components/spi_flash/include -ID:/esp32-4-1/esp32-4-1-idf/components/mbedtls/port/include -ID:/esp32-4-1/esp32-4-1-idf/components/mbedtls/mbedtls/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/fast_provisioning/CMakeFiles/__idf_fast_provisioning.dir/ble_mesh_fast_prov_client_model.c.obj -MF esp-idf\fast_provisioning\CMakeFiles\__idf_fast_provisioning.dir\ble_mesh_fast_prov_client_model.c.obj.d -o esp-idf/fast_provisioning/CMakeFiles/__idf_fast_provisioning.dir/ble_mesh_fast_prov_client_model.c.obj   -c D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_client_model.c
D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_client_model.c: In function 'example_fast_prov_client_recv_status':
D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_client_model.c:382:77: error: 'ESP_BLE_MESH_CID_NVAL' 
undeclared (first use in this function); did you mean 'ESP_BLE_MESH_SIG_MODEL'?
                                        ESP_BLE_MESH_MODEL_ID_GEN_ONOFF_CLI, ESP_BLE_MESH_CID_NVAL);
                                                                             ^~~~~~~~~~~~~~~~~~~~~
                                                                             ESP_BLE_MESH_SIG_MODEL
D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_client_model.c:382:77: note: each undeclared identifier is reported only once for each function it appears in
[... cut ...]
D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_operation.c:153:23: note: each undeclared identifier is reported only once for each function it appears in
D:/esp32-4-1/esp32-4-1-idf/examples/bluetooth/esp_ble_mesh/common_components/fast_provisioning/ble_mesh_fast_prov_operation.c:158:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1.exe: some warnings being treated as errors
[1135/1142] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ninja: build stopped: subcommand failed.
终端进程已终止，退出代码: 1

Beibixian, 2020-07-10:
> Hi @Beibixian
>
> The latest App is v1.1.0, you could download it from [here](https://github.com/EspressifApp/EspBLEMeshForAndroid/releases/tag/v1.1.0).
>
> When using the App for fast provisioning, [...]

More logs are listed below：

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6844
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF GIT-NOTFOUND 2nd stage bootloader
I (41) boot: compile time 14:16:27
I (41) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 2MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d07c (184444) map
I (178) esp_image: segment 1: paddr=0x0003d0a4 vaddr=0x3ffbdb60 size=0x02f6c ( 12140) load
I (183) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xa9ae4 (695012) map
0x400d0018: _stext at ??:?

I (433) esp_image: segment 3: paddr=0x000e9b04 vaddr=0x3ffc0acc size=0x012cc (  4812) load
I (435) esp_image: segment 4: paddr=0x000eadd8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/esp32/esp-idf/components/freertos/xtensa_vectors.S:1778

I (440) esp_image: segment 5: paddr=0x000eb1e0 vaddr=0x40080400 size=0x12af8 ( 76536) load
I (493) boot: Loaded app from partition at offset 0x10000
I (493) boot: Disabling RNG early entropy source...
I (494) cpu_start: Pro cpu up.
I (497) cpu_start: Application information:
I (502) cpu_start: Project name:     ble_mesh_fast_prov_server
I (508) cpu_start: App version:      1
I (513) cpu_start: Compile time:     Jul 10 2020 14:16:01
I (519) cpu_start: ELF file SHA256:  ecbcb5974cd48c5e...
I (525) cpu_start: ESP-IDF:          GIT-NOTFOUND
I (530) cpu_start: Starting app cpu, entry point is 0x400812a8
0x400812a8: call_start_cpu1 at D:/esp32/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (541) heap_init: Initializing. RAM available for dynamic allocation:
I (548) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (554) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (560) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (566) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (572) heap_init: At 3FFCF988 len 00010678 (65 KiB): DRAM
I (578) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (585) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (591) heap_init: At 40092EF8 len 0000D108 (52 KiB): IRAM
I (597) cpu_start: Pro cpu start user code
I (615) spi_flash: detected chip: generic
I (616) spi_flash: flash io: dio
[... 10 lines cut ...]
I (24977) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer: PB-GATT
I (25607) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer: PB-GATT
I (25607) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (25617) FAST_PROV_SERVER_DEMO: net_idx: 0x0001, unicast_addr: 0x0002
I (25627) FAST_PROV_SERVER_DEMO: flags: 0x00, iv_index: 0x00000000
W (55597) BT_GATT: gatt_rsp_timeout disconnecting...
W (79137) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013
I (83367) FAST_PROV_SERVER_DEMO: example_ble_mesh_config_server_cb, event = 0x00, opcode = 0x0000, addr: 0x0001
I (83367) FAST_PROV_SERVER_DEMO: Config Server get Config AppKey Add
I (97817) FAST_PROV_SERVER_DEMO: example_ble_mesh_custom_model_cb: Fast prov server receives msg, opcode 0xc602e5
I (97817) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_MODEL_SEND_COMP_EVT, err_code 0
I (97827) FAST_PROV_SERVER: example_handle_fast_prov_status_send_comp_evt: opcode 0xc702e5
I (102617) FAST_PROV_SERVER_DEMO: event 0x00, opcode 0x8203, src 0x0001, dst 0x0002
I (102617) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (102627) FAST_PROV_SERVER_DEMO: onoff 0x01
I (103727) FAST_PROV_SERVER_DEMO: event 0x00, opcode 0x8203, src 0x0001, dst 0x0002
I (103727) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (103737) FAST_PROV_SERVER_DEMO: onoff 0x00
I (112627) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_SET_FAST_PROV_ACTION_COMP_EVT, status_action 0x00
E (112627) BLE_MESH: ble_mesh_model_send_msg, Invalid destination address 0x0000
E (112637) FAST_PROV_SERVER: example_handle_fast_prov_action_set_comp_evt: Failed to send Fast Prov Status message
E (112647) FAST_PROV_SERVER_DEMO: example_ble_mesh_provisioning_cb: Failed to handle Fast Prov Action Set complete event

In (24977), I used app "EspBLEMesh" to fast provision the first board, but the other two boards were not provisioned by the first board. Then (83367) I connected the first board and turned LED on and off. After a few seconds, the errors came. 
Hoping to get your advise.

Beibixian, 2020-07-11:
> Hi @Beibixian
>
>
>
> The latest App is v1.1.0, you could download it from [here](https://github.com/EspressifApp/EspBLEMeshForAndroid/releases/tag/v1.1.0).
>
>
>
> When using the App for fast provisioning, [...]

I have tested the example you provided, but the problem still exists. 
I wanna know the three detailed logs when the devices work properly. Hoping to get your answers.

Campou, 2020-07-30:
Hi @Beibixian 

Does this issue still exist with the `EspBLEMesh` App? Seen you submitted another issue regarding the fast provisioning examples.

Thanks.

Beibixian, 2020-07-30:
> Hi @Beibixian 
> 
> 
> 
> Does this issue still exist with the `EspBLEMesh` App? Seen you submitted another issue regarding the fast provisioning examples.
> 
> 
> 
> Thanks.

Yes, it still exists. I have to replace the app with a board which runs fast provisioning client.

Campou, 2020-07-30:
Hi @Beibixian 

Found the cause of the issue from the log, i.e.

I (112627) FAST_PROV_SERVER_DEMO: ESP_BLE_MESH_SET_FAST_PROV_ACTION_COMP_EVT, status_action 0x00
E (112627) **BLE_MESH: ble_mesh_model_send_msg, Invalid destination address 0x0000**
E (112637) FAST_PROV_SERVER: example_handle_fast_prov_action_set_comp_evt: Failed to send Fast Prov Status message
E (112647) FAST_PROV_SERVER_DEMO: example_ble_mesh_provisioning_cb: Failed to handle Fast Prov Action Set complete event

Due to the invalid destination address, which causes the first device failing to send the response successfully. If convenient, could you please attach the test example here?

Thanks.

Beibixian, 2020-07-30:
> Hi @Beibixian
>
>
>
> The latest App is v1.1.0, you could download it from [here](https://github.com/EspressifApp/EspBLEMeshForAndroid/releases/tag/v1.1.0).
>
>
>
> When using the App for fast provisioning, [...]

The code was downloaded from here.

Campou, 2020-08-10:
Hi @Beibixian 

Really sorry here, I tries a few times using the latest `release/v4.0` branch, can't reproduce this issue. BTW, just noticed some differences here from the above logs.

[fast_prov_server.log](https://github.com/espressif/esp-idf/files/5049755/fast_prov_server.log)

The above is the attachment of the example test log on my side.

Thanks.

closed 2025-03-14 by Alvin1Zhang