espressif/esp-idf #5549: BLE BR/EDR Cross Transport Key Derivation (CTKD)

caleedubya, 2020-07-07:
Hi Everyone,

With ESP32 being Bluetooth 4.2 SIG compliant I would think it should be able to do CTKD but it's certainly not easy to figure it out. 

What is CTKD? For anyone using Dual Mode Bluetooth it's the ability to bond to either one of BLE or BT and have the both modes connected through calculating and deriving the key for the other mode not explicitly bonded by the user. 

I've been rooting around at this for quite some time and I've been able to find the following function within the BT Stack that looks promising... [smp_calculate_long_term_key_from_link_key](https://github.com/espressif/esp-idf/blob/f1b8723996d299f40d28a34c458cf55a374384e1/components/bt/host/bluedroid/stack/smp/smp_keys.c#L1962) but the question is how do we call this from app_main?

Having an example that shows how to perform this task while running Coex BLE would be very handy for the community. 

Thanks!

Wth-Esp, 2020-07-08:
Hi,
ESP32 does not support this feature yet.
Thanks

drewandre, 2020-07-08:
@Wth-Esp any chance this is on the product roadmap at all? @caleedubya and I have been active on other forums like [reddit](https://www.reddit.com/r/esp32/comments/hkm799/ble_a2dp_coexistence_iphone_shows_duplicate/) and the [esp32 forum](https://www.esp32.com/viewtopic.php?f=2&t=5537&start=10) trying to figure this out. We both use BLE and A2DP and having duplicate devices appear in the phone BT settings menu is frustrating for us and for my users. I had previously tried going down the MFi path to achieve a seamless dual mode integration but it proved too difficult to source MFi processors reliably and receive tech support for MFi/esp32 development. Having CTDK would solve this issue by consolidating dual mode esp32 profiles into one device while also preventing users from needing to manually pair the BT classic profile.

I think at the very least, the esp32 BT [coex example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex)'s readme could be updated since the profile names "ESP_COEX_BLE_DEMO" and "ESP_COEX_A2DP_DEMO" do not actually show up separately as suggested - at least on iOS, these are both consolidated into one name, but on what appear to a user as two devices, despite being two different BT profiles.

Wth-Esp, 2020-07-08:
@drewandre I have to talk with my teammates and share you with our decision ASAP.

chegewara, 2020-07-08:
Yes, it would be nice to have it.
I already have ble + classic, but it was single device app, not production, so i didnt bother and my app is showing 2 devices with the same name, both with pin code. Not very elegant solution, because i have to use classic bt with public address and ble with RPA.

caleedubya, 2020-07-08:
Hey guys,

What about this quote from the BTStack Manual:

"5.7.7. Cross-transport Key Derivation for LE Secure Connections. In a
dual- mode configuration, BTstack automatically generates an BR/EDR Link
Key from the LE LTK via the Link Key Conversion function h6. It is then
stored in the link key db.

To derive an LE LTK from a BR/EDR link key, the Bluetooth controller needs
to support Secure Connections via NIST P-256 elliptic curves and the LE
Secure Connections needs to get established via the LE Transport. BTstack
does not support LE Secure Connections via LE Transport currently."

When I read that it suggests to me that getting a BR/EDR key from a LE LTK
should be possible via the Link Key Conversion function. Going the other
way though and deriving a LE LTK from a BR/EDR link key looks to be
unsupported at this time.

So if I reading this correctly, to make this work we just need to call the
Link Key Conversion function. Any idea how we might do that?

Thoughts?

Dallas

Wth-Esp, 2020-07-09:
Hi all,

Thanks for your support, we agree to do this feature.
But this year we have much work to do on the new chips, so we decide to develop this feature in Q4 this year.

Thanks
Weitianhua

drewandre, 2020-07-10:
@Wth-Esp thank you, this is pretty exciting. This is the last step I was looking to overcome before bringing an esp32-based BT product to market this winter.

caleedubya, 2020-07-10:
I"ll echo's Drew's comments. Thank you and this is also my last step in
bringing a new ESP32-based product to market. Sincerely

drewandre, 2020-10-01:
@Wth-Esp the esp-idf [v4.2-beta1 release notes](https://github.com/espressif/esp-idf/releases/tag/v4.2-beta1) mention bluetooth 5.0 compatibility. With v4.2 stable looking likely to be released Q4 this year, should be expect cross transport key derivation support? This release looks really exciting. Thank you for all of your work!

erpalma, 2020-10-07:
I would really suggest you to have a look to [this](https://arxiv.org/abs/2009.11776) paper before moving on with CTKD.

drewandre, 2020-10-15:
Aren't these proposed vulnerabilities the responsibility of Bluetooth SIG and not Espressif though? I would imagine it's Espressif's goal to be in compliance with the bluetooth 4.2 standard whether or not it includes vulnerabilities. Either way, CTKD seems to be the only viable option we have to avoid user's having to manage two bluetooth connections for a product that requires bluetooth audio streaming alongside BLE commands (we've tried the MFi route, and LE audio isn't widely adopted yet). Thanks for sharing this though @erpalma, pretty interesting.

caleedubya, 2020-12-13:
> Hi all,
> 
> Thanks for your support, we agree to do this feature.
> But this year we have much work to do on the new chips, so we decide to develop this feature in Q4 this year.
> 
> Thanks
> Weitianhua

Hi @Wth-Esp, just wanted to check in to see if you guys are still on track for a Q4 release of CTKD functionality? There are a few of us where this feature is the last thing we need to release a ESP32 based product so we're anxiously awaiting it's release. :D

Wth-Esp, 2020-12-14:
Hi @caleedubya 

First of all, thanks for your patience. We have evaluated the development and the test work of this part. I'm so sorry that we can not push the feature as per the timeline, because of the security phase issue which may affect the Authentication and Encryption procedure of both BREDR and BLE. Now I cannot give you the timeline of the work, we have to test it more. But I will update the stage of development and test work here as soon as possible. Once again, sorry for the delay.

Thanks

caleedubya, 2022-05-26:
Hi @Wth-Esp, it's been a couple years since we last touched on this. Wanted to see if there is any chance to take another shot at this?

xupenghu, 2023-03-10:
Hi @ghost I also have the same need, please push as soon as possible.

blueMoodBHD, 2023-03-10:
We are very sorry that we have no development plan for this feature at the moment.

We have put this feature in featuer introduced list, and we will let you know if we starts working on it.