espressif/esp-idf #5550: Error of mbedtls in IDFv4.1-beta2

ROYMURLI, 2020-07-07:
Hi,
I was working with compiler version 5.2.0 and gcc v 6.4.0 in msys2 and was using IDFv4.1-dev-1086-g93a8603c5. Now I upgraded toolchiain to 8.2.0 and gcc v 9.1.0 in msys2 and using IDFv4.2-beta2. I am getting error in medtls components. Tried to include and exclude various components but not working. Screenshoot of error is:-
[image]

Suggest me what should be the changes in menuconfig so that it will run. latest SDK config is:-

[sdkconfig1.txt](https://github.com/espressif/esp-idf/files/4882958/sdkconfig1.txt)

mahavirj, 2020-07-08:
@ROYMURLI Can you please try enabling `MBEDTLS_ECP_C` from https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L639?

ROYMURLI, 2020-07-10:
Thanks @mahavirj . It worked.

closed 2020-07-10 by ROYMURLI