espressif/esp-idf #5553: HTTP empty header values handled incorrectly

edwin-oetelaar, 2020-07-08:
Http response headers with empty value (not sure if that is valid) copy remaining header, not an empty string.
The parser does not crash, it parses the remaining header correctly, it just duplicates the remaining header (including some body) as the value of the header.
Depending on the contents of the remaining http response and body, this may lead to out of memory situation.

example using this URL: http://22603.live.streamtheworld.com:80/TLPSTR20.mp3
The server responds with an empty header value.
```
icy-description:
```
but the callback returns 
```
D (36711) HTTP_CLIENT: HEADER=icy-description:icy-url: https://www.radio10.nl/80s-hits
icy-genre: Top 40
icy-br: 128
icy-metaint: 16000
Server: MediaGateway 5.4.0-0158

���
```

// wireshark transcript from other http client
```
GET /TLPSTR20.mp3 HTTP/1.0
Host: 22603.live.streamtheworld.com
Accept: text/html, text/plain, text/sgml, */*;q=0.01
Accept-Encoding: gzip, compress, bzip2
Accept-Language: en
User-Agent: Lynx/2.8.9dev.16 libwww-FM/2.14 SSL-MM/1.4.1 GNUTLS/3.5.17
```

```
HTTP/1.0 200 OK
Expires: Thu, 01 Dec 2003 16:00:00 GMT
Cache-Control: no-cache, must-revalidate
Pragma: no-cache
Access-Control-Allow-Methods: GET, HEAD
Access-Control-Allow-Origin: *
Access-Control-Allow-Credentials: true
Set-Cookie: uuid=1ad7a8d7-f45f-4346-9516-57762ce66aa4; expires=Fri, 07 Aug 2020 12:44:31 GMT; path=/; domain=.live.streamtheworld.com
Set-Cookie: uuid-s=1ad7a8d7-f45f-4346-9516-57762ce66aa4; expires=Fri, 07 Aug 2020 12:44:31 GMT; path=/; domain=.live.streamtheworld.com; SameSite=None; Secure; HttpOnly
Content-Type: audio/mpeg
Connection: close
icy-name: Radio 10 80s hits
icy-description: 
icy-url: https://www.radio10.nl/80s-hits
icy-genre: Top 40
icy-br: 128
Server: MediaGateway 5.4.0-0158

...d
```

use the normal example ```https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c```

with above url to test

edwin-oetelaar, 2020-07-09:
I fixed it, the bug was in the semantics of the 
```components/esp_http_client/lib/http_utils.c```
the function
```
char *http_utils_assign_string(char **str, const char *new_str, int len)
```

if called with a zero length so :  ```len == 0``` which is valid, it does the wrong thing, it allocated memory.

New semantics:
- if len < 0 : allocate memory

this means all calls had to change which called it with len==0

See attached patch. It has been tested.

```diff --git a/components/esp_http_client/esp_http_client.c b/components/esp_http_client/esp_http_client.c
index bca80fad9..957d110f9 100644
--- a/components/esp_http_client/esp_http_client.c
+++ b/components/esp_http_client/esp_http_client.c
@@ -418,12 +418,12 @@ static esp_err_t _set_config(esp_http_client_handle_t client, const esp_http_cli
     }
 
     if (config->transport_type == HTTP_TRANSPORT_OVER_SSL) {
-        http_utils_assign_string(&client->connection_info.scheme, "https", 0);
+        http_utils_assign_string(&client->connection_info.scheme, "https", -1);
         if (client->connection_info.port == 0) {
             client->connection_info.port = DEFAULT_HTTPS_PORT;
         }
     } else {
-        http_utils_assign_string(&client->connection_info.scheme, "http", 0);
+        http_utils_assign_string(&client->connection_info.scheme, "http", -1);
         if (client->connection_info.port == 0) {
             client->connection_info.port = DEFAULT_HTTP_PORT;
         }
@@ -767,10 +767,10 @@ esp_err_t esp_http_client_set_url(esp_http_client_handle_t client, const char *u
             if (password) {
                 *password = 0;
                 password ++;
-                http_utils_assign_string(&client->connection_info.password, password, 0);
+                http_utils_assign_string(&client->connection_info.password, password, -1);
                 HTTP_MEM_CHECK(TAG, client->connection_info.password, return ESP_ERR_NO_MEM);
             }
-            http_utils_assign_string(&client->connection_info.username, username, 0);
+            http_utils_assign_string(&client->connection_info.username, username, -1);
             HTTP_MEM_CHECK(TAG, client->connection_info.username, return ESP_ERR_NO_MEM);
             free(user_info);
         } else {
@@ -782,7 +782,7 @@ esp_err_t esp_http_client_set_url(esp_http_client_handle_t client, const char *u
     if (purl.field_data[UF_PATH].len) {
         http_utils_assign_string(&client->connection_info.path, url + purl.field_data[UF_PATH].off, purl.field_data[UF_PATH].len);
     } else {
-        http_utils_assign_string(&client->connection_info.path, "/", 0);
+        http_utils_assign_string(&client->connection_info.path, "/", -1);
     }
     HTTP_MEM_CHECK(TAG, client->connection_info.path, return ESP_ERR_NO_MEM);
 
@@ -840,7 +840,7 @@ bool esp_http_client_is_complete_data_received(esp_http_client_handle_t client)
     return true;
 }
 
-int esp_http_client_read(esp_http_client_handle_t client, char *buffer, int len)
+int esp_http_client_read(esp_http_client_handle_t client, char * const buffer, int len)
 {
     esp_http_buffer_t *res_buffer = client->response->buffer;
 
diff --git a/components/esp_http_client/lib/http_header.c b/components/esp_http_client/lib/http_header.c
index b3dda8eea..977efcba7 100644
--- a/components/esp_http_client/lib/http_header.c
+++ b/components/esp_http_client/lib/http_header.c
@@ -86,10 +86,10 @@ static esp_err_t http_header_new_item(http_header_handle_t header, const char *k
 
     item = calloc(1, sizeof(http_header_item_t));
     HTTP_MEM_CHECK(TAG, item, return ESP_ERR_NO_MEM);
-    http_utils_assign_string(&item->key, key, 0);
+    http_utils_assign_string(&item->key, key, -1);
     HTTP_MEM_CHECK(TAG, item->key, goto _header_new_item_exit);
     http_utils_trim_whitespace(&item->key);
-    http_utils_assign_string(&item->value, value, 0);
+    http_utils_assign_string(&item->value, value, -1);
     HTTP_MEM_CHECK(TAG, item->value, goto _header_new_item_exit);
     http_utils_trim_whitespace(&item->value);
     STAILQ_INSERT_TAIL(header, item, next);
diff --git a/components/esp_http_client/lib/http_utils.c b/components/esp_http_client/lib/http_utils.c
index 267e39e6d..725ab2b6b 100644
--- a/components/esp_http_client/lib/http_utils.c
+++ b/components/esp_http_client/lib/http_utils.c
@@ -45,7 +45,7 @@ char *http_utils_assign_string(char **str, const char *new_str, int len)
         return NULL;
     }
     char *old_str = *str;
-    if (l <= 0) {
+    if (l < 0) { // fix
         l = strlen(new_str);
     }
     if (old_str) {
```

Do you want a pull request instead?

Best regards,
Edwin van den Oetelaar

shubhamkulkarni97, 2020-08-13:
Hi @edwin-oetelaar,

Fix looks good and we can go ahead with a pull request if you would like. Otherwise, I'll try to add this fix.

Thanks,
Shubham

edwin-oetelaar, 2020-08-19:
Just apply the patch. Just a few lines. Thanks, Edwin

closed 2020-09-16 by espressif-bot