espressif/esp-idf #5555: CMake or CMakeFiles.txt bug

ammaree, 2020-07-08:
## Environment
- Development Kit:      [none]
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1905-g625bd5eb1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp-2020r2
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [Eclipse]]

## Problem Description
If I hardcode the list of PRIV_REQUIRED components as per the WORKING CMakeLists.txt (listed below) the output in WORKING OUTPUT is produced and the compile succeeds.

If I programmatically construct the PRIV_REQUIRES list of components as per the FAILING CMakeLists.txt (listed below) the output in FAILING output is produced.

What is visible is that in the FAILING example the 2 additional components "pca9555" and "onewire" are NOT added to the PRIV_REQUIRES list although they are present in exactly the same format in ${priv_requires} prior to idf_component_register() being called.

### Expected Behavior
Both options to perform exactly the same.

### Actual Behavior
Programmatic option fails.

### Code to reproduce this issue
WORKING CMakeLists.txt
>get_property( MODEL_SEL GLOBAL PROPERTY MODEL_SEL )
message( "MODEL=${MODEL_SEL}  COMPONENT=${COMPONENT_NAME}" )
set(srcs "commands.c" )
set(include . )
set(priv_include_dirs . )
set(requires "common" )
set(priv_requires "pca9555" "onewire" "rtos-support" "mqtt actuators" "rules" "identity" "http" "stringsX" "printf" "syslog" "systiming" "telnet_server" "hal_esp32" "lwip" )
idf_component_register( SRCS "${srcs}"
						INCLUDE_DIRS ${include} 
						PRIV_INCLUDE_DIRS ${priv_include_dirs} 
						REQUIRES ${requires} 
						PRIV_REQUIRES ${priv_requires} )	
idf_component_get_property(VAR ${COMPONENT_NAME} REQUIRES)
message("########### NEW REQUIRES: ${VAR}" )
idf_component_get_property(VAR ${COMPONENT_NAME} PRIV_REQUIRES)
message("###### NEW PRIV_REQUIRES: ${VAR}" )

WORKING output
>MODEL=1  COMPONENT=commands
NEW REQUIRES: common;cxx;newlib;freertos;heap;log;lwip;soc;esp_rom;esp_common;esp_system;xtensa;esp32
NEW PRIV_REQUIRES: pca9555;onewire;rtos-support;mqtt;actuators;rules;identity;http;stringsX;printf;syslog;systiming;telnet_server;hal_esp32;lwip

FAILING CMakeLists.txt
>get_property( MODEL_SEL GLOBAL PROPERTY MODEL_SEL )
message( "MODEL=${MODEL_SEL}  COMPONENT=${COMPONENT_NAME}" )
set(srcs "commands.c" )
set(include . )
set(priv_include_dirs . )
set(requires "common" )
if( MODEL_SEL MATCHES 1 )
  set(priv_requires "pca9555" "onewire" "rtos-support" "mqtt" "actuators" "rules" "identity" "http" "stringsX" "printf" "syslog" "systiming" "telnet_server" "hal_esp32" "lwip" )
else()
  set(priv_requires "rtos-support" "mqtt" "actuators" "rules" "identity" "http" "stringsX" "printf" "syslog" "systiming" "telnet_server" "hal_esp32" "lwip" )
endif()
idf_component_register( SRCS "${srcs}"
						INCLUDE_DIRS ${include}
						PRIV_INCLUDE_DIRS ${priv_include_dirs}
						REQUIRES ${requires}
						PRIV_REQUIRES ${priv_requires} )
message( "Step 0: requires = ${requires}" )
message( "Step 0: priv_requires = ${priv_requires}" )
idf_component_get_property( var ${COMPONENT_NAME} REQUIRES )
message( "Step 2: REQUIRES: ${var}" )
idf_component_get_property( var ${COMPONENT_NAME} PRIV_REQUIRES )
message( "Step 2: PRIV_REQUIRES: ${var}" )

FAILING output
>MODEL=1  COMPONENT=commands
Step 0: requires = common
Step 0: priv_requires = pca9555;onewire;rtos-support;mqtt;actuators;rules;identity;http;stringsX;printf;syslog;systiming;telnet_server;hal_esp32;lwip
Step 2: REQUIRES: common;cxx;newlib;freertos;heap;log;lwip;soc;esp_rom;esp_common;esp_system;xtensa;esp32
Step 2: PRIV_REQUIRES: rtos-support;mqtt;actuators;rules;identity;http;stringsX;printf;syslog;systiming;telnet_server;hal_esp32;lwip

FAILING error:
>C:/Dropbox/devs/ws/z-components/commands/commands.c:42:11: fatal error: pca9555.h: No such file or directory
  #include "pca9555.h"

Any and all help highly appreciated.

ammaree, 2020-07-08:
I have tried to achieve the same objective, programmatically building up the PRIV_REQUIRES list, using a number of different steps.
I have also tried the different steps in both the top level CMakeLists.txt as well as in a component CMakeLists.txt.
Certain steps seem to work in the top level project but not in a component, especially when some steps are optionally added subject to conditionals tests.
The following sequence works with selected lines commented out, as required.
>set( priv_requires )
set( priv_requires ${priv_requires} pca9555 )
set( priv_requires ${priv_requires} onewire )
set( priv_requires ${priv_requires} app_update buffers endpoints http-parser identity irmacos libyuarel nvs_flash rtos-support stringsX syslog printf systiming x_sockets )

The sequence above works perfectly if I manually comment out specific lines based on device capabilities
As soon as I try to add logic checks to automatically adapt the flow based on global  variables, the process fails.
>set( priv_requires app_update buffers endpoints http-parser identity irmacos libyuarel nvs_flash rtos-support stringsX syslog printf systiming x_sockets )
get_property(MODEL_DEV GLOBAL PROPERTY MODEL_DEV )
get_property(MODEL_GUI GLOBAL PROPERTY MODEL_GUI )
message("${COMPONENT_NAME}: MODEL_DEV=${MODEL_DEV}  MODEL_GUI=${MODEL_GUI}" )
if(${MODEL_GUI} MATCHES 1)
  set( priv_requires ${priv_requires} ili9341 )
endif()
if(${MODEL_DEV} MATCHES 1)
  set( priv_requires ${priv_requires} pca9555 )
  set( priv_requires ${priv_requires} onewire )
endif()
message( "priv_requires: ${priv_requires}" )
idf_component_register( SRCS ${srcs} INCLUDE_DIRS ${include_dirs} PRIV_INCLUDE_DIRS ${priv_include_dirs} REQUIRES ${requires} PRIV_REQUIRES ${priv_requires} )
idf_component_get_property( var ${COMPONENT_NAME} PRIV_REQUIRES )
message( "PRIV_REQUIRES: ${var}" )

Alvin1Zhang, 2020-07-09:
Thanks for reporting, we will look into.

renzbagaporo, 2020-07-09:
The build system has two passes. The first pass is where the requirements are evaluated and is done in a different CMake invocation, and thus, no access to the parent's properties (MODEL_SEL in your example). Thus in the first passes' context, the check will always evaluated to false.

Only the following are accessible in the first pass:

- IDF build/component properties
- Kconfig configurations

ammaree, 2020-07-09:
@renzbagaporo Thanks for the feedback.

I originally used the esp-idf/components/freertos/CMakeLists.txt file as a reference to test my idea.
On a second reading of the same file lines 41/2 states
* app_trace is required by FreeRTOS headers only when CONFIG_SYSVIEW_ENABLE=y
* but requirements can't depend on config options, so always require it.

Does this mean that even if I change by MODEL_SEL & BUILD_SEL variables to become CONFIG_???? and move them to my Kconfig.projbuild file I will still not be successful, due to line 42?

renzbagaporo, 2020-07-13:
Yes, requirements cannot depend on configuration.

ammaree, 2020-07-14:
@renzbagaporo  I have largely resolved the problem by doing the following.
In the top level project CMakeLists.txt file I create a variable containing a list of the "optional" components required by the specific HW model eg.
```
set( OC_LIST onewire pca9555 )
set( OCD_LIST  ${PC}/onewire  ${PC}/pca9555 )
```

Should it be required to add additional components these are APPENDed eg.
```
set( OC_LIST ${OC_LIST} fonts ili9341 )
set( OCD_LIST  ${OCD_LIST}  ${PC}/ili9341  ${PC}/fonts )
```

The additional components are then added to the COMPONENT list...
```
set( COMPONENTS esp32 main esptool_py ${OC_LIST} )
set(EXTRA_COMPONENT_DIRS	${PC}/actuators ${PC}/buffers ${PC}/commands ${PC}/common ${PC}/endpoints ${PC}/hal_${PH} 
                    ${PC}/http ${PC}/identity ${PC}/irmacos ${PC}/log ${PC}/mqtt ${PC}/printf ${PC}/rtos-support
                    ${PC}/rules ${PC}/sitewhere ${PC}/sntp_client ${PC}/statistics ${PC}/stringsX ${PC}/syslog
                    ${PC}/systiming ${PC}/telnet_server ${PC}/values ${PC}/x_json ${PC}/x_sockets
                    ${PS}/http-parser ${PS}/libyuarel ${PS}/nanopb
                    ${OCD_LIST} )
```
Lastly, in each components that [PRIV_]REQUIRES any of the additional components I use the similar CMakeLists.txt files:
```
set( srcs "commands.c" )
set( include_dirs "." )
set( priv_include_dirs "." )
set( requires common )
set( priv_requires rtos-support mqtt actuators rules identity http stringsX printf syslog systiming telnet_server hal_esp32 lwip )
idf_build_get_property( build_components BUILD_COMPONENTS )
foreach( req ${OC_LIST} )
    if( req IN_LIST build_components)
		set( priv_include_dirs ${priv_include_dirs} ${PC}/${req} )
		set( priv_requires ${priv_requires} ${req} )
    endif()
endforeach()
idf_component_register( SRCS ${srcs}
                    INCLUDE_DIRS ${include_dirs} 
                    PRIV_INCLUDE_DIRS ${priv_include_dirs} 
                    REQUIRES ${requires} 
                    PRIV_REQUIRES ${priv_requires}
                    )
```
This mechanism works perfectly for the compile stage, all components are found, the include directories are added to the lists and the libraries are built. However, during the link stage the additional COMPONENT libraries are not found and a list of undefined references are reported.

> [6/7] Linking CXX executable c77a5342-06ae-49eb-99e8-bf0d84da7829.elf
FAILED: c77a5342-06ae-49eb-99e8-bf0d84da7829.elf
cmd.exe /C "cd . && C:\Users\andre\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\c77a5342-06ae-49eb-99e8-bf0d84da7829.elf.rsp  -o c77a5342-06ae-49eb-99e8-bf0d84da7829.elf  && cd ."
c:/users/andre/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/commands/libcommands.a(commands.c.obj):(.literal.vCommandInterpret+0x20): undefined reference to `m90e26Report'
c:/users/andre/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/commands/libcommands.a(commands.c.obj):(.literal.vCommandInterpret+0x24): undefined reference to `m90e26LoadNVSConfig'
c:/users/andre/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/commands/libcommands.a(commands.c.obj):(.literal.vCommandInterpret+0x2c): undefined reference to `ssd1306Report'
c:/users/andre/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/commands/libcommands.a(commands.c.obj): in function `vCommandInterpret':
C:/Dropbox/devs/ws/z-components/commands/commands.c:408: undefined reference to `m90e26Report'

Any suggestions to resolve this last obstacle will be appreciated.
@igrr If you can chip in with some advice, doubly appreciated

renzbagaporo, 2020-07-14:
Does the component `commands` have all its requirements set as well? That is, the component the contains definitions for `m90e26LoadNVSConfig` and `vCommandInterpret` is included in its `REQUIRES` or `PRIV_REQUIRES`?

Alvin1Zhang, 2020-08-13:
@ammaree Thanks for reporting, would you please help share if any updates for this issue? Thanks.

ammaree, 2020-08-14:
@Alvin1Zhang 
The issue has not been resolved, I have simply had to go back to the manual process and add the additional/optional components whenever I change the device model.

I am still in need of a solution to automate this and would appreciate some help.

renzbagaporo, 2020-08-17:
Hi @ammaree. I have a suspicion that this part:

```
foreach( req ${OC_LIST} )
    if( req IN_LIST build_components)
		set( priv_include_dirs ${priv_include_dirs} ${PC}/${req} )
		set( priv_requires ${priv_requires} ${req} )
    endif()
endforeach()
```

Does not actually work. Since during the expansion (before `idf_component_register`), `BUILD_COMPONENTS` is not evaluated yet, and it does not know `OC_LIST` as well.

This seem to be a weak (optional) dependencies case, similar to something in ESP-IDF: https://github.com/espressif/esp-idf/blob/master/components/esp_common/CMakeLists.txt#L37-L56. In this, `esp_common` only links in the components enumerated in `optional_reqs` if they are in the build.

Important: Notice this is done after `idf_component_register`.

ammaree, 2020-08-17:
Ultimately it seems I am not going to get this working, so will close the issue.

closed 2020-08-17 by ammaree

AronRubin, 2021-03-03:
I have just altered the esp-nimble-cpp library with a workaround. That library needs to depend on the arduino component iff it exists.
```cmake
idf_build_get_property(__hack_component_targets __COMPONENT_TARGETS)

if("esp-nimble-component" IN_LIST BUILD_COMPONENTS OR "__esp-nimble-component" IN_LIST __hack_component_targets)
    list(APPEND ESP_NIMBLE_PRIV_REQUIRES esp-nimble-component)
elseif("nimble" IN_LIST BUILD_COMPONENTS OR "__nimble" IN_LIST __hack_component_targets)
    list(APPEND ESP_NIMBLE_PRIV_REQUIRES nimble)
endif()

if("arduino" IN_LIST BUILD_COMPONENTS OR __hack_component_targets MATCHES "__idf_arduino")
    list(APPEND ESP_NIMBLE_PRIV_REQUIRES arduino)
endif()

idf_component_register(
  REQUIRED_IDF_TARGETS
    "esp32"
    "esp32s3"
    "esp32c3"
  INCLUDE_DIRS
    "src"
  SRCS
    "src/FreeRTOS.cpp"
    "src/NimBLE2904.cpp"
    "src/NimBLEAddress.cpp"
    "src/NimBLEAdvertisedDevice.cpp"
    "src/NimBLEAdvertising.cpp"
    "src/NimBLEBeacon.cpp"
    "src/NimBLECharacteristic.cpp"
    "src/NimBLEClient.cpp"
    "src/NimBLEDescriptor.cpp"
    "src/NimBLEDevice.cpp"
    "src/NimBLEEddystoneTLM.cpp"
    "src/NimBLEEddystoneURL.cpp"
    "src/NimBLEHIDDevice.cpp"
    "src/NimBLERemoteCharacteristic.cpp"
    "src/NimBLERemoteDescriptor.cpp"
    "src/NimBLERemoteService.cpp"
    "src/NimBLEScan.cpp"
    "src/NimBLESecurity.cpp"
    "src/NimBLEServer.cpp"
    "src/NimBLEService.cpp"
    "src/NimBLEUtils.cpp"
    "src/NimBLEUUID.cpp"
  REQUIRES
    bt
    nvs_flash
  PRIV_REQUIRES
    ${ESP_NIMBLE_PRIV_REQUIRES}
)
```