espressif/esp-idf #5557: Secure Boot v2 bootloader signing during build fails on Windows

devanlai, 2020-07-08:
## Environment

- Development Kit:      Custom PCBA
- Module or chip used:  [ESP32-WROVER-E]
- IDF version: v4.2-dev-1905-g625bd5eb1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: VSCode with official Espressif IDF extension
- Power Supply:         USB

## Problem Description

The firmware bootloader signing step first copies the unsigned bootloader to `bootloader-unsigned.bin`, then signs it to produce `bootloader.bin` and generate `.signed_bin_timestamp`.

In the custom command it uses to do so, it directly invokes the `cp` command, which is not available in a native Windows command prompt environment.

This originates from [line 108 in bootloader/subproject/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/a9ccc5e5c8cdeb6a051023101bb11b577d971823/components/bootloader/subproject/CMakeLists.txt#L108):
```
COMMAND cp "${CMAKE_BINARY_DIR}/${PROJECT_BIN}" "${CMAKE_BINARY_DIR}/${bootloader_unsigned_bin}"
```

I'm not very familiar with CMake, but based on a [StackOverflow answer](https://stackoverflow.com/a/18602513), I replaced it with the following line:
```
COMMAND ${CMAKE_COMMAND} -E copy "${CMAKE_BINARY_DIR}/${PROJECT_BIN}" "${CMAKE_BINARY_DIR}/${bootloader_unsigned_bin}"
```

This seems to resolve it for me, but as I'm not well-versed in CMake, I can't say if that's the right way to resolve the issue.

### Expected Behavior

The bootloader signing step should work on Windows.

### Actual Behavior

The bootloader signing step fails on Windows:

### Steps to reproduce

1. Open the [`getting_started/hello_world` example](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world)
2. In menuconfig, select:
   * Component config > ESP32-specific > Minimum Supported ESP32 Revision > Rev 3
   * Security features > Enable hardware Secure Boot in bootloader
   * Under Security Features, "Sign Binaries during build" should be enabled by default
3. Generate a secure boot signing key using `espsecure generate_signing_key --version 2 secure_boot_signing_key.pem`
4. Run `idf.py build`

### Code to reproduce this issue
N/A

## Debug Logs

Excerpt from the build log:
```
[101/103] Linking C executable bootloader.elf
==============================================================================
Bootloader built. Secure boot enabled, so bootloader not flashed automatically.
Secure boot enabled, so bootloader not flashed automatically.
        C:\Users\Devan\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe C:/Users/Devan/Documents/GitHub/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
==============================================================================
[102/103] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
[103/103] Generated the signed Bootloader
FAILED: .signed_bin_timestamp
cmd.exe /C "cd /D C:\Users\Devan\Documents\GitHub\esp-idf\examples\get-started\hello_world\build\bootloader && cp C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader-unsigned.bin && C:\Users\Devan\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe C:/Users/Devan/Documents/GitHub/esp-idf/components/esptool_py/esptool/espsecure.py sign_data --version 2 --keyfile C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/secure_boot_signing_key.pem -o C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader-unsigned.bin && C:\Users\Devan\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E echo "Generated signed binary image C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin" "from C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader-unsigned.bin" && C:\Users\Devan\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E md5sum C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin > C:/Users/Devan/Documents/GitHub/esp-idf/examples/get-started/hello_world/build/bootloader/.signed_bin_timestamp"
'cp' is not recognized as an internal or external command, operable program or batch file.
ninja: build stopped: subcommand failed.
[462/944] Building C object esp-idf/dr...iles/__idf_driver.dir/spi_common.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

## Other items if possible

- [sdkconfig](https://github.com/espressif/esp-idf/files/4893621/sdkconfig.txt) sdkconfig file

Alvin1Zhang, 2020-07-09:
Thanks for the detailed report and letting us know, we will look into.

closed 2020-08-03 by espressif-bot

melvinItude, 2020-10-13:
Why isn't this merged in any v4x releases yet???

projectgus, 2020-10-14:
Hi @melvinItude ,

I'm sorry that these fixes aren't in any stable release yet.

They were backported to release/v4.1 branch in https://docs.espressif.com/projects/esp-idf/en/stable/versions.html#updating-to-a-release-branch and release/v4.2 branch in 156afc0c0587b67726b74e6edfe74e699940d9d2. So the fix will be part of the v4.1.1 release and the v4.2 release, once they are available.

Unfortunately it was not merged in time for the v4.2-beta1 pre-release (because we have an internal release test process, there is some delay between when a commit is merged and when it gets included in the release).

If you need to apply the fix now, you can choose to check out the release branch of ESP-IDF as described here: https://docs.espressif.com/projects/esp-idf/en/stable/versions.html#updating-to-a-release-branch

lajollalerner, 2020-10-30:
Hi all,
I followed the manual procedures for signing and buring the efuses for V2 secure boot. I am able to verify that the signed code (both bootloader and app) are signed using the espsecure verify_signature command. my two questions are, 
1. Do I use espsecure.py sign_data on the bootloader.bin and the app.bin or is there a different way to sign the bootloader? 
2. when I used espefuse.py burn_key I noticed the printout of blk2 data is reversed from the bin file produced by espsecure.py digest_rsa_public_key. Is that correct?

Thank you,
Ian

Alvin1Zhang, 2020-12-10:
IDF v4.2 has released at https://github.com/espressif/esp-idf/releases/tag/v4.2