espressif/esp-idf #5558: Build fails for target ESP32-S2 with optimisation for performance (-O2)

boarchuz, 2020-07-09:
## Environment

- Development Kit: ESP32-S2-SAOLA-1M
- IDF version: v4.2-dev-1905-g625bd5eb1
- Build System: CMake
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- IDE: VSCode with ESP-IDF Extension

## Problem Description

Project fails to compile with target set to ESP32-S2 and CONFIG_COMPILER_OPTIMIZATION_PERF=y:
```
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
```
See:
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/freertos/tasks.c#L648

## Steps to reproduce

* Create new directory
* Open in VSCode
* ESP-IDF: Create project -> template-app (creates basic IDF structure, main.c, CMakeLists, etc)
* main.c -> `void app_main(void) {}`
* ESP-IDF: Set Espressif device target -> ESP32-S2
* ESP-IDF: Launch gui configuration tool -> Compiler Options -> Optimization Level -> Optimize for Performance (-O2)
* Build

Note that it builds fine with these alternate configs:
* ESP32S2 with default config (ie. -Og)
* ESP32 with -O2
* ESP32 with CONFIG_FREERTOS_UNICORE=y
* ESP32 with CONFIG_FREERTOS_UNICORE=y and -O2

## Full Log
<details>
<summary>Click</summary>

```
> Executing task: cmake --build . <

[1/396] Performing build step for 'bootloader'
ninja: no work to do.
[19/394] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj
FAILED: esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj 
/home/matt/Espressif/Tools/master/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -D_ESP_FREERTOS_INTERNAL -Iconfig -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos/include -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos/xtensa/include -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos/include/freertos -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos/xtensa/include/freertos -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos/xtensa -I/home/matt/Espressif/IDF/master/esp-idf/components/freertos -I/home/matt/Espressif/IDF/master/esp-idf/components/newlib/platform_include -I/home/matt/Espressif/IDF/master/esp-idf/components/heap/include -I/home/matt/Espressif/IDF/master/esp-idf/components/log/include -I/home/matt/Espressif/IDF/master/esp-idf/components/lwip/include/apps -I/home/matt/Espressif/IDF/master/esp-idf/components/lwip/include/apps/sntp -I/home/matt/Espressif/IDF/master/esp-idf/components/lwip/lwip/src/include -I/home/matt/Espressif/IDF/master/esp-idf/components/lwip/port/esp32/include -I/home/matt/Espressif/IDF/master/esp-idf/components/lwip/port/esp32/include/arch -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/src/esp32s2/. -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/src/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_rom/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_common/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_system/include -I/home/matt/Espressif/IDF/master/esp-idf/components/xtensa/include -I/home/matt/Espressif/IDF/master/esp-idf/components/xtensa/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/driver/include -I/home/matt/Espressif/IDF/master/esp-idf/components/driver/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_ringbuf/include -I/home/matt/Espressif/IDF/master/esp-idf/components/efuse/include -I/home/matt/Espressif/IDF/master/esp-idf/components/efuse/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/espcoredump/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_timer/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_ipc/include -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/soc/esp32s2/. -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/soc/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/soc/soc/esp32s2/../include -I/home/matt/Espressif/IDF/master/esp-idf/components/vfs/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_wifi/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_wifi/esp32s2/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_event/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_netif/include -I/home/matt/Espressif/IDF/master/esp-idf/components/esp_eth/include -I/home/matt/Espressif/IDF/master/esp-idf/components/tcpip_adapter/include -I/home/matt/Espressif/IDF/master/esp-idf/components/app_trace/include -mlongcalls   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1905-g625bd5eb1\" -DESP_PLATFORM -MD -MT esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj -MF esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj.d -o esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj   -c /home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'xTaskCreatePinnedToCore':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript -1 is below array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'vTaskPrioritySet':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'vTaskResume':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'xTaskResumeFromISR':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'xTaskRemoveFromEventList':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
  TCB_t *curTCB = pxCurrentTCB[xCoreID];
                  ~~~~~~~~~~~~^~~~~~~~~
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c: In function 'xTaskRemoveFromUnorderedEventList':
/home/matt/Espressif/IDF/master/esp-idf/components/freertos/tasks.c:648:30: error: array subscript [0, 0] is outside array bounds of 'TCB_t * volatile[1]' {aka 'struct tskTaskControlBlock * volatile[1]'} [-Werror=array-bounds]
[... 1090 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

```
</summary>

Alvin1Zhang, 2020-07-09:
Thanks for the detailed report and letting us know, we will look into.

projectgus, 2020-07-09:
Hi @boarchuz,

Yes, thanks for that. This was actually reported last week as #5525 but the reporter closed the issue. Nevertheless we have a fix ready to merge, sorry for the inconvenience.

boarchuz, 2020-07-09:
Oops didn't see that one, thanks @projectgus. I'll close this now then.

projectgus, 2020-07-09:
The other one was closed as well (don't know why), so we can keep this one open. :)

igrr, 2020-09-02:
Fixed in https://github.com/espressif/esp-idf/commit/39bdbd7e33dd47c7f8dc3f24a40b93dcd3b940c5.

closed 2020-09-02 by igrr