espressif/esp-idf #5559: how to tx/rx data base on the vendor model through the BLE Mesh Network

NOA-Vernon, 2020-07-09:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

How do I use the vendor model for TX and RX base on the ble_mesh_vendor_model/vendor_client demo,　　delete the provision function and left only vendor model on ble_mesh_vendor_model/vendor_client demo

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wroom-32]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it):
     v4.2-dev-1905-g625bd5eb1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [|Yes (Eclipse)]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
The devices and TX/RX base on the vendor model through BLE MESH network.

### Actual Behavior
one the client device can't send a message when the press the button(BOOT0), the error code tells it's an invalid parameter.

### Steps to reproduce

1. create the project base on the vendor_client demo on Eclipse IDE;
2. delete the provision function and left the vendor client model;
3. I have paste the main.c on the following section;

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
/* main.c - Application main entry point */

/*
 * Copyright (c) 2018 Espressif Systems (Shanghai) PTE LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */
[... 457 lines cut ...]
[0;32mI (8214) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT[0m
[0;32mI (8214) EXAMPLE: net_idx 0x000, addr 0x0002[0m
[0;32mI (8214) EXAMPLE: flags 0x00, iv_index 0x00000000[0m
[0;33mW (8214) BOARD: led green is already off[0m
[0;31mE (11564) BT_GATT: GATT_SendServiceChangeIndication can't send service change indication manually, please configure the option through menuconfig[0m
[0;33mW (14574) BLE_MESH: Composition page 255 not available[0m
[0;33mW (15064) BLE_MESH: No matching TX context for ack[0m
[0;32mI (17054) EXAMPLE: ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD[0m
[0;32mI (17054) EXAMPLE: net_idx 0x0000, app_idx 0x0000[0m
[0;32mI (17054) AppKey: ea 86 ad 82 84 6d 16 df af f8 90 e6 e3 6d 67 48 [0m
[0;32mI (24744) EXAMPLE: ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND[0m
[0;32mI (24744) EXAMPLE: elem_addr 0x0002, app_idx 0x0000, cid 0x02e5, mod_id 0x0000[0m
[0;32mI (24744) EXAMPLE_NVS: Store, key "vendor_client", length 5[0m
[0;32mI (24744) EXAMPLE_NVS: Store, data: 00 00 00 00 07 [0m
[0;31mE (29614) EXAMPLE: example_ble_mesh_send_vendor_message[0m
[0;31mE (29614) EXAMPLE: Failed to send vendor message err: 102[0m
[0;31mE (29614) EXAMPLE: Failed to send vendor message 0xc002e5[0m
[0;31mE (31854) EXAMPLE: example_ble_mesh_send_vendor_message[0m
[0;31mE (31854) EXAMPLE: Failed to send vendor message err: 102[0m
[0;31mE (31854) EXAMPLE: Failed to send vendor message 0xc002e5
```

## Other items if possible

[sdkconfig_elf.zip](https://github.com/espressif/esp-idf/files/4894964/sdkconfig_elf.zip)

Campou, 2020-07-22:
Hi @NOA-Vernon 

Sorry for the late reply.

1. Please use `esp_ble_mesh_client_model_init(&vnd_models[0])` to intialize the vendor client model after `esp_ble_mesh_init`.
2. Also I found that in the log `GATT_SendServiceChangeIndication can't send service change indication manually` happened, please choose the **Manual GATTS Service Change Mode** in the bluedroid menuconfig options.

Thanks.

Alvin1Zhang, 2020-09-01:
Thanks for reporting, feel free to reopen if any more updates. Thanks.

closed 2020-09-01 by Alvin1Zhang

busadhruv, 2023-10-02:
In this vendor model code i want publish subscribe message functionality because i use node as a low power and friend node and my data is also string ..

issue 
in example_ble_mesh_custom_model_cb this call back function not generate event for publish?
it only generate event for send message as it do before.