espressif/esp-idf #5560: Memory leak in bta_ag_sco_event(...)?

howroyd, 2020-07-09:
IDF release v4.1 in file `esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c` line [768](https://github.com/espressif/esp-idf/blob/84b51781c80740fda92784dafcfc96c13b0d8b66/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c#L768).

`break` and then `return` without freeing the memory at `p_buf`.

For some reason, `BTM_WriteScoData` is failing for me but this issue is that when it does, the heap is immediately corrupt because the memory at `p_buf` is not freed before `break`.

edwin-oetelaar, 2020-07-11:
Yes, I think you found a bug. 
Adding ```osi_free(p_buf);``` before the ```break;``` on lin 768 would fix it.

xiongweichao, 2020-09-10:
If `BTM_WriteScoData` is  failed, the `p_buf` memory will be freed before `BTM_WriteScoData` return.
IDF release v4.1 in file   `/esp-idf/components/bt/host/bluedroid/stack/btm/btm_sco.c` line 459.
```
457    if (status != BTM_SUCCESS) {
458        BTM_TRACE_WARNING ("stat %d", status);
459        osi_free(p_buf);
460    }
461   return (status);
```

Alvin1Zhang, 2020-10-29:
Thanks for reporting, feel free to reopen.

closed 2020-10-29 by Alvin1Zhang

Wth-Esp, 2020-10-29:
Hi
@xiongweichao  is right. The `p_buf` is freed in function `BTM_WriteScoData`.

And we have fixed an issue of data path recently and it may need several days to sync on Github.
Please refer to https://github.com/espressif/esp-idf/issues/4967. 

Thanks