espressif/esp-idf #5561: ESP-IDF SD SPI not working

muaiydit, 2020-07-09:
Hi,

I have seen this issue in some post but wasn't really helpful.
I use the SD card example in the ESP-IDF and I have enabled the SPI mode by uncommenting the #ifdef USE_SPI_MODE
however I am getting this errors:
E (412) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (432) example: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
Same hardware with same connections and pins define was used successfully with Arduino IDE. Any idea?

igrr, 2020-07-09:
Hi @muaiydit, which SPI clock frequency have you used in Arduino IDE? Have you tried reducing the clock frequency when running SD card example in IDF?

muaiydit, 2020-07-09:
Hi,
Thank you for your replay.
I haven't change the clock and I have used the lib as it.
What I noticed on the oscilloscope the Arduino IDE send the 80 clock on CS high then it start sending the commands. While, the IDF on starting sending commands without switching it to spi mode.

muaiydit, 2020-07-14:
Any idea?

igrr, 2020-07-14:
@muaiydit Arduino SD library by default uses 4MHz bus clock, while sdspi in ESP-IDF uses 20MHz bus clock. Invalid CRC errors are almost always related to signal integrity issues. You might have insufficiently strong pull-up resistors, too large capacitance (due to wire length) or ground bounce due to the return current.
If you have an oscilloscope, please try comparing the signal shape between the working and non-working case.

muaiydit, 2020-07-14:
Thank you very much for your suggestions. 
On oscilloscope the clock and the data are perfectly sharp on both Arduino code and ESP-IDF code. However, the clock speed on Arduino is 125K while it is a 400K on the IDF. Is that cause the issue? how to reduce the clock?

muaiydit, 2020-07-14:
I have been able to change the ESP-IDF clock to 125K and still not working

igrr, 2020-07-15:
Which version of ESP-IDF are you using? (please run `git describe --tags` in ESP-IDF directory)

Could you please post your code? `sdmmc_check_scr` function mentioned in the error message is specific to SD interface, it is not used when talking to the card over SPI.

muaiydit, 2020-07-15:
The ESP-IDF is v4.0.1
I used the code in examples\storage\sd_card\ only modification was uncoment this line to activate the SPI mode
#define USE_SPI_MODE

MrCurio, 2021-01-21:
Any updates on this issue? I am facing the same problems as @muaiydit 

I am using a SD card Module, which has a 5V pin and a AMS1117 regulator to step-down to 3.3V. I am powering the module through the VIN pin of my devkitC board. When using Arduino SD Example, the program runs as expected. However, when using the esp-idf example, it simply doesn't run and sometimes returns "Invalid CRC" , "Timeout"...etc.

I have already tried with recommended 10k pull-up resistors and nothings seems to make effect. Any recommendations? I have seen similar problems with SD cards around  the SPI driver. 

Is there any SD module that you have already tested with the esp-idf driver? Could you provide a link to the module itself?
My application desperately needs a record of logs and i'm afraid i'm stuck with this problem. Any ideas?

Thanks!

chegewara, 2021-01-21:
> Any updates on this issue? I am facing the same problems as @muaiydit
> 
> I am using a SD card Module, which has a 5V pin and a AMS1117 regulator to step-down to 3.3V. I am powering the module through the VIN pin of my devkitC board. When using Arduino SD Example, the program runs as expected. However, when using the esp-idf example, it simply doesn't run and sometimes returns "Invalid CRC" , "Timeout"...etc.
> 
> I have already tried with recommended 10k pull-up resistors and nothings seems to make effect. Any recommendations? I have seen similar problems with SD cards around the SPI driver.
> 
> Is there any SD module that you have already tested with the esp-idf driver @igrr ? Could you provide a link to the module itself?
> My application desperately needs a record of logs and i'm afraid i'm stuck with this problem. Any ideas?
> 
> Thanks!

I have the same issues, but i found solution. In my case i have exactly the same hardware setup and the same code on esp32 (S1) and esp32S2. S2 works fine, but spits error. The solution is to decrease SPI speed, in my case 10MHz works on S1, even if S2 works with 20MHz. What is strange S2 wont run faster even if i setup 26MHz or more.

MrCurio, 2021-01-21:
Hi @chegewara 
Which esp-idf are you using? I'm using v4.2 

Could you point out where speed must be changed? I am just trying to run the example provided on GitHub, but i haven't moved forward.

Thanks!

chegewara, 2021-01-21:
```
    sdmmc_card_t *sd_card;
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.flags = SDMMC_HOST_FLAG_SPI | SDMMC_HOST_FLAG_DEINIT_ARG | SDMMC_HOST_FLAG_1BIT;
    host.max_freq_khz = 10000; <--- THIS
```
I am also using sd card adapter with AMS1117. Arduino sd card code is setup to use 4MHz by default, i think, but in my case 10MHz is ok.

MrCurio, 2021-01-21:
Okay, i have just tried and i could finally make it work!

Many many thanks @chegewara !

pacucha42, 2022-02-21:
Thank you @chegewara for helping to solve the issue. Closing the bug report, please, reopen if needed

closed 2022-02-21 by pacucha42