espressif/esp-idf #5562: SNTP Crash

drony, 2020-07-09:
IDF 4.0.1 when i start sntp server after 3 or 4 seconds crash and restart esp and works fine second time boot

here is the code
` ESP_LOGI(tag, "Initializing SNTP");
  sntp_setoperatingmode(SNTP_OPMODE_POLL);
  sntp_setservername(0, "pool.ntp.org");
  sntp_init();`

`
Decoding stack results
0x400966cd: invoke_abort at C:/SysGCC/esp32/esp-idf/v4.0.1/components/esp32/panic.c line 155
0x40096a51: abort at C:/SysGCC/esp32/esp-idf/v4.0.1/components/esp32/panic.c line 172
0x401e07b3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401e07fa: std::terminate() at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x401e09d5: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc line 300
0x401e0a90: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc line 54
0x400e205d: loopp::core::MainLoop::handle_timers() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\ext/new_allocator.h line 99
0x400e39a9: loopp::core::MainLoop::run() at ../../../components/loopp/src/core/MainLoop.cpp line 341
0x400ddabe: Main::main_task() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr_base.h line 1285
0x40225991: std::_Function_handler   >::_M_invoke(std::_Any_data const&) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/invoke.h line 89
0x400e1369: std::function ::operator()() const at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/std_function.h line 260
0x400e3a0a: loopp::core::Task::run(void*) at ../../../components/loopp/src/core/Task.cpp line 35
0x4009928d: vPortTaskWrapper at C:/SysGCC/esp32/esp-idf/v4.0.1/components/freertos/port.c line 143`

da-vics, 2020-07-09:
Similar Issue

Alvin1Zhang, 2020-07-10:
Thanks for reporting and letting us know, we will look into.

KonstantinKondrashov, 2020-07-13:
Hi @drony!
I have run [the SNTP example](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/protocols/sntp) from IDF v4.0.1-296-gf73cd1d and I do not observe any issue with that. Could you give an example that I can run to reproduce your issue?
(It will be very useful if you fill the issue template).
Thanks.

da-vics, 2020-07-16:
@KonstantinKondrashov try turning off the internet connectivity of your wireless transceiver. that would reprodice the error, when the internet connectivity is weak before the timeout results the the bug

closed 2021-12-27 by Alvin1Zhang