espressif/esp-idf #5566: error failed to load RF calibration data (0x1101), falling back to full calibration

coffice12, 2020-07-10:
esp32 woom

code：
esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/

error：
I (812) system_api: read default base MAC address from EFUSE
E (812) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (822) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
I (992) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 2
E (1232) BT_OSI: config_new: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
W (1232) BT_BTC: btc_config_init unable to load config file; starting unconfigured.

E (1242) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1252) BT_OSI: config_save, err_code: 0x2

E (1382) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1382) BT_OSI: config_save, err_code: 0x2

E (1392) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1402) BT_OSI: config_save, err_code: 0x2

E (1402) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1412) BT_OSI: config_save, err_code: 0x2

E (1422) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1432) BT_OSI: config_save, err_code: 0x2

I (1452) BLE_COMP: create attribute table successfully, the number handle = 11

I (1472) BLE_COMP: (0) ***** advertising start successfully *****

WCCWCC, 2020-08-13:
Hi, @coffice12 

Does this problem still exist?

You can provide a link to example to reproduce this problem？

boarchuz, 2020-08-13:
You need to initialize NVS first. The log explains this many times.

Nurali125, 2020-09-23:
@boarchuz, are you stupid or what? I have the same problem and in the code that I used nvs_flash_init() was already called, but anyway this problem appeared.  You are telling like, for example, if I had my pc broken and called expert to fix it, but expert comes and tells me "Oh, you have a blue screen, your pc is broken, you need to fix it". Wtf, I called you to help me fix the problem, not to just tell me that I have a problem! The same thing you are telling here man

igrr, 2020-09-23:
@Nurali125 Please refrain from abusive comments on this issue tracker. If you have met a similar issue, but you do have an nvs_flash_init call included, please open a new issue and fill in the issue template, so that we have enough information to help you.

iSWORD, 2021-03-18:
Hi, I'm having the same issue trying to run `ble_compatibility_test` from the examples without making any modification to its code. I can see it's attempting to [init NVS](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_compatibility_test/main/ble_compatibility_test.c#L623) but I'm still getting stuck in a bootloop.

iSWORD, 2021-03-18:
Update: updating to `master` instead of `v4.2` fixed this issue. Thank you Espressif team & community ❤️

Alvin1Zhang, 2021-04-13:
Thanks for sharing the updates, feel free to reopen.

closed 2021-04-13 by Alvin1Zhang

Ashishel095, 2022-06-23:
I have updated the master "boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader" but I am still getting the same error.

pnt325, 2022-10-18:
I have the same issue. Cause open other nvs partition without close before call BLE/WIFI init.

Call `nvs_close()` before initialize BLE/WIFI