espressif/esp-idf #5567: Bt classic & esp now coex on real time demanding application

lyris85, 2020-07-10:
Hello, my application uses 20 esp32 devices connected together with espnow and 1 mobile application. Bt classic spp is used to exchange data between the application and the esp network. The esp connected to the mobile application route/forward in real time all incoming spp data to espnow such that each esp32 are getting updated as new bt data arrives. 

It works well, I can reach high speed transfer without any delay. However, once the bt transmission is done, the router device is not able to listen to the esp now network for  like 2 or 3 seconds. This is a problem because I want to acknowledge the process once bt transmission is done via espnow.

I know that the radio is shared. I don't expect to listen to esp now and bt spp at the same time. However, it should be possible to listen to espnow immediatly after a bt transmission is done. Is there any way I can configure this ?

Greetings

jack0c, 2020-12-08:
@lyris85 Sorry for the late reply. To say the truth, we do not have an ideal solution for this. ESP-NOW is designed for device without sleep and coexistence. But we can try to analyze it and give a solution. @blueMoodBHD please help.

lyris85, 2020-12-08:
We implemented everything using wifi instead of bluetooth and it worked out. However a spp solution would have been more elegent

blueMoods, 2020-12-09:
@lyris85  As we don't support coexistence between bluetooth and ESP-NOW, So ESP-NOW can work only when bt not work,such as no bt connection or in sniff mode. After  bt transmission is done, it will enter sniff mode after a while, Then ESP-NOW can works well.

Could you please provide the commit of esp-idf which you used? And about the SPP, ESP32 is master or slave?

jack0c, 2020-12-09:
@blueMoodBHD  `it will enter sniff mode after a while`, can we configure the `a while`?

lyris85, 2020-12-09:
I don't have access to the computer anymore. I used the latest release of june.

blueMoods, 2020-12-09:
We don't have any APIs to config the time to enter sniff. But you can test it with some changes in esp-idf, in `components/bt/host/bluedroid/common/include/common/bt_target.h`,    `BTA_FTS_OPS_IDLE_TO_SNIFF_DELAY_MS` and `BTA_FTC_OPS_IDLE_TO_SNIFF_DELAY_MS` means that how long to wait before activating sniff mode after entering the idle state， you can change them and test can it work or not.

jack0c, 2023-04-13:
@lyris85 Since esp-idf v4.3, we support power save when Wi-Fi disconnected as well as esp-now window. If you use esp-now window, the coexistence will work properly.

jack0c, 2023-07-19:
@lyris85 Are you still working on this? will close this issue, feel free to reopen it if necessary.

closed 2023-07-19 by jack0c