espressif/esp-idf #5568: a provisioning problem of ESP-BLE-MESH "fast_prov_client" and "fast_prov_server"

Beibixian, 2020-07-11:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.0.1
- Build System:         [idf.py]
- Compiler version: (crosstool -NG esp-2019r2) 8. 2. 0
- Operating System:     [Windows]
- Using an IDE?: [vs code]
- Power Supply:         [USB]

## Problem Description

During the test of project ESP-BLE-MESH **"fast_prov_client" and "fast_prov_server"**, I build and flash the "fast_prov_client" project into the first developing board and then build and flash the "fast_prov_server" project into the second and third developing board. 
The first board(Top provisioner) configures the second board successfully and the second board becomes the primary provisioner.
[image]
Everything goes correctly before **the second board prepares to configure the third board**. 
Debug Logs are provided below.

## Debug Logs of the second board

I (7513) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT
I (7513) EXAMPLE: provisioner_prov_link_open: bearer PB-ADV
I (7513) EXAMPLE: ESP_BLE_MESH_PROVISIONER_ADD_UNPROV_DEV_COMP_EVT, err_code: 0
**E (9133) BLE_MESH: bt_mesh_provisioner_check_is_addr_dup, Not a unicast address 0x0000
E (9133) BLE_MESH: send_prov_data, Assigned address 0x0006 is duplicated**
I (10643) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT
I (10643) EXAMPLE: provisioner_prov_link_close: bearer PB-ADV, reason 0x02
I (13513) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT
I (13513) EXAMPLE: provisioner_prov_link_open: bearer PB-ADV
E (15103) BLE_MESH: bt_mesh_provisioner_check_is_addr_dup, Not a unicast address 0x0000
E (15103) BLE_MESH: send_prov_data, Assigned address 0x0006 is duplicated
I (16553) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT
I (16553) EXAMPLE: provisioner_prov_link_close: bearer PB-ADV, reason 0x02
......(repeating the error)

Alvin1Zhang, 2020-07-13:
@Beibixian Thanks for the detailed report, we will look into.

Campou, 2020-07-13:
Hi @Beibixian 

Please try with the [release/v4.0 branch](https://github.com/espressif/esp-idf/tree/release/v4.0), this is a known issue, and I have fixed this on the release/v4.0 branch.

Thanks.

closed 2020-07-13 by Beibixian