espressif/esp-idf #5569: Its will be good to not pay for unused features

SiarheiVolkau, 2020-07-11:
I have found a sort of dead code issue.

My setup does not use BluFi but use GATTS.
In the .map file i encounter that BluFi code and data are inculded in my binary.

I propose to refactor the BLE code to compile in only used features.
Or add flag to enable/(disable?) BluFi explicitly in Kconfig.

I have collect all blufi related data size, just to show size of the problem:
`.data` section: 100 bytes
`COMMON` section: 52 bytes
`.rodata` section: 1692 bytes
`.literal` section: 184 bytes
`.text` section: 4198 bytes

i suspect there are more unused features which increases final binary by dead code and data in RAM.

Will be good if you review the code and refactor it properly.

BR,
Siarhei

WCCWCC, 2020-07-13:
Hi, @SiarheiVolkau 

You said that the problem does exist, we have to discuss it internally, some code is indeed not controlled by macros

jack0c, 2022-05-20:
It is fixed since idfv4.3 and backported to v4.2, v4.1, v4.0, v3.3

closed 2022-05-20 by jack0c