espressif/esp-idf #5570: sdio: Fails to link sdio_slave.elf on ESP32s2

htruong, 2020-07-12:
## Environment

- Development Kit:     ESP32-S2-SAOLA-M
- Kit version: v1.2
- Module or chip used:  ESP32-S2-WROVER
- IDF version (run ``git describe --tags`` to find it):
    v4.2-dev-2243-gcf056a7d0
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         Irrelevant

## Problem Description

I tried to compile the sdio example for the ESP32s2, and encountered the error:

```bash
cp -r $IDF_PATH/examples/peripherals/sdio/ .
cd sdio/slave
idf.py set-target esp32s2
idf.py build
```

### Expected Behavior

SDIO example gets build.

### Actual Behavior

The following message appeared:

```
[103/103] Generating binary image from built executable
esptool.py v3.0-dev
Generated /home/wasabi/develop/sdio/slave/build/bootloader/bootloader.bin
[926/927] Linking CXX executable sdio_slave.elf
FAILED: sdio_slave.elf
// ldcommand...
/home/blah/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(app_main.c.obj):(.literal.event_cb+0x4): undefined reference to `sdio_slave_read_reg'
/home/blah/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(app_main.c.obj):(.literal.event_cb+0x8): undefined reference to `sdio_slave_write_reg'
/home/blah/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(app_main.c.obj):(.literal.task_hostint+0x0): undefined reference to `sdio_slave_send_host_int'
```

Alvin1Zhang, 2020-07-13:
Thanks for reporting, we will look into.

renzbagaporo, 2020-07-13:
ESP32-S2 does not have SDIO peripheral.

htruong, 2020-07-13:
Thanks, that explains the issue.

closed 2020-07-13 by htruong