espressif/esp-idf #5571: gpio_uninstall_isr_service() Interrupt wdt timeout

drony, 2020-07-12:
when i call **gpio_uninstall_isr_service()** 

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

`
Decoding 18 results
0x4009beab: vListInsert at C:/SysGCC/esp32/esp-idf/v4.0.1/components/freertos/list.c line 188
0x4009b22d: vTaskPlaceOnEventList at C:/SysGCC/esp32/esp-idf/v4.0.1/components/freertos/tasks.c line 3508
0x40099dff: xQueueGenericReceive at C:/SysGCC/esp32/esp-idf/v4.0.1/components/freertos/queue.c line 2044
0x400d416c: esp_ipc_call_and_wait at C:/SysGCC/esp32/esp-idf/v4.0.1/components/esp_common/src/ipc.c line 116
0x400d4232: esp_ipc_call_blocking at C:/SysGCC/esp32/esp-idf/v4.0.1/components/esp_common/src/ipc.c line 128
0x400d327a: esp_intr_free at C:/SysGCC/esp32/esp-idf/v4.0.1/components/esp32/intr_alloc.c line 783
0x401585b2: gpio_uninstall_isr_service at C:/SysGCC/esp32/esp-idf/v4.0.1/components/driver/gpio.c line 430
0x400efc88: bbcore::drivers::GPIODriver::stop() at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/include/bbcore/utils/json.hpp line 17886
0x400e6bd1: Main::on_provisioning(std::__cxx11::basic_string  , std::allocator  > const&) at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/include/bbcore/utils/json.hpp line 20862
0x400e6f09: Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}::operator()(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >) const at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/include/bbcore/utils/json.hpp line 20862
:  (inlined by) void std::__invoke_impl   , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&, std::__cxx11::basic_string  , std::allocator  >&, Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&>(std::__invoke_other, Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&, std::__cxx11::basic_string  , std::allocator  >&, Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/invoke.h line 60
:  (inlined by) std::__invoke_result   , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&, std::__cxx11::basic_string  , std::allocator  >&, Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&>::type std::__invoke   , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&, std::__cxx11::basic_string  , std::allocator  >&, Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&>(std::__invoke_result&&, (Main::on_mqtt_connected(bool)::{lambda(std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1}&)...) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/invoke.h line 95
:  (inlined by) void std::_Bind   , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1} (std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)>::__call (std::tuple&&, std::_Index_tuple0u, 1u>) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0/functional line 400
:  (inlined by) void std::_Bind   , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1} (std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)>::operator(), void>() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0/functional line 484
:  (inlined by) std::_Function_handler    , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)#1} (std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >)> >::_M_invoke(std::_Any_data const&) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/std_function.h line 297
0x400e7825: std::function ::operator()() const at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/std_function.h line 687 (discriminator 1)
0x400e956f: bbcore::core::MainLoop::handle_queue() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/hashtable.h line 1440
0x400e9e6b: bbcore::core::MainLoop::run() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/hashtable.h line 1440
0x400e1a9a: Main::main_task() at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/include/bbcore/utils/json.hpp line 20862
0x402143a5: void std::__invoke_impl (std::__invoke_memfun_deref, void (Main::*&)(), Main*&) at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/include/bbcore/utils/json.hpp line 20862
:  (inlined by) std::__invoke_result ::type std::__invoke (void (Main::*&)(), Main*&) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/invoke.h line 95
:  (inlined by) void std::_Bind ::__call (std::tuple&&, std::_Index_tuple0u>) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0/functional line 400
:  (inlined by) void std::_Bind ::operator(), void>() at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0/functional line 484
:  (inlined by) std::_Function_handler   >::_M_invoke(std::_Any_data const&) at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/std_function.h line 297
0x400e7825: std::function ::operator()() const at c:\sysgcc\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/std_function.h line 687 (discriminator 1)
0x400e9ede: bbcore::core::Task::run(void*) at C:\Users\drony\source\repos\PR14.EMB.IR.Next\build\VisualGDB\Debug/../../../components/bbcore/src/core/Task.cpp line 35
0x4009928d: vPortTaskWrapper at C:/SysGCC/esp32/esp-idf/v4.0.1/components/freertos/port.c line 403
`

Alvin1Zhang, 2020-07-13:
@drony Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

closed 2021-12-09 by espressif-bot

AxelLin, 2022-02-20:
@Alvin1Zhang 
v4.3 does not include this fix.