espressif/esp-idf #5572: Update ESP32-Ethernet-Kit docs for v1.2

yarrick, 2020-07-12:
**Is your feature request related to a problem? Please describe.**

I got an ESP32-Ethernet-Kit v1.2 board and it is not mentioned in the documentation.

**Describe the solution you'd like**

docs/en/hw-reference/esp32/get-started-ethernet-kit.rst describing the v1.2 board and changes from v1.1.

krzychb, 2020-07-22:
Hi @yarrick, 

I am sorry for missing documentation for this board. 

Here is the list of differences between V1.2 and V1.1:

1. Correct the silkscreen marking at the function switch. Below is respective note from V1.1 documentation
   [image]

2. Value of C1, C2, C42, and C43 is changed to 20 pF.
3. Replace ESP32-WROVER-B with ESP32-WROVER-E.

Updated schematic of ESP32-Ethernet-Kit_A_V1.2, is attached below.
[SCH_ESP32-Ethernet-Kit_A_V1.2_20200528.pdf](https://github.com/espressif/esp-idf/files/4959010/SCH_ESP32-Ethernet-Kit_A_V1.2_20200528.pdf)

New documentation will be published soon.

krzychb, 2020-08-03:
Resolved in a6c18343274e349fc5b18e8b73f439b1ad3adad3. 
Thanks @yarrick !

closed 2020-08-03 by krzychb