espressif/esp-idf #5573: RMT driver (TX) does not work properly if the inverted GPIO output is used

joy4eg, 2020-07-12:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:  [any ESP32]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-2243-gcf056a7d0
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

**TL;DR:** RMT driver (TX) does not work properly if the inverted GPIO output is used.

Details: I'm trying to write an alternate firmware for Xiaomi Universal IR controller (MJYKQ01CM). This device is based on a single core ESP32 chip with BLE and WiFi, 4MB of SPI flash.
It has IR receiver that connected to `GPIO_NUM_27`, and seven IR transmitters, they connected to `GPIO_NUM_26`.

Some pictures - https://imgur.com/a/qWfD7xt

The thing is that `GPIO_NUM_26` connected as the **inverted** output. But RMT drivers doesn't have an API to set that, it contains a hardcoded function call:  https://github.com/espressif/esp-idf/blob/master/components/driver/rmt.c#L490
```c
.. rmt_set_pin() ...
   if (mode == RMT_MODE_TX) {
...
      esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 0, 0);                                                                                                                                             
...
   }
```
As result, the transmitter does not work properly (and that's how I burned half of the diodes on that board because they remains in the ON state after transmitting :D )

To solve my issue I changed that line to:
```c
 esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 1, 0);   
```
After that change all peripherals works as expected.

### Expected Behavior
An API to tell RMT driver that GPIO is inverted or not.

### Actual Behavior
There is no such API.

Alvin1Zhang, 2020-07-13:
Thanks for reporting, we will look into.

wilhelmw201, 2020-07-13:
Hi,

sorry about your board :( Would enabling the idle output of RMT and setting it to high (both are in the rmt_tx_config_t) give you the correct output?

joy4eg, 2020-07-13:
Hey @wilhelmw201 

My tx init function looks like:
```c
esp_err_t ir_init_tx(void)
{
	rmt_config_t config = RMT_DEFAULT_CONFIG_TX(DEVICE_GPIO_IR_TX, DEVICE_RMT_CHANNEL_TX);
	esp_err_t err;

	ESP_LOGI(TAG, "TX: gpio #%d, channel %d", DEVICE_GPIO_IR_TX, DEVICE_RMT_CHANNEL_TX);

	config.clk_div = 255;
	config.tx_config.loop_en = false;
	config.tx_config.carrier_duty_percent = 50;
	config.tx_config.carrier_freq_hz = 38000;
	config.tx_config.carrier_level = 1;
	config.tx_config.carrier_en = true;
	config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
	config.tx_config.idle_output_en = true;

	err = rmt_config(&config);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "TX: rmt_config failed: %s", esp_err_to_name(err));
		return err;
	}

	err = rmt_driver_install(DEVICE_RMT_CHANNEL_TX, 1000, 0);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "TX: rmt_driver_install failed: %s", esp_err_to_name(err));
		return err;
	}
	return ESP_OK;
}
```

I changed `config.tx_config.idle_level` to `RMT_IDLE_LEVEL_HIGH`, and reverted the GPIO change (`esp_rom_gpio_connect_out_signal` was called with `0, 0`). 
Result: transmitter is always ON, and it cannot produce correct IR commands (my LED lamp ignores all of them).

wilhelmw201, 2020-07-13:
Hi,

can you elaborate on this sentence: "The thing is that GPIO_NUM_26 connected as the inverted output."? I assume you mean that you have inverted the GPIO prior to calling the listed ir_init_tx, but this was overwritten by the RMT init function? 

(If yes, it is not necessary to inverse the output, you can set the RMT idle level to the idle level you want and inverse your signal data.)

joy4eg, 2020-07-13:
What I tried:
- `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 1, 0)` and `RMT_IDLE_LEVEL_HIGH`: always on, doesn't work
- `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 1, 0)` and `RMT_IDLE_LEVEL_LOW`: works as expected
- `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 0, 0)` and `RMT_IDLE_LEVEL_HIGH`: always on, doesn't work
- `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 0, 0)` and `RMT_IDLE_LEVEL_LOW`: always on, doesn't work

So it is not just the inverted signal itself, LED should stay in the **OFF** state if there is no transmission at the current moment.

wilhelmw201, 2020-07-13:
I see, did you enable the carrier? Then in this case this is problematic.

We'll put an inverse function in the GPIO, but for now you may have to call esp_rom_gpio_connect_out_signal manually :(

joy4eg, 2020-07-13:
>did you enable the carrier?

I did :)

>may have to call esp_rom_gpio_connect_out_signal manually
That's what I'm doing now. Hope it will fixed by some flag/argument in RMT driver API.

costaud, 2020-08-04:
> * `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 1, 0)` and `RMT_IDLE_LEVEL_LOW`: works as expected
> * `esp_rom_gpio_connect_out_signal(gpio_num, RMT_SIG_OUT0_IDX + channel, 0, 0)` and `RMT_IDLE_LEVEL_HIGH`: always on, doesn't work

What seems strange to me is that this two cases should have the same output if carrier mode is on.

suda-morris, 2022-05-16:
Invertion of GPIO output signal was added :
1. in the legacy driver, there's a flag `RMT_CHANNEL_FLAGS_INVERT_SIG` to control this
2. in the new driver, there's a `invert_out` flag in `rmt_tx_channel_config_t` configuration structure.

See [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv4N23rmt_tx_channel_config_t10invert_outE) for how to use it.

closed 2022-05-16 by suda-morris