espressif/esp-idf #5574: ESP-IDF mDNS functionality does not include query ID in response

george-hawkins, 2020-07-12:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROVER
- IDF version: v4.0.1
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

When an mDNS query is sent it includes a transaction ID, this ID should be included in the response. The ESP-IDF mDNS functionality does not do this, it always just includes an ID of 0 in the response.

Many mDNS libraries ignore this issue. As a result, you can see different behavior depending on which tool you use. E.g. `avahi-resolve` ignores this issue:

```
$ avahi-resolve --name espressif.local
espressif.local	192.168.0.248
```

Whereas `dig` treats this as an error and will not resolve the given name:

```
$ dig +short -p 5353 @224.0.0.251 espressif.local
;; Warning: ID mismatch: expected ID 39315, got 0
;; Warning: ID mismatch: expected ID 39315, got 0
;; Warning: ID mismatch: expected ID 39315, got 0
;; connection timed out; no servers could be reached
```

I've confirmed what's going on using Wireshark. Here's a screenshot of the relevant packets. First, I queried a host that I knew would correctly respond with the given transaction ID. You can see the query and the response for `daedalus.local` include the ID 0x00007f63. Then I queried an ESP32 devise, called `ding-5cd80b3.local`. Here you can see that the query includes the ID 0x00009993 but the responses (`dig` retries twice) always include the ID 0x00000000.

[image]

Note: the mDNS hostname is being set using the `mdns_hostname_set` function as described in the ESP-IDF [mDNS documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mdns.html).

## Debug Logs

[mdns.pcapng.gz](https://github.com/espressif/esp-idf/files/4908651/mdns.pcapng.gz) - this is the pcap file corresponding to the image included above.

george-hawkins, 2020-07-20:
You can see here in [`esp-idf/.../mdns.c:4852`](https://github.com/espressif/esp-idf/blob/ab8cab1/components/mdns/mdns.c#L4852) that the ID is correctly parsed from the incoming message:

```c
header.id = _mdns_read_u16(data, MDNS_HEAD_ID_OFFSET);
```

However, it's ignored after this point and not included in the response.

Interestingly, [Nicko van Someren](https://github.com/nickovs) has created an mDNS implementation that runs in MicroPython (that runs on-top of the ESP-IDF) that parses out this ID and, unlike the current ESP-IDF C implementation, also correctly includes it in the response.

See [`slimDNS/.../slimDNS.py`](https://github.com/nickovs/slimDNS/blob/638c461/slimDNS.py), there you can see it first parse out the ID (into `pkt_id`) on line [244](https://github.com/nickovs/slimDNS/blob/638c461/slimDNS.py#L244):

```python
(pkt_id, flags, qst_count, ans_count, _, _) = unpack_from("!HHHHHH", buf, 0)
```

And then on line [280](https://github.com/nickovs/slimDNS/blob/638c461/slimDNS.py#L280), it includes this ID in the response:

```python
pack_into("!HHHHHH", buf, 0,
          pkt_id, _FLAGS_QR_RESPONSE | _FLAGS_AA,
          0, len(matches), 0, 0)
```

It's this bit, where the ID is included in the response, that's missing in the current ESP-IDF implementation. And it's this bit that makes the difference between `dig` and other name resolution tools accepting the response or rejecting it.

`header.id` needs to be passed on down, and included in the response, and not ignored as is done at the moment.

Alvin1Zhang, 2020-08-13:
Thanks for reporting and sorry for slow turnaround. We have a fix under internal reviewing, and the issue will be closed automatically once the fix is available on GitHub. Thanks.

closed 2020-09-16 by espressif-bot