espressif/esp-idf #5575: twai_transmit don't work

nopnop2002, 2020-07-12:
twai_receive() work fine.
But An error occurs in twai_transmit().

My Environment:

	freeRTOS version:V8.2.0
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.2-dev-2243-gcf056a7d0
	ets_get_cpu_frequency:160

CAN Transceiver:
SN65HVD230

My test code:

	#include <stdio.h>
	#include <stdlib.h>
	#include <string.h>
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "freertos/queue.h"
	#include "freertos/semphr.h"
	#include "esp_err.h"
	#include "esp_log.h"
	#include "driver/twai.h" // Update from V4.2

	/* --------------------- Definitions and static variables ------------------ */
	//Example Configuration
	#define RX_TASK_PRIO    9
	#define TX_GPIO_NUM             21
	#define RX_GPIO_NUM             22
	#define TAG                             "CANDUMP"

	static const twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

	static const twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();

	static const twai_general_config_t g_config =
	        TWAI_GENERAL_CONFIG_DEFAULT(TX_GPIO_NUM, RX_GPIO_NUM, TWAI_MODE_NORMAL);

	void app_main()
	{
	        //Install and start TWAI driver
	        ESP_ERROR_CHECK(twai_driver_install(&g_config, &t_config, &f_config));
	        ESP_LOGI(TAG, "Driver installed");
	        ESP_ERROR_CHECK(twai_start());
	        ESP_LOGI(TAG, "Driver started");

	        //xTaskCreate(twai_task, "TWAI", 4096, NULL, RX_TASK_PRIO, NULL);

	        twai_message_t rx_msg;
	        twai_message_t tx_msg;
	        int counter = 0;

	        while (1) {
	                //twai_message_t rx_msg;
	                //twai_receive(&rx_msg, portMAX_DELAY);
	                esp_err_t rtn = twai_receive(&rx_msg, 0);
	                if (rtn == ESP_OK) {
	                        ESP_LOGI(pcTaskGetTaskName(0),"twai_receive identifier=0x%x flags=0x%x-%x-%x data_length_code=%d",
	                                rx_msg.identifier, rx_msg.flags, rx_msg.extd, rx_msg.rtr, rx_msg.data_length_code);

	                } else {
	                        counter++;
	                        if (counter == 100) {
	                                tx_msg.identifier = 0x123;
	                                tx_msg.extd = 0;
	                                tx_msg.rtr = 0;
	                                tx_msg.ss = 0;
	                                tx_msg.self = 0;
	                                tx_msg.dlc_non_comp = 0;
	                                tx_msg.data_length_code = 1;
	                                tx_msg.data[0] = 0x10;
	                                ESP_LOGI(pcTaskGetTaskName(0),"twai_transmit .......");
	                                esp_err_t ret = twai_transmit(&tx_msg, pdMS_TO_TICKS(1000));
	                                if (ret == ESP_OK) {
	                                        ESP_LOGI(pcTaskGetTaskName(0),"twai_transmit success");
	                                } else {
	                                        ESP_LOGE(pcTaskGetTaskName(0), "twai_transmit Fail %s", esp_err_to_name(ret));
	                                }
	                                counter = 0;
	                        }
	                }
	                vTaskDelay(10);
	        }

	}

Test result:

	I (0) cpu_start: Starting scheduler on APP CPU.
	I (310) CANDUMP: Driver installed
	I (310) CANDUMP: Driver started
	I (1120) main: twai_receive identifier=0x101 flags=0x0-0-0 data_length_code=3
	I (2120) main: twai_receive identifier=0x10000001 flags=0x1-1-0 data_length_code=4
	I (3120) main: twai_receive identifier=0x102 flags=0x0-0-0 data_length_code=4
	I (4120) main: twai_receive identifier=0x10000002 flags=0x1-1-0 data_length_code=5
	I (5120) main: twai_receive identifier=0x103 flags=0x0-0-0 data_length_code=5
	I (6120) main: twai_receive identifier=0x10000003 flags=0x1-1-0 data_length_code=6
	I (7120) main: twai_receive identifier=0x104 flags=0x0-0-0 data_length_code=6
	I (8120) main: twai_receive identifier=0x10000004 flags=0x1-1-0 data_length_code=7
	I (9120) main: twai_receive identifier=0x105 flags=0x0-0-0 data_length_code=7
	I (10120) main: twai_receive identifier=0x10000005 flags=0x1-1-0 data_length_code=8
	I (11120) main: twai_receive identifier=0x106 flags=0x0-0-0 data_length_code=8
	I (11320) main: twai_transmit .......
	I (11320) main: twai_transmit success
	I (12120) main: twai_receive identifier=0x10000006 flags=0x3-1-1 data_length_code=0
	I (21420) main: twai_transmit .......
	E (21420) main: twai_transmit Fail ESP_ERR_INVALID_STATE
	I (31420) main: twai_transmit .......
	E (31420) main: twai_transmit Fail ESP_ERR_INVALID_STATE

Dazza0, 2020-07-14:
@nopnop2002 `twai_transmit()` will return `ESP_ERR_INVALID_STATE` if the peripheral has entered the Bus Off state. Could you call `twai_get_status_info()` to check the state of the peripheral?

nopnop2002, 2020-07-14:
Thank you for response.
I call twai_get_status_info() to check the state of the peripheral.
Although twai_transmit is successful in 11320, Frame is not output to CAN-BUS.

	I (0) cpu_start: Starting scheduler on APP CPU.
	I (310) CANDUMP: Driver installed
	I (310) CANDUMP: Driver started
	I (520) main: twai_receive identifier=0x10e flags=0x0-0-0 data_length_code=2
	I (1520) main: twai_receive identifier=0x1000000e flags=0x1-1-0 data_length_code=3
	I (2520) main: twai_receive identifier=0x10f flags=0x0-0-0 data_length_code=3
	I (3520) main: twai_receive identifier=0x1000000f flags=0x1-1-0 data_length_code=4
	I (4520) main: twai_receive identifier=0x100 flags=0x0-0-0 data_length_code=4
	I (5520) main: twai_receive identifier=0x10000000 flags=0x1-1-0 data_length_code=5
	I (6520) main: twai_receive identifier=0x101 flags=0x0-0-0 data_length_code=5
	I (7520) main: twai_receive identifier=0x10000001 flags=0x1-1-0 data_length_code=6
	I (8520) main: twai_receive identifier=0x102 flags=0x0-0-0 data_length_code=6
	I (9520) main: twai_receive identifier=0x10000002 flags=0x1-1-0 data_length_code=7
	I (10520) main: twai_receive identifier=0x103 flags=0x0-0-0 data_length_code=7
	I (11320) main: status_info.state=1
	I (11320) main: twai_transmit .......
	I (11320) main: twai_transmit success
	I (21320) main: status_info.state=2
	I (21320) main: twai_transmit .......
	E (21320) main: twai_transmit Fail ESP_ERR_INVALID_STATE
	I (31320) main: status_info.state=2
	I (31320) main: twai_transmit .......
	E (31320) main: twai_transmit Fail ESP_ERR_INVALID_STATE

nopnop2002, 2020-07-15:
I changed the terminating resistance from 120 ohms to 150 ohms.

It worked correctly.

[image]

closed 2020-07-15 by nopnop2002