espressif/esp-idf #5576: hello_world example for esp32-s2 build results in fail referencing bootloader.elf

roblatour, 2020-07-12:
## Environment

- Development Kit:        ESP32-S2
- Kit version                   v1
- Module or chip used:  ESP32-S2-Saola-1
- IDF version                   v4.2-dev-2243-gcf056a7d0
- Build System:               idf.py
- Compiler version         xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0  
- Operating System:      Windows 10
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?:            Yes - ESP-IDE
- Power Supply:            USB

## Problem Description

following the instruction here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#get-started-get-prerequisites

Got to Step 8
ran the command
idf.py build

got the following error message:

[102/103] Linking C executable bootloader.elf
FAILED: bootloader.elf
cmd.exe /C "cd . && C:\esp\tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe -mlongcalls   CMakeFiles/bootloader.elf.dir/project_elf_src.c.obj  -o bootloader.elf  esp-idf/soc/libsoc.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=C:/Users/Rob Latour/Documents/esp/hello_world/build/bootloader/bootloader.map  -fno-rtti  -fno-lto  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/soc/libsoc.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/soc/soc/esp32s2/libsoc_esp32s2.a  esp-idf/soc/libsoc.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/soc/soc/esp32s2/libsoc_esp32s2.a  -L C:/esp/esp-idf/components/esp32s2/ld  -T esp32s2.peripherals.ld  -Wl,--gc-sections  -L C:/esp/esp-idf/components/esp_rom/esp32s2/ld  -T esp32s2.rom.api.ld  -T esp32s2.rom.ld  -T esp32s2.rom.newlib-funcs.ld  -T esp32s2.rom.libgcc.ld  -T esp32s2.rom.spiflash.ld  -L C:/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s2  -T bootloader.ld  -T bootloader.rom.ld && cd ."
xtensa-esp32s2-elf-gcc.exe: error: Latour/Documents/esp/hello_world/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[589/925] Linking CXX static library esp-idf\tcpip_adapter\libtcpip_adapter.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

### Expected Behavior

build should work 

### Actual Behavior

build fails

### Steps to reproduce

follow instructions at
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#get-started-get-prerequisites

### Code to reproduce this issue
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU cores, WiFi%s%s, ",
            CONFIG_IDF_TARGET,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Free heap: %d\n", esp_get_free_heap_size());

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

## Debug Logs

## Other items if possible
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/4909892/CMakeLists.txt)

renzbagaporo, 2020-07-13:
Hi, can you make sure that your copy of `esp-idf` is clean? You can use git to check if there are modifications.

Also, you state that the board you are targetting ESP32-S2, so you make sure you are not forgetting `set-target`.

roblatour, 2020-07-13:
not sure how to do that - but I just installed it yesterday following these instructions
(https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#get-started-get-prerequisites

chegewara, 2020-07-13:
It is possible you did try to build with other esp-idf version on location. Try this:
`idf.py fullclean` or just delete build folder and try again.

roblatour, 2020-07-14:
idf.py fullclean
idf.py fullclean build
or deleting the build directory and its contents and running
idf.py buld
all end up with the same problem as reported above.

However, I notice in the error message it says
Latour/Documents/esp/hello_world/build/bootloader/bootloader.map: No such file or directory

It should say, I would think
C:/Users/Rob Latour/Documents/esp/hello_world/build/bootloader/bootlader.map

and as such I'm wondering if the space in my account name, Rob Latour, is causing a problem.

Also, of note, there is no file called 
C:/Users/Rob Latour/Documents/esp/hello_world/build/bootloader/bootlader.map

on my system the time of the error.

projectgus, 2020-07-15:
Hi @roblatour,

> and as such I'm wondering if the space in my account name, Rob Latour, is causing a problem.

You are correct. At the moment all paths used in the ESP-IDF build system (path containing ESP-IDF, path containing the project, path containing any other IDF components in the build) must not contain spaces. We're working on fixing this, or at minimum producing a clearer error to avoid the hard-to-debug errors which this currently produces.

Sorry for the inconvenience.

Will keep this issue open until we have either fixed this or fixed the error which is produced.

GeorgeFlorian, 2021-08-06:
`You are correct. At the moment all paths used in the ESP-IDF build system (path containing ESP-IDF, path containing the project, path containing any other IDF components in the build) must not contain spaces.`

Is this mentioned somewhere ?
Because I just encountered this and I don't remember reading it in no tutorial.
Thank you.

igrr, 2021-08-06:
It's mentioned in the getting started section, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-1-install-prerequisites (regarding spaces in ESP-IDF path) and https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-5-start-a-project (regarding spaces in project path).

zikalino, 2023-06-01:
This issue seems to be already resolved. I have verified that it's possible to build with spaces in a path.
Closing the issue - feel free to create new tickets for any specific corner cases if they occur.

closed 2023-06-01 by zikalino

Gisbert12843, 2024-01-27:
This Error has not been resolved as of: 5.1.2

cortices, 2024-05-13:
This issue still isn't resolved; it should not be closed. @zikalino what led you to believe it is resolved?
Today, four years later, I received from ESP-IDF 

    Error: Output directory contains spaces: `/home/xx/y/z/a b c/d/e/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-0583900747d45821/out`.

@projectgus as you said:
> Will keep this issue open until we have either fixed this or fixed the error which is produced.

igrr, 2024-05-13:
Currently we do have some test coverage for building with paths with spaces (in [tools/test_build_system/test_spaces.py](https://github.com/espressif/esp-idf/blob/master/tools/test_build_system/test_spaces.py). The issue was closed because these tests got added and the issues found by these tests got resolved. There might still be issues we haven't found — we would appreciate issue reports! If anyone stumbles upon an issue with paths-with-spaces in ESP-IDF v5.2 or later, please do open a new issue, we will try to solve it.