espressif/esp-idf #5581: loss "crypto/aes.h"

snailwft, 2020-07-13:
Hi, Why does compiling the application in ESP-IDF-v4-0-1 cause this error?I find the idf compents is under/components/wpa_supplicant/SRC/crypto/aes. H of this file!

/home/share/esp32/esp_light/InfoWaterFall/main/middle/include/divoom.h:141:10: fatal error: crypto/aes.h: No such file or directory
 #include "crypto/aes.h"

Alvin1Zhang, 2020-07-14:
Thanks for reporting @snailwft . Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

snailwft, 2020-07-14:
Hi, I have a list of things that you can look at.
文档：2020.7.14 基于esp-idf-v4.0.1编译应用...
链接：http://note.youdao.com/noteshare?id=93fa8a58957dde5e2e6d7cc4f63b1a87&sub=F2ADC47A798A4EF3A3533A7451261246

snailwft, 2020-07-14:
这个问题我已经解决了，谢谢您的支持。

closed 2020-07-14 by snailwft