espressif/esp-idf #5582: esp-idf/examples/provisioning/legacy/ble_prov/

coffice12, 2020-07-13:
cafe@CafedeMac-mini esp_prov % python esp_prov.py --transport ble --service_name PROV_90DB50 --sec_ver 1 --pop abcd1234 --ssid SE-Wifi --passphrase cafe888888888
BLE client is running in console mode
	This could be due to your platform not being supported or dependencies not being met
	Please ensure all pre-requisites are met to run the full fledged client
BLECLI >> Please connect to BLE device `PROV_90DB50` manually using your tool of choice
BLECLI >> Was the device connected successfully? [y/n] y
BLECLI >> List available attributes of the connected device
BLECLI >> Is the service UUID '021a9004-0382-4aea-bff4-6b3f1c5adfb4' listed among available attributes? [y/n] y
BLECLI >> Is the characteristic UUID '021a9000-0382-4aea-bff4-6b3f1c5adfb4' listed among available attributes? [y/n] y
BLECLI >> Is the characteristic UUID '021a9000-0382-4aea-bff4-6b3f1c5adfb4' listed among available attributes? [y/n] y
BLECLI >> Is the characteristic UUID '021a9000-0382-4aea-bff4-6b3f1c5adfb4' listed among available attributes? [y/n] y

==== Starting Session ====
BLECLI >> Write following data to characteristic with UUID '021a9000-0382-4aea-bff4-6b3f1c5adfb4' :
	>> 10015a25a201220a2038171451f73b9970d0ba02391e7d99ff31e93f29ae3e622f602fc74851dc8f01
BLECLI >> Enter data read from characteristic (in hex) :
	<< 
Incorrect sec scheme
cafe@CafedeMac-mini esp_prov % 
cafe@CafedeMac-mini esp_prov % 
cafe@CafedeMac-mini esp_prov %

Alvin1Zhang, 2020-12-18:
Thanks for reporting, feel free to reopen with more details, thanks.

closed 2020-12-18 by Alvin1Zhang

ankayca, 2024-07-01:
Hi,
I have the same issue reported here https://www.esp32.com/viewtopic.php?f=13&t=40630&p=134184#p134184.