espressif/esp-idf #5583: Problems sending - MQTT - WSS publish request

FocosiGit, 2020-07-13:
Describe your problem:
Problems using MQTT over WSS:

I am using the components: wifi & mqtt via websockets.
After a few successful messages, the system receives the following error messages:

3. Include logs:

E (34851) TRANS_SSL: ssl_poll_write select error 104, errno = Connection reset by peer, fd = 55
W (34851) TRANS_SSL: Poll timeout or error, errno=Success, fd=55, timeout_ms=10000
E (34861) TRANSPORT_WS: Error write header
I (34861) MQTTWSS_EXAMPLE: MQTT_EVENT_ERROR
E (34871) MQTT_CLIENT: Error write data or timeout, written len = -1, errno=0

4. Problem information to locate and fix the problem:

My system:
ESP32 devboard.
The dev environment is esp-idf 3.3.2 latest version.

Board Boot information:

I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v3.3.2 2nd stage bootloader
I (43) boot: compile time 12:04:07
I (43) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed : 40MHz
I (51) boot: SPI Mode : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label Usage Type ST Offset Length
I (70) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (78) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (85) boot: 2 factory factory app 00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1fa44 (129604) map
I (158) esp_image: segment 1: paddr=0x0002fa6c vaddr=0x3ffb0000 size=0x005a4 ( 1444) load
I (159) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x80c18 (527384) map
I (350) esp_image: segment 3: paddr=0x000b0c38 vaddr=0x3ffb05a4 size=0x03cc0 ( 15552) load
I (357) esp_image: segment 4: paddr=0x000b4900 vaddr=0x40080000 size=0x00400 ( 1024) load
I (357) esp_image: segment 5: paddr=0x000b4d08 vaddr=0x40080400 size=0x13c7c ( 81020) load
I (412) boot: Loaded app from partition at offset 0x10000
I (412) boot: Disabling RNG early entropy source...
I (412) cpu_start: Pro cpu up.
I (416) cpu_start: Application information:
I (421) cpu_start: Project name: mqtt_websocket_secure
I (427) cpu_start: App version: 1
I (431) cpu_start: Compile time: Jul 13 2020 12:04:16
I (437) cpu_start: ELF file SHA256: ee02cc3725eb7aa6...
I (443) cpu_start: ESP-IDF: v3.3.2
I (448) cpu_start: Starting app cpu, entry point is 0x40081458
I (0) cpu_start: App cpu up.
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (471) heap_init: At 3FFBA4E8 len 00025B18 (150 KiB): DRAM
I (478) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (484) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (490) heap_init: At 4009407C len 0000BF84 (47 KiB): IRAM
I (497) cpu_start: Pro cpu start user code
I (179) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

The program that I am testing is:

#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>
[... 207 lines cut ...]
sent publish successful, msg_id=46910
MQTT_EVENT_PUBLISHED, msg_id=46910
sent publish successful, msg_id=53454
E (21571) TRANSPORT_WS: Error read data
E (21581) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
E (21581) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
sent publish successful, msg_id=10775
MQTT_EVENT_DISCONNECTED
I (21591) MQTT_CLIENT: Client force reconnect requested
I (21601) MQTTWSS_EXAMPLE: Other event id:7
I (27691) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
MQTT_EVENT_CONNECTED - sent subscribe successful, msg_id=47903
sent subscribe successful, msg_id=10841Waiting Mqtt
MQTT_EVENT_SUBSCRIBED, msg_id=47903
MQTT_EVENT_SUBSCRIBED, msg_id=10841
sent publish successful, msg_id=25799
E (34851) TRANS_SSL: ssl_poll_write select error 104, errno = Connection reset by peer, fd = 55
W (34851) TRANS_SSL: Poll timeout or error, errno=Success, fd=55, timeout_ms=10000
E (34861) TRANSPORT_WS: Error write header
I (34861) MQTTWSS_EXAMPLE: MQTT_EVENT_ERROR
E (34871) MQTT_CLIENT: Error write data or timeout, written len = -1, errno=0
MQTT_EVENT_DISCONNECTED
I (34881) MQTT_CLIENT: Client force reconnect requested
I (34891) MQTTWSS_EXAMPLE: Other event id:7
sent publish successful, msg_id=-1

FocosiGit, 2020-07-13:
I did some additional tests and with the smaller message size with a size of 347 bytes, it worked very well. My problem is that I need to send bigger messages.

ex:
		static char  					Msg1[]			= "{\"cmd\":\"M_PROC\",\"st\":\"OK\",\"val\":0,\"ts\":1594641010,\"lc\":1594641010,\"Mac\":\"B4:E6:2D:99:73:0B\",\"txt\":{\"CMD\":[\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,16,49]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[15,16,49]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,8,41]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,8,41]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,8,41]\"]}}";

Alvin1Zhang, 2020-07-14:
@FocosiGit Thanks for the detailed report and letting us know, we will look into.

ESP-Marius, 2020-07-14:
Hi @FocosiGit 

I wasn't able to reproduce any issues using 3.3.2 and sending your string. But I suspect this is simple the broker shutting you down for sending too much, it's a public test broker after all. 

Do you have another broker you can test with? I tested with both `test.mosquitto.org` and `mqtt.eclipse.org:443/mqtt`, and neither showed any issues, but that might just be because the load on the broker is lower at the moment.

FocosiGit, 2020-07-14:
Please, find below a new version of the program with new URL and additional information regarding memory:

#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>
#include "esp_wifi.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event_loop.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"

#include "esp_log.h"
#include "mqtt_client.h"

static const char *TAG = "MQTTWSS_EXAMPLE";

		static EventGroupHandle_t 		wifi_event_group;
const 	static int 						CONNECTED_BIT 		= BIT0;

		static char  					Topic1[]		= "FAC/0/NULL/NULL/NULL/NULL/NULL/NULL/NULL/TBLWFCLK/B4_E6_2D_99_73_0B/D_FDdb90a2b0c5bf90a1";
		static char  					Msg1[4*1024]	= "{\"cmd\":\"M_PROC\",\"st\":\"OK\",\"val\":0,\"ts\":1594641010,\"lc\":1594641010,\"Mac\":\"B4:E6:2D:99:73:0B\",\"txt\":{\"CMD\":[\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,16,49]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[15,16,49]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[15,8,41]\",\"L_MAN#[5,0,5]\",\"L0@5\",\"L1#[20,7,12]\",\"L2#[15,8,41]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[13,44,19]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[13,44,19]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[13,35,7]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[13,35,7]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[13,30,46]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[13,30,46]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,12]\",\"L2#[13,24,41]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,12]\",\"L2#[13,24,41]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[23,40,52]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,11]\",\"L2#[23,40,52]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[22,35,9]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,11]\",\"L2#[22,35,9]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[18,34,0]\",\"L_MAN#[5,0,5]\",\"L0@1\",\"L1#[20,7,11]\",\"L2#[18,34,0]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[17,47,15]\",\"L_MAN#[5,0,5]\",\"L0@5\",\"L1#[20,7,11]\",\"L2#[17,47,15]\",\"L_MAN#[2,0,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[17,40,35]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[17,36,56]\",\"L_MAN#[0,0,8]\",\"L1#[20,7,11]\",\"L2#[17,36,53]\",\"L_MAN#[0,0,7]\",\"L1#[20,7,11]\",\"L2#[17,36,51]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[17,35,28]\",\"L_MAN#[0,0,8]\",\"L1#[20,7,11]\",\"L2#[17,35,25]\",\"L_MAN#[0,0,7]\",\"L1#[20,7,11]\",\"L2#[17,35,21]\",\"L_MAN#[2,0,0]\",\"L0@Quinta-feira\",\"L1#[20,7,11]\",\"L2#[17,5,23]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[17,3,46]\",\"L_MAN#[1,0,11]\",\"L0@PGM2\",\"L1#[20,7,11]\",\"L2#[17,3,46]\",\"L_MAN#[1,1,11]\",\"L0@PGM2\",\"L1#[20,7,11]\",\"L2#[17,3,45]\",\"L_MAN#[0,0,11]\",\"L1#[20,7,11]\",\"L2#[17,3,42]\",\"L_MAN#[0,0,10]\",\"L1#[20,7,11]\",\"L2#[17,3,41]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[17,3,39]\",\"L_MAN#[0,0,8]\",\"L1#[20,7,11]\",\"L2#[17,3,36]\",\"L_MAN#[0,0,7]\",\"L1#[20,7,11]\",\"L2#[17,3,32]\",\"L_MAN#[1,0,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,41,1]\",\"L_MAN#[2,0,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,39,11]\",\"L_MAN#[1,1,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,37,0]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[15,37,0]\",\"L_MAN#[1,0,10]\",\"L0@PGM1\",\"L1#[20,7,11]\",\"L2#[15,37,0]\",\"L_MAN#[2,0,10]\",\"L0@PGM1\",\"L1#[20,7,11]\",\"L2#[15,35,1]\",\"L_MAN#[1,1,10]\",\"L0@PGM1\",\"L1#[20,7,11]\",\"L2#[15,30,8]\",\"L_MAN#[0,0,10]\",\"L1#[20,7,11]\",\"L2#[15,30,8]\",\"L_MAN#[1,0,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,30,8]\",\"L_MAN#[1,1,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,29,59]\",\"L_MAN#[2,0,2]\",\"L0@Sábado\",\"L1#[20,7,11]\",\"L2#[15,29,59]\",\"L_MAN#[0,0,9]\",\"L1#[20,7,11]\",\"L2#[15,7,3]\",\"L_MAN#[1,0,11]\",\"L0@PGM2\",\"L1#[20,7,11]\",\"L2#[15,7,3]\",\"L_MAN#[1,1,11]\",\"L0@PGM2\",\"L1#[20,7,11]\",\"L2#[14,56,27]\"]}}";

		static bool						Wifi_Connected 	= false;
		static bool						Mqtt_Connected	= false;
esp_mqtt_client_handle_t				Mqtt_Client;

static esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
        case SYSTEM_EVENT_STA_START:
            esp_wifi_connect();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
            Wifi_Connected 	= true;

            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            esp_wifi_connect();
            xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
            break;
        default:
            break;
    }
    return ESP_OK;
}

static void wifi_init(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid 		= "xxx",
            .password 	= "yyy",
        },
[... 213 lines cut ...]
E (28952) TRANSPORT_WS: Error read data
E (28952) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (28952) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
MQTT_EVENT_DISCONNECTED
I (28962) MQTT_CLIENT: Client force reconnect requested
I (28972) MQTTWSS_EXAMPLE: Other event id:7
I (35092) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
[... cut ...]
I (45352) MQTTWSS_EXAMPLE: Other event id:7
Published,msg_id=55964,size=2688,Free memory: 203136 bytes
I (51472) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
MQTT_EVENT_CONNECTED - Waiting Mqtt
E (56602) TRANSPORT_WS: Error read data
E (56602) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (56602) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
MQTT_EVENT_DISCONNECTED
I (56612) MQTT_CLIENT: Client force reconnect requested
I (56622) MQTTWSS_EXAMPLE: Other event id:7

ESP-Marius, 2020-07-15:
@FocosiGit I tested your program (only changing the wifi password and the broker to `wss://test.mosquitto.org:8081`) and everything still works fine here, published for several minutes without any errors.

Did you change anything in your sdkconfig/menuconfig? Are you sure there is no other changes to your project expect the ones you posted here?

FocosiGit, 2020-07-15:
This is the project directory without "build" sub-directory to avoid wasting space, to re-generate just compile again.
[Mqtt_wss.zip](https://github.com/espressif/esp-idf/files/4925266/Mqtt_wss.zip)

is there a possibility to be latency problem ?

ESP-Marius, 2020-07-17:
Hi again @FocosiGit 

Unfortunately still no luck reproducing your errors. Your latency looks OK to me.

Some more suggestions to help us get to the source of the issue:

- Could you attach a log where it fails, with the log level set to verbose? (Setting can be found by doing `idf.py` `menuconfig`, and then Component Config->Log Output->Default Log verbosity

- If feasible, test against a MQTT broker which you have access to the logs on. 

- See if switching to a better WiFI connection solves the issue

FocosiGit, 2020-07-17:
Thanks for the response.

I am using wifi network via cable company, and the problem is still happening.

I did one additional test, I started on my smart phone the function to work as a router, and I connected to the internet via smart phone, for my surprise everything worked perfectly, no more problems.

But anyway, I need to identify what is happening when I connect to the cable company, because other users may face the same issue.

I changed a bit my program to clarify the sent message step  and the confirmation when the feedback, when the program receives the confirmation from the server.

When it send the it prints: Sent message, waiting confirmation...,msg_id=38946,size=2688,Free memory: 161640 bytes

when it receives conformation from the server: MQTT_EVENT_PUBLISHED, msg_id=38946

below are the code changed:

   		msg_id = esp_mqtt_client_publish(Mqtt_Client, 							// esp_mqtt_client_handle_t
									     Topic1, 								// const char *topic
									     Msg1, 									// char *data
									     strlen(Msg1), 							// int len
									     1, 		 							// int qos
									     0);									// int retain
   		printf("Sent message, waiting confirmation...,msg_id=%d,size=%d,Free memory: %d bytes\n", msg_id, strlen(Msg1), esp_get_free_heap_size() );

and in the confirmation:

           case MQTT_EVENT_PUBLISHED:
                printf("MQTT_EVENT_PUBLISHED, msg_id=%d\n", event->msg_id);

**Below is the log that I captured as requested using my cable company:**

Toolchain path: /home/carlos/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "mqtt_websocket_secure" version: 1
CC build/main/app_main.o
Python requirements from /home/carlos/esp/esp-idf/requirements.txt are satisfied.
AR build/main/libmain.a

Generating libmain.a.sections_info
Generating esp32.project.ld
LD build/mqtt_websocket_secure.elf
esptool.py v2.8
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____.....___
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:b5:d4:b9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 25632 bytes to 15275...
Wrote 25632 bytes (15275 compressed) at 0x00001000 in 1.4 seconds (effective 151.7 kbit/s)...
Hash of data verified.
Compressed 776160 bytes to 473854...
Wrote 776160 bytes (473854 compressed) at 0x00010000 in 42.0 seconds (effective 147.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1757.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[... 588 lines cut ...]
V (114542) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (115562) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (116592) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (117622) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (118642) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (119692) MQTT_CLIENT: Sending fragmented message, remains to send 1761 bytes of 2688
D (119692) MQTT_CLIENT: mqtt_enqueue_oversized id: 19106, type=3 successful
D (119702) OUTBOX: ENQUEUE msgid=19106, msg_type=3, len=2783, size=8349
D (119712) MQTT_CLIENT: Sending fragmented message, remains to send 737 bytes of 2688
V (120442) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
Sent message, waiting confirmation...,msg_id=19106,size=2688,Free memory: 155396 bytes
V (121452) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/carlos/esp/esp-idf/tools/idf_monitor.py", line 134, in _run_outer
    self.run()
  File "/home/carlos/esp/esp-idf/tools/idf_monitor.py", line 228, in run
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/home/carlos/.local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

ESP-Marius, 2020-07-22:
@FocosiGit 

Thanks for the debug log and sorry for the late response. 

It definitely seems to be related/provoked by something related to your connection then. I'll continue to try to stress test it tomorrow and see if I can get a similar failure.

ESP-Marius, 2020-07-28:
@FocosiGit Hi again, unfortunately no luck reproducing this.

Could this be caused by some kind of smart firewall, just breaking the connection after some time (e.g. encrypted communication on uncommon port number being interrupted as it "resembles" some potentially dangerous packets). Check if there's anything you might have forgot to configure/ turn off/add exceptions in the company network and check the firewall logs.

If this doesn't turn up anything then another suggestion would be to send us a wireshark capture and use plain mqtt or ws (on the same port = 8081) so we can see if we see something in the packet capture.

FocosiGit, 2020-07-29:
Hi, it is not a case of any firewall because when using the 4G connection it works perfectly.   I imagine  the problem is related to latency, do you know if it is possible to increase the settings for timeout ?

I will give a look at the wireshark

ESP-Marius, 2020-07-31:
Wouldn't that actually support the theory that there is a firewall issue though? Since it works fine when on 4G/smart phone which bypasses the firewall, but has problem when connected to the company network (which might have a firewall setup)

You can adjust TCP/IP related parameters (including stuff that would cause timeouts) in the LWIP component in menuconfig as well as `MQTT_NETWORK_TIMEOUT_MS` in `mqtt_config.h`.

It should be possible to capture the packets on a wifi network as well. But doing it on the MQTT server side is also possible.

Alvin1Zhang, 2020-08-13:
@FocosiGit Thanks for reporting, would you please help share if any updates for this issue? Thanks.

ESP-Marius, 2020-09-17:
Closing issue due to inactivity, feel free to reopen if this is still an ongoing issue.

closed 2020-09-17 by ESP-Marius

MBuratto, 2022-07-28:
I'm having the same problem with large packages. Did anyone manage to solve it?

Manimekalai-Devendran, 2023-07-23:
@ESP-Marius  and @Alvin1Zhang I'm also having the same problem with large packets. But it's happening only I'm connecting to my phone network. Did anyone solve it?