espressif/esp-idf #5584: ESP32 sprintf snprintf无法将source赋值给dest, source大小大于dest

iot1688, 2020-07-14:
你好
在ESP32里边，版本4.0：

snprintf(sysCfg.com_stop_bits,sizeof(sysCfg.com_stop_bits)-1,"%s",mqttupdatavar[0].var_value_array_bak[3]);

提示BUG:
error: '%s' directive output may be truncated writing up to 9 bytes into a region of size 1 [-Werror=format-truncation=]
      snprintf(sysCfg.com_stop_bits,sizeof(sysCfg.com_stop_bits)-1,"%s",mqttupdatavar[0].var_value_array_bak[3]);

在ESP8266 WINDOWS环境下编译通过，查的应该是编译版本问题。
请问这个有什么好的处理方法？

snailwft, 2020-07-14:
同问，这个版本出现这种问题有点烦人。

Alvin1Zhang, 2020-07-14:
感谢反馈。这个编译警告加入在了 gcc 8 中，ESP8266 SDK 仍在使用旧版本的 gcc，因此该编译警告未出现。该警告用来避免因误裁剪字符串引发的问题，此类裁剪可能引入安全相关的问题。

以下四种方案中，任意一种方案都可以解决该问题：

- 在你的项目中使能该 config 选项以关闭在 gcc 8 中加入的所有警示：
https://docs.espressif.com/projects/esp-idf/en/v4.0/api-reference/kconfig.html#config-compiler-disable-gcc8-warnings
- 验证 snprintf 的返回值，这样编译器就知道，你已了解并不是所有的 bytes 都会被写入
#include <assert.h>
int r = snprintf(sysCfg.com_stop_bits,sizeof(sysCfg.com_stop_bits)-1,"%s",mqttupdatavar[0].var_value_array_bak[3]);
assert(r >= 1);

- 如果你只需要拷贝一个字符:
sysCfg.com_stop_bits[0] = mqttupdatavar[0].var_value_array_bak[3][0]
sysCfg.com_stop_bits[1] = 0;

- 只在你的组件里禁用/关闭该警告:
如果使用的是 GNU 构建系统，将以下信息加入到你项目中的 main/component.mk 及其他 component.mk 中；
CFLAGS += -Wno-format-truncation 
如果使用的是 CMake 构建系统，将以下信息加入到你项目中的 main/CMakeLists.txt 及其他 CMakeLists.txt 中；
target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-format-truncation) 

（对于 CMake，该行应该置于 idf_component_register() 行之后。）

iot1688, 2020-07-16:
TKS

closed 2020-07-16 by iot1688