espressif/esp-idf #5585: Error -76 from MbedTLS

schroedermarius, 2020-07-14:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.2-dev-1303-gcd4fa46f5
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     macOS
- Using an IDE?: VSCode
- Power Supply:         USB

## Problem Description

Running esp-azure and sending telemetry data. After a time im getting following error on each send:
Error: Time:Tue Jul 14 10:04:30 2020 File:/Users/marius/esp/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_send_async Line:552 tlsio_esp_tls_send_async without a prior successful open
Error: Time:Tue Jul 14 10:04:30 2020 File:/Users/marius/esp/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:sendPacketItem Line:395 Failure sending control packet data
Error: Time:Tue Jul 14 10:04:30 2020 File:/Users/marius/esp/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:mqtt_client_publish Line:1101 Error: mqtt_client_publish send failed
Error: Time:Tue Jul 14 10:04:30 2020 File:/Users/marius/esp/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c Func:publish_mqtt_telemetry_msg Line:898 Failed attempting to publish mqtt message

After this happened a few times following error is printed:

Error: Time:Tue Jul 14 10:04:44 2020 File:/Users/marius/esp/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_destroy Line:153 tlsio_esp_tls_destroy called while not in TLSIO_STATE_CLOSED.
I (2889204) azure_client: 

Connection Status result:IOTHUB_CLIENT_CONNECTION_UNAUTHENTICATED, Connection Status reason: IOTHUB_CLIENT_CONNECTION_EXPIRED_SAS_TOKEN

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Core  0 register dump:
PC      : 0x6572757a  PS      : 0x00060630  A0      : 0x800e48ae  A1      : 0x3ffd90b0  
A2      : 0x3ffd7944  A3      : 0x00000000  A4      : 0x3ffd7974  A5      : 0x40a68000  
A6      : 0x7ff00000  A7      : 0x00000200  A8      : 0x801a0f92  A9      : 0x3ffd9080  
A10     : 0x3ffd77ac  A11     : 0x00000000  A12     : 0x3f40342c  A13     : 0x002c15f4  
A14     : 0x3f40b1cc  A15     : 0x7ff5fb1c  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5  

Backtrace:0x65727577:0x3ffd90b0 |<-CORRUPTED

Im using the light sleep mode, if this could be helpflu.

### Expected Behavior

No errors.

Alvin1Zhang, 2020-07-14:
Thanks for reporting, we will look into.

schroedermarius, 2020-07-15:
Thanks!
To me it really seems that it is connected to the light sleep.
Is it possible that my tasks which uses the esp-azure to send telemetry data have to be delayed after sleep, so the wifi can connect?
Is there a possibility to wait for wifi after awake of light sleep?

schroedermarius, 2020-07-17:
Update: This also occurs on the current stable release (4.0.1). ESP-IDF current master branch.