espressif/esp-idf #5586: Other Possibilities than RSSI to judge STA Link Quality (Noise Floor?)

Curclamas, 2020-07-14:
**Is your feature request related to a problem? Please describe.**

As a developer I would like to read more information than just RSSI in order to judge ESP32 STA Mode connection quality with the AP.

**Describe the solution you'd like**

I would like to know if there is more than just RSSI available, in the `reserved` part, that would let me judge the quality of the link
https://github.com/espressif/esp-idf/blob/release/v3.3/components/esp32/include/esp_wifi_types.h#L246-L255

Interesting for me are:
- noise floor (why only available in promiscous mode?)
- Rx quality
- Tx quality

**Describe alternatives you've considered**

As we need those data in order to analyse the link quality to our device there is no real alternative. Of course we could do things like speed-tests etc. 
But that would waste mobile data and would be a much "higher level" of analysis

**Additional context**

We have a large scale of devices deployed in the field which are connected with WiFi-LTE routers. We would like to internally judge the link quality, not just with respect to signal strength (RSSI) but also other parameters (noise environment, antenna matching, disturbed signals due to Metal in close vicinity etc.) 

Sometimes we have the situation that RSSI is good, but speed is still poor, despite the LTE connection being stable and fast.

TLDR: we have no control what our customers do with the devices and want to be able to remotely judge the connection quality as good as possible.

Alvin1Zhang, 2020-07-15:
Thanks for raising this feature request.

liuzfesp, 2020-07-15:
Hi @Curclamas, as you pointed out, currently only RSSI. I think we can provide API to get noise floor, but it requires some time based on current release plan. Will update here once it's ready.

Curclamas, 2020-07-19:
@liuzfesp thx for the quick response! I would have three questions:

- What APIs could you imagine for that functionality? How would it work?
- What could be a rough time schedule for this?
- Is there a way to "try out" these features before the official release? Would be interesting for my company to experiment a little around this!

Thanks in advance!

liuzfesp, 2020-07-20:
HI @Curclamas,

- Maybe we can provide a new API `esp_wifi_get_noise_floor()`
- I just can't promise the exact date, currently we still don't have a final decision that this API should be supported in future IDF releases. Currently we have very strict criteria for supporting new WiFi APIs. Anyway, I will update you if we decide to support it in IDF.
- Yes, we can do it. Please @esp-HarveyRong build a private WiFi lib for it.

Curclamas, 2020-07-20:
@liuzfesp thanks for the quick answer! I am looking forward to test it! However @esp-HarveyRong when you build it: will it be compatible to v3.3.2? (https://github.com/espressif/esp-idf/tree/v3.3.2)
Since this is the version we are working with.

liuzfesp, 2020-07-22:
Hi @Curclamas ,

Please overwrite your libpp.a under $IDF_PATH/components/esp32/lib and do a test. The debug API is: int esp_wifi_get_noise_floor(void), you need to extern it before using it.

[libpp.a.zip](https://github.com/espressif/esp-idf/files/4957642/libpp.a.zip)

Alvin1Zhang, 2020-08-13:
@Curclamas Thanks for reporting, would you please help share if any updates? Thanks.

AxelLin, 2020-08-13:
Hi @liuzfesp 

I test it with master tree (v4.3-dev-771-gc77c4ccf6c43) by replacing components/esp32/lib/libpp.a file.
However, I got undefined reference to `esp_wifi_get_noise_floor' build error.

liuzfesp, 2020-08-13:
Hi @AxelLin the libpp.a is based on v3.3.2, not v4.3.

AxelLin, 2020-08-13:
> Hi @AxelLin the libpp.a is based on v3.3.2, not v4.3.

Hi @liuzfesp 
Got it. Just thought such new function should be tested on master first rather than stable branches.

liuzfesp, 2020-08-13:
Yes, @AxelLin, the libpp.a is just a test patch for @Curclamas

AxelLin, 2020-08-28:
> Yes, @AxelLin, the libpp.a is just a test patch for @Curclamas
hi @liuzfesp 
Just wondering how to move on since there is no response from @Curclamas for 2 weeks.
Do you consider adding this function to master so more people can test it?

Curclamas, 2020-10-14:
Hi @AxelLin @liuzfesp sorry for the late response! We have some delays with our rollout of v3.3.2 (actually there have been new releases since then?) 
Does the code also work with the latest 3.3.x? We will test it as soon as we are up to date!

TianaESP, 2021-11-16:
@Curclamas Sorry, 3.3.x won't support this, please switch to a higher IDF version.

AxelLin, 2021-11-16:
> @Curclamas Sorry, 3.3.x won't support this, please switch to a higher IDF version.

@TianaESP 
Which esp-idf version supports this? Using which API?

TianaESP, 2021-11-16:
@AxelLin  IDF v4.0 and above, API:  Add an API to get WiFi statistics.

AxelLin, 2021-11-16:
> @AxelLin IDF v4.0 and above, API: Add an API to get WiFi statistics.

@TianaESP 
I don't find the commit "Add an API to get WiFi statistics" in any branch.
Can your provide the commit id or the function name?

TianaESP, 2021-11-18:
@AxelLin  Sorry, I made a mistake. I will check again and come back.

mhdong, 2021-11-18:
@AxelLin You can get noise floor information by esp_wifi_set_cai_rx_cb(wifi_csi_cb_t cb, void* ctx).

[image]

[image]

AxelLin, 2021-11-18:
> @AxelLin You can get noise floor information by esp_wifi_set_cai_rx_cb(wifi_csi_cb_t cb, void* ctx).
> 

@mhdong 
I don't find this function in esp-idf. Which branch includes this function?

jack0c, 2022-01-06:
Should be esp_wifi_set_csi_rx_cb

AxelLin, 2022-01-06:
> Should be esp_wifi_set_csi_rx_cb

I see, but it seems overkill to enable csi just for getting noise_floor.

jack0c, 2023-06-02:
@AxelLin You are right, Sorry for the delay, we will add some APIs to do this.

AxelLin, 2023-07-24:
> @AxelLin You are right, Sorry for the delay, we will add some APIs to do this.

Any update?
What's the target timelne to add API for getting noise floor?

AxelLin, 2023-09-08:
> @AxelLin You are right, Sorry for the delay, we will add some APIs to do this.

But there is no update for 3 Months? @jack0c @mhdong

hansw123, 2023-10-19:
> > @AxelLin You are right, Sorry for the delay, we will add some APIs to do this.
> 
> But there is no update for 3 Months? @jack0c @mhdong

HI AxeILin，We have completed the development of this API and are now in the testing phase. It will be updated in later versions as planned.

hansw123, 2023-10-24:
@AxelLin  Hi AxelLin, The NOISE FLOOR we're acquiring now is brought inside the chip, determined by the chip itself, is that useful to you? By the way, what is the exact definition of NOISE FLOOR that you  are looking for.

AxelLin, 2023-11-11:
> @AxelLin Hi AxelLin, The NOISE FLOOR we're acquiring now is brought inside the chip, determined by the chip itself, is that useful to you? By the way, what is the exact definition of NOISE FLOOR that you are looking for.

@hansw123 
Do you mean the NOISE FLOOR output by the new API  is different from the noise_floor values got via esp_wifi_set_csi_rx_cb()?

See the origianl issue report https://github.com/espressif/esp-idf/issues/5586#issue-656602871,
the main point is to help diagnostic some wifi issues, especially when RSSI is good but speed is poor.

hansw123, 2023-11-14:
> > @AxelLin Hi AxelLin, The NOISE FLOOR we're acquiring now is brought inside the chip, determined by the chip itself, is that useful to you? By the way, what is the exact definition of NOISE FLOOR that you are looking for.
> 
> @hansw123 Do you mean the NOISE FLOOR output by the new API is different from the noise_floor values got via esp_wifi_set_csi_rx_cb()?
> 
> See the origianl issue report [#5586 (comment)](https://github.com/espressif/esp-idf/issues/5586#issue-656602871), the main point is to help diagnostic some wifi issues, especially when RSSI is good but speed is poor.

We may have misinterpreted the exact definition of noise floor from the beginning; noise floor does not represent the energy level of ambient noise

AxelLin, 2024-06-06:
@0xjakob @mhdong 

There is no improvement for this issue?
I do hope esp32 can provide some API to help debug issues when device has unexpected behavior.
e.g. some device is extreme slow at certain environment even though the device is very closed to AP but we don't know why because there is no error at all (just slow for unknown reason).