espressif/esp-idf #5587: cmake error in esp-idf 4.3 with respect to \ symbol in Windows 10

vsupacha, 2020-07-14:
## Environment

- Development Kit:      Cucumber RS board by Gravitech
- Module or chip used:  ESP32-S2-WROVER
- IDF version: v4.3-dev-472-gcf056a7d0
- Build System:        CMake 3.16.4 in VS Code with ESP-IDF Extension 0.3.0
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows 10
- (Windows only) environment type: Plain Command Prompt in VS Code environment
- Using an IDE?: Visual Studio Code with ESP-IDF Extension 0.3.0
- Power Supply:         USB

## Problem Description
After updated to ESP-IDF 4.3, VS Code with ESP-IDF extension can not build any project. Test with HelloWorld example. Track reported errors as cmake issue

**Invalid character escape '\U'**

to line 2 of **build_properties.temp.cmake** file

set(PYTHON C:\Users\vsupa\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe)

Expect **\Users** which is mandatory in Windows environment to cause the problem.

### Expected Behavior
No problem for building an example HelloWorld project in VS Code with ESP-IDF extension. Previous version (ESP-IDF 4.2) was ok.

### Actual Behavior
Error for commands in VS Code with ESP-IDF extension.

### Steps to reproduce

1. Use ESP-IDF extension to download ESP-IDF master (4.3) and ESP-IDF tools (2020r2). Existing Python is 3.7 from python.org.
2. Generate "Hello World" project.
3. Select ESP-IDF command - select target > error in VS Code Extensions. 
4. Manually select ESP32-S2 as target.
5. Select ESP-IDF command to build code > error in VS Code Extensions. Check ESP-IDF task output and found issues.

### Code to reproduce this issue
Example HelloWorld code, no modifications.

## Debug Logs

```
Python requirements from C:\Users\vsupa\esp-idf\requirements.txt are satisfied.
CMake Error at C:/Users/vsupa/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at build_properties.temp.cmake:2 (set):
    Syntax error in cmake code at
      C:/Users/vsupa/Desktop/CucumberRS/tmp_project/hello_world/build/build_properties.temp.cmake:2
    when parsing string
      C:\Users\vsupa\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe  

    Invalid character escape '\U'.

  Call Stack (most recent call first):
    C:/Users/vsupa/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)
Call Stack (most recent call first):
  C:/Users/vsupa/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  C:/Users/vsupa/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)      
  CMakeLists.txt:6 (project)
-- Configuring incomplete, errors occurred!
```

ammaree, 2020-07-14:
Fix is as in #5512 
Change in idf.py at line 52

> import pathlib
os.environ["PYTHON"] = pathlib.Path(sys.executable).as_posix()

vsupacha, 2020-07-14:
After manually reverse ESP-IDF version to 4.1-beta2 and re-install ESP-IDF tools, everything is ok.

Alvin1Zhang, 2020-07-15:
Thanks for reporting and notes, let us close this ticket and keep tracking in https://github.com/espressif/esp-idf/issues/5512. Thanks.

closed 2020-07-15 by Alvin1Zhang