espressif/esp-idf #5588: ESP32-S2: TinyUSB 100% broken by a4d0033

atanisoft, 2020-07-14:
## Environment

- Development Kit:      ESP32-S2
- Kit version (for WroverKit/PicoKit/DevKitC): WROVER
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): a4d0033 (latest master)
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0r2
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply:         USB or external 3.3V

## Problem Description

commit a4d0033 changed the behavior of configure_pins() leading to LoadProhibited when calling tinyusb_driver_install:

### Code to reproduce this issue

tinyusb example app will reproduce this.

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40085400  PS      : 0x00060c30  A0      : 0x8008548e  A1      : 0x3ffddb40  
A2      : 0x00004000  A3      : 0x3f0214f4  A4      : 0x00000000  A5      : 0x00000030  
A6      : 0x00000000  A7      : 0x3ffddb40  A8      : 0x000000b8  A9      : 0x0000003a  
A10     : 0x00060c23  A11     : 0x00000000  A12     : 0x00060c20  A13     : 0x00000000  
A14     : 0x3f02bc86  A15     : 0x00000011  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00004000  LBEG    : 0x00060c20  LEND    : 0x00000000  LCOUNT  : 0x4002a5c1  
0x4002a5c1: _xt_user_exc at .../esp-idf/components/freertos/xtensa/xtensa_vectors.S:627
Backtrace:0x400853fd:0x3ffddb40 0x4008548b:0x3ffddb70 0x400854f6:0x3ffddbb0 0x40090e21:0x3ffddbf0 0x4002f1cc:0x3ffdddb0
0x400853fd: gpio_ll_input_enable at tinyusb.c:?
0x4008548b: configure_pins at tinyusb.c:?
0x400854f6: tinyusb_driver_install at ??:?
0x40090e21: app_main at ..../esp32s2io/build/../main/esp32s2io.cpp:431
0x4002f1cc: main_task at .../esp-idf/components/freertos/xtensa/port.c:511
```

The fix for this is to update https://github.com/espressif/esp-idf/blob/master/components/tinyusb/port/esp32s2/src/tinyusb.c#L37 to be:
```
if (iopin->pin != GPIO_MATRIX_CONST_ONE_INPUT && iopin->pin != GPIO_MATRIX_CONST_ZERO_INPUT) {
  gpio_ll_input_enable(&GPIO, iopin->pin);
}
```
since gpio_ll_input_enable does no bounds checking (something else that should be done!)

chegewara, 2020-07-14:
Yes. console CDC works more or less pretty good, but thats all. Tinyusb component in app does not work at all (yes, console is UART).

an-dr, 2020-07-16:
HI! Indeed, we broke it, thank you for the report. I'll fix it soon

closed 2020-08-03 by espressif-bot