espressif/esp-idf #5589: Firmware for compliance testing (RF)

renanr2, 2020-07-14:
Hello everyone!

I'm facing a RF test for my project with ESP32 to get a national standard certification by ANATEL (Brazil).

I need to do a few configurations on my device, for each test.

I read the IDF user guide, but i can't found out a way to set the bluetooth modulation between GFSK and 8-DPSK.

It has a kind of default firmware for compliance testing? Or, a way to set bluetooth modulation by API?

Please help me.

renanr2, 2020-07-14:
I found it at espressif idf tools.
https://www.espressif.com/sites/default/files/tools/ESP32%26ESP8266_RF_Performance_Test_EN.zip

closed 2020-07-14 by renanr2

mfalconf, 2022-08-09:
Hi @renanfgpascoal, Did the ESPRFTestTool help you with the certification requested by ANATEL?

menachg, 2022-10-02:
do we have the source for the EspRftestTool firmware?
i need it for rf testing in conjunction with other modules enabled

natanael127, 2023-09-21:
I have the same issues of @mfalconf and @menachg

guilhermep-git, 2024-05-08:
I have the same issues to set the different types of modulation for Bluetooth Classic, BLE and Wi-Fi  with EspRFTestTool_v3.6 software. Aparently the software not able to choose modultaion. I need this to get certification of my product by ANATEL.

The Certificate of Conformity for ESP32-WROOM-32D in Brazil (ANATEL) show several modulations.
[image]

DevxMike, 2024-08-22:
Hello,
@guilhermep-git @menachg @natanael127 @mfalconf can you please share some insights regarding this topic?

guilhermep-git, 2024-08-23:
Hello @DevxMike,
ESP RF Test Tool worked in my case. It was used by certification laboratory with success.
The ESP RF Test Guide was used as reference.

DevxMike, 2024-08-26:
Hello @guilhermep-git ,
Thank you very much.