espressif/esp-idf #5590: Unable to Build project due to Undefined reference error.

noel-ED, 2020-07-14:
Hi,
I was using a Linux environment to develop my code but now I have upgraded my system and I am working on Windows and I have set it up following the "Get-Started" Guide available online for ESP32 website.
I am not sure of the previous IDF version which I used in the Linux env but all I know is that it was a September 2019 release and I would use make flash command to get to program my controller.
Now after migrating my code, I extracted a project from the examples folder and I built it using idf.py commands.
My current IDF version is: **IDF version: v4.2-dev-1905-g625bd5eb1**

I am using the mqtt/ssl example. When built straight after copying it works fine. But after I replace the files with my previously developed files it gives me an Undefined reference error: 
`FAILED: mqtt_ssl.elf
cmd.exe /C "cd . && D:\ESP\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\mqtt_ssl.elf.rsp  -o mqtt_ssl.elf  && cd ."
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.c.obj):(.literal.app_main+0x48): undefined reference to `SenseValues'
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.c.obj):(.literal.app_main+0x50): undefined reference to `mqtt_app_start'
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.c.obj): in function `app_main':
d:\esp\examples\ssl\build/../main/app_main.c:128: undefined reference to `mqtt_app_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1'

once I comment the faulty function calls the code builds fine and I am able to flash it. The definitions are proper since it has worked in the previous IDF version.

Could you help me I am not well versed with this environment.

Hope to hear from you soon.

Regards,
Noel

igrr, 2020-07-14:
Hi Noel, could you please post what your project layout looks like (which files are in which directories) and also your CMakeLists.txt file? Also please indicate where the SenseValues function is defined.

noel-ED, 2020-07-14:
Hi,
I have created all of my files and located them in the same folder (main) as is my app_main.c file.
I have a mqtt_handler.c/.h file and a sensor_values.c/.h file in created in order to handle individual tasks. Sense value is a task created in the main function and is defined in the sensor_value.c file to which a declaration is included in the sensor_value.h file and it in included in app_main.c. Similarly it goes for mqtt_app_start.
The CMakeLists in the parent directory is as follows 
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(mqtt_ssl)

target_add_binary_data(mqtt_ssl.elf "main/mqtt_eclipse_org.pem" TEXT)

 and the on within the main folder is as follows:
idf_component_register(SRCS "app_main.c"
                    INCLUDE_DIRS ".")

please check the snip of the two directories
[image]

[image]

Regards,
Noel

igrr, 2020-07-14:
Could you try adding the source files (as_cJSON.c, mqtt_handler.c, sensor_values.c) to the list of source files, after "app_main.c" in CMakeLists.txt? Otherwise these files are going to be not compiled.

(See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#minimal-component-cmakelists for an example and documentation)

noel-ED, 2020-07-14:
Thank you!! It worked.
I am really surprised with your prompt reply. Thank you again.

closed 2020-07-14 by noel-ED