espressif/esp-idf #5591: esp_ota_begin() breaks wifi connection

dreske, 2020-07-14:
### Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-dev-1856-g00148cd0c
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description
Calling esp_ota_begin() with OTA_SIZE_UNKNOWN randomly disconnects wifi.
Found this old issue #2083 with exactly the same behavior.

I've already set some 'aggressive' settings for spi flash I think (50 Ticks yield every 1ms):
```
CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=1
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=50
```

Tested with different modules, same behavior.

### Expected Behavior
Flash is erased successfully without wifi interruption

### Actual Behavior
Wifi breaks in some (around 30%) cases

### Steps to reproduce
Calling esp_ota_begin() in a loop and log output

### Code to reproduce this issue

```cpp
void start_ota(void* param)
{
    while(1) {
        esp_err_t err;        
        esp_ota_handle_t update_handle = 0 ;
        const esp_partition_t *update_partition = NULL;

        ESP_LOGI(TAG, "Starting OTA example");
        const esp_partition_t *configured = esp_ota_get_boot_partition();
        const esp_partition_t *running = esp_ota_get_running_partition();

        if (configured != running) {
            ESP_LOGW(TAG, "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
                 configured->address, running->address);
            ESP_LOGW(TAG, "(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
        }
        ESP_LOGI(TAG, "Running partition type %d subtype %d (offset 0x%08x)",
                running->type, running->subtype, running->address);

        update_partition = esp_ota_get_next_update_partition(NULL);
        update_partition->subtype, update_partition->address);
        
        ESP_LOGI(TAG, "esp_ota_begin");
        err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "esp_ota_begin failed (%s)", esp_err_to_name(err));
            continue;
        }
        ESP_LOGI(TAG, "esp_ota_begin succeeded");
        vTaskDelay(5000 / portTICK_RATE_MS);
    }
}

void app_main(void)
{
    // Initialize NVS.
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // 1.OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // 2.NVS partition contains data in new format and cannot be recognized by this version of code.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
    */
    ESP_ERROR_CHECK(example_connect());

#if CONFIG_EXAMPLE_CONNECT_WIFI
    /* Ensure to disable any WiFi power save mode, this allows best throughput
     * and hence timings for overall OTA operation.
     */
    esp_wifi_set_ps(WIFI_PS_NONE);
#endif // CONFIG_EXAMPLE_CONNECT_WIFI

    xTaskCreate(&start_ota, "advanced_ota_example_task", 1024 * 8, NULL, 1, NULL);
}
```

## Debug Logs

```
I (893078) advanced_https_ota_example: Starting OTA example                                                                                                                                                                                  
I (893078) advanced_https_ota_example: Running partition type 0 subtype 0 (offset 0x00010000)                                                                                                                                                
I (893078) advanced_https_ota_example: Writing to partition subtype 16 at offset 0x110000                                                                                                                                                    
I (893088) advanced_https_ota_example: esp_ota_begin                                                                                                                                                                                         
I (896548) wifi:state: run -> init (cf00)                                                                                                                                                                                                    
I (896548) wifi:pm stop, total sleep time: 0 us / 17210844 us                                                                                                                                                                                
                                                                                                                                                                                                                                             
I (896548) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1                                                                                                                                                                        
I (896558) example_connect: Wi-Fi disconnected, trying to reconnect...                                                                                                                                                                       
I (896798) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1                                                                                                                                                                        
I (896798) wifi:state: init -> auth (b0)
I (896798) wifi:state: auth -> assoc (0)
I (896808) wifi:state: assoc -> run (10)
I (896818) wifi:connected with home, aid = 1, channel 6, BW20, bssid = 18:e8:29:c1:dd:75
I (896818) wifi:security: WPA2-PSK, phy: bgn, rssi: -54
I (896818) wifi:pm start, type: 0

I (896888) advanced_https_ota_example: esp_ota_begin succeeded

```

Alvin1Zhang, 2020-07-15:
Thanks for reporting, we will look into.

copercini, 2020-07-27:
I'm having the exactly same issue but with 100% of fail rate. Here I'm using a webserver to upload the new .bin file to OTA, and it only happens in WiFi STA mode, when using WiFi AP or ethernet, the connection with WiFi STA still drops, but OTA works

shubhamkulkarni97, 2020-07-29:
Hi @copercini @dreske,

I'm enable to reproduce this issue. Is there anything specific change in configuration which causes this issue?
Are you facing this issue with custom code or with IDF examples as well?

Can you share sdkconfig file for the same?

dreske, 2020-07-29:
Hi, also with the OTA examples. 
I'll share my sdkconfig later, but its default, except the wifi config and my changes regarding spi flash (mentioned above)

Alvin1Zhang, 2020-08-21:
@dreske Thanks for reporting, would you please help share if any updates for this issue? Thanks.

shubhamkulkarni97, 2020-09-03:
Hello @copercini, thanks for the update.

I could reproduce the issue on mentioned IDF version, however it is fixed in https://github.com/espressif/esp-idf/commit/4761c4059ab6dd68aa5d2c009946135d4edd118d.

@dreske @copercini, updating your branch with latest changes should fix the issue.

Thanks,
Shubham

shubhamkulkarni97, 2020-09-14:
@dreske @copercini Any updates on the issue?

dreske, 2020-09-19:
Looks good after some quick tests. Thank you!

Alvin1Zhang, 2020-09-22:
Good to hear the issue fixed. Feel free to reopen. Thanks.

closed 2020-09-22 by Alvin1Zhang

chrismerck, 2021-03-12:
> I could reproduce the issue on mentioned IDF version, however it is fixed in 4761c40.   [release/v4.2]

Yikes, has this fix been backported to IDF v3?

A new ESP32-WROVER-E product we are launching takes 14+ seconds for `esp_ota_begin`. Sometimes it even triggers the task watchdog triggers shortly `esp_ota_begin` returns.

[image]

Alvin1Zhang, 2021-03-15:
@chrismerck Thanks for sharing the updates. After preliminary analysis, seems what you shared may be a new issue, would you please help create a new ticket per the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Thanks.