espressif/esp-idf #5592: ESP32 BLE mesh + Apple ANCS

owensc84, 2020-07-14:
Is it possible to run the BLE mesh stack with various server and client models along side Apple ANCS on the ESP32-WROOM-32E without rewriting extensive parts of the IDF?

I need the ESP32 to be a BLE mesh proxy node using the GATT proxy service and connect to an iPhone via ANCS at the same time.

If so, is it as simple as adding the ANCS demo code to my BLE mesh project? I suspect not.

Any advice or tips on how BLE mesh and Apple ANCS can coexist would be greatly appreciated.

Campou, 2020-07-30:
Hi @owensc84 

For Apple ANCS, ESP32 needs to advertise the corresponding packets. After connected with an iPhone, ESP32 will work as a GATT client for receiving notifications, etc.

And for BLE Mesh, ESP32 needs to advertise the mesh specific packets, and after connected with another phone, ESP32 will work as a GATT server for receiving mesh proxy packets.

Based on the above scenario, maybe we can enable Apple ANCS firstly. When ESP32 is connected with the iPhone, BLE Mesh can be initialized and enabled for provisioning.

Thanks.

owensc84, 2020-07-31:
@Campou , thank you for the response
When I try your suggestion the ANCS connects successfully but when I enable BLE mesh provisioning and attempt to provision the ANCS gattc callback function is responding to BLE mesh provisioning requests from my phone.  This is of course causing issues.  Is there a way to distinguish between the ANCS requests and BLE Mesh requests?  Or distinguish between ANCS connection and BLE Mesh connection.

I have tried the opposite way, provision BLE Mesh and then connect to the esp32 via GATT proxy server with the nrfMesh App.  I then send to a model that initializes the ANCS module and starts ANCS advertising.  When I do this I get some errors and my iphone does not see the ANCS advertisement.  The ANCS gattc callback also starts to respond to BLE mesh requests as described above.

Any help is greatly appreciated!

Campou, 2020-08-10:
Hi @owensc84 

Sorry for the late reply.

You can have a try with the [gattc_gatts_coex](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/gattc_gatts_coex). In this example, when the device is connected with a phone, the connected event will be notified to the application layer 3 times, one for gattc, and the other two for gatts.

And the coexistence of ANCS with BLE Mesh is like the above example (i.e. gattc for ANCS and gatts for BLE Mesh). The difference is that the gatts events for BLE Mesh will be handled internally (without notifying to the application layer), while the gattc events for ANCS will be notified to the application layer.

Here the `conn_id` could be used to distinguish the connections for ANCS and BLE Mesh. For example, we can record the `conn_id` for ANCS when it's connected. So later if another disconnection event for BLE Mesh comes, the `conn_id` should be different with the previous value of ANCS.

Another challenge here is how to manage the BLE advertising for ANCS and BLE Mesh. Since BLE Mesh advertises almost all the time, so the better approach is to use the mesh adv queue to interleave the advertising of ANCS packets and BLE Mesh packets.

And for BLE Mesh, we have added a option for this operation, just enable the `"Support sending normal BLE advertising packets"` in BLE Mesh menuconfig, then we can use the `esp_ble_mesh_start_ble_advertising` function to post the ANCS advertising packets to mesh adv queue. But this approach has impact on the BLE Mesh advertising, so we should use se this function to advertise the ANCS packets just for a certain period, and after the connection for ANCS is established, the ANCS packets should not be sent any more, except when the connection of ANCS is disconnected.

Thanks.

owensc84, 2020-08-11:
Hi @Campou 
I appreciate the information.  We have decided to try development of our product with ESP WiFi Mesh instead.

Alvin1Zhang, 2020-08-12:
Thanks for reporting and updates, feel free to reopen, thanks.

closed 2020-08-12 by Alvin1Zhang