espressif/esp-idf #5596: blufi heap corrupt?

chegewara, 2020-07-15:
Past few days i am struggling with heap corruption. It is fine as long as i wont connect with blufi app. After connecting with blufi app and exchanging few requests app crash on `heap_caps_check_integrity`:

```
ASSERT_PARAM(0 10), in lld_pdu.c at line 537
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008617d  PS      : 0x00060634  A0      : 0x80089df1  A1      : 0x3ffbec10  
0x4008617d: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdcb4  A7      : 0x60008050  A8      : 0x8008617d  A9      : 0x3ffbebf0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400860b5  LEND    : 0x400860bc  LCOUNT  : 0x00000000  
0x400860b5: r_assert_param at ??:?

0x400860bc: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40009203  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008617d
0x4008617d: r_assert_param at ??:?

ELF file SHA256: c68ee3b0646fbc31

Backtrace: 0x4008617d:0x3ffbec10 0x40089dee:0x3ffbec30 0x400485a5:0x3ffbec60 0x4004868d:0x3ffbec80 0x4008a686:0x3ffbeca0 0x4008b65f:0x3ffbecc0 0x40082d19:0x3ffbece0 0x4008b8a5:0x3ffd0640 0x400924ce:0x3ffd0650 0x40095069:0x3ffd0680 0x400d4e3b:0x3ffd06c0 0x400d4e84:0x3ffd06f0 0x400e1f1c:0x3ffd0720 0x400dd918:0x3ffd0780 0x4009154b:0x3ffd07b0
0x4008617d: r_assert_param at ??:?

0x40089dee: r_lld_pdu_rx_handler at ??:?

0x4008a686: r_rwble_isr at ??:?

0x4008b65f: r_rwbtdm_isr_wrapper at intc.c:?

0x40082d19: _xt_lowint1 at /home/chegewara/esp/v3.3/components/freertos/xtensa_vectors.S:1154

0x4008b8a5: btdm_bb_isr at intc.c:?

0x400924ce: vTaskExitCritical at /home/chegewara/esp/v3.3/components/freertos/tasks.c:5064

0x40095069: multi_heap_internal_unlock at /home/chegewara/esp/v3.3/components/heap/multi_heap.c:700
 (inlined by) multi_heap_check at /home/chegewara/esp/v3.3/components/heap/multi_heap.c:680

0x400d4e3b: heap_caps_check_integrity at /home/chegewara/esp/v3.3/components/heap/heap_caps.c:416

0x400d4e84: heap_caps_check_integrity_all at /home/chegewara/esp/v3.3/components/heap/heap_caps.c:425
```

IDF: ESP-IDF v3.3.2-270-gf4333c8e3
module: wroom

In meantime i tested all options i think, with and without soft coex too.

Thanks

chegewara, 2020-07-15:
I found few things, maybe will help to solve this issue:
- communication with blufi app stuck, no more data exchange,
- app is still connected to device, no disconnect logs,
- it is always when custom data are processed in callback, so i am wondering if i should use Queue to handle custom data requests instead

WCCWCC, 2020-07-16:
Hi, @chegewara ：

* ASSERT_PARAM(0 10), in lld_pdu.c at line 537  : This error indicates that the event tag of the received packet does not match the current event.  Received an adv packet while the connection event was running.

 * In the event callback of ble, when the data interaction is relatively fast, we do not recommend doing time-consuming operations when processing events

* Is the status of your device: that is sending adv packets and connecting packets?  Can this problem be reproduced steadily?

chegewara, 2020-07-17:
I have some additional info about blufi in release/v3.3 and probably i should open new issue about it.
I believe there is some problem with encrypting custom data. This is example of corrupted data received in blufi app, which, as i mentioned, all has been working before update esp-idf:
```
0, 4, 1, 1, 0, -55, 0, 0, 104, 116, 116, 112, 58, 47, 47, 54, 50, 46, 49, 51, 56, 46, 50, 49, 46, 56, 58, 53, 53, 53, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12, 1, 0, 0, 40, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 23, 0, 0, 0, 0, 0, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0
```
Those data are "random noise" which should be all 0s:
```
12, 1, 0, 0, 40, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 23, 0, 0, 0, 0, 0, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0
```
Also byte 5 (-55) should never be negative value. It is reproduced steadily in our application. Thanks

WCCWCC, 2020-07-20:
Hi, @chegewara 

 * Can you tell me how to reproduce the problem you said.  
 * From your description, I cannot fix this problem. You should provide more detailed information, such as log, etc.

chegewara, 2020-07-20:
@WCCWCC 
Ok, i will ask client to let me provide source code.

Thanks

chegewara, 2020-07-20:
In meantime here is part of code that may give you a lead:
```
typedef struct {
  uint8_t  server_status ;
  char      server_ip[200];
}grms_info_st;

...

  custom_data_t response_data;
  memset (&response_data, 0, sizeof(custom_data_t));

...

        response_data.flow           = READ;
        response_data.data_type      = GRMS_INFO;
        response_data.start_flag     = FIRST_PACKAGE;
        response_data.number_package = 1;
        response_data.data_lenght    = sizeof(grms_info_st);

        grms_data = (grms_info_st*) &response_data.data[0];
        memset (grms_data, 0, sizeof(grms_info_st));

        if (http_connection_status()) {
          grms_data->server_status = 1;
        } else {
          grms_data->server_status = 0;
        }
        uint8_t size_of_http_url = strlen (flashConfig.http_url_data);
        if ( size_of_http_url > 200) { size_of_http_url = 200; }

        for(uint8_t i = 0; i < size_of_http_url; i++){
          grms_data->server_ip[i] = flashConfig.http_url_data[i];
        }
        esp_blufi_send_custom_data((uint8_t*)&response_data, HEADER_SIZE + sizeof(grms_info_st));
```

Problem is when `flashConfig.http_url_data` is short (around 30 chars) and there is a lot trailing 0s. This may be normal AES encryption behavior if there is so many 0s in data to encrypt thou.

I "fixed" it in my code, so its nothing important now. 

Thanks

closed 2021-12-19 by chegewara