espressif/esp-idf #5597: EXTRA_CFLAGS not working

hassance3, 2020-07-15:
If you need add an extra argument for ESP8266, you need just to do this: 
`make app ESPPORT="MyPortNumber" EXTRA_CFLAGS="-DMyDefine=123456 -Wno-unused-variable"`

**hello_world_main.c**
```
void app_main(void){
   printf( MyDefine "\n" );
   ...
}
```

But this it's not working for ESP32 (IDF), I tried apply [this documentation](https://docs.espressif.com/projects/esp-idf/en/v3.1.5/api-guides/build-system.html) but it's seem the argument done at linking stage not compiling stage.

Alvin1Zhang, 2020-07-16:
@hassance3 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

hassance3, 2020-07-16:
First, I don't know if [IDF](https://github.com/espressif/esp-idf) support this or not, but [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) dose.

- 1: Take the hello world example.
- 2: Add `printf( MyDefine "\n" );` into `app_main(void)`.
- 3: build with this command make app EXTRA_CFLAGS="-DMyDefine=123456"
- 4: Flash & Run.

Basically we give the compiler an extra arguments, it's like `#define MyDefine "123456"`.
So, this work fine for [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) but not for [IDF](https://github.com/espressif/esp-idf).

hassance3, 2020-07-16:
For IDF this is the result:

**Build**
`idf.py build EXTRA_CFLAGS="-Dfoo=123456"`

**Log**
```
Executing action: all (aliases: build)
Running ninja in directory e:\code\sdk\esp_idf_master\examples\get-started\hello_world\build
Executing "ninja all"...
[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[2/6] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
ccache e:\code\SDK\ESP_IDF_Tools\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32/. -I../../../../components/soc/src/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/driver/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32/. -I../../../../components/soc/soc/esp32/include -I../../../../components/soc/soc/esp32/../include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/perfmon/include -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/includes -I../../../../components/esp-tls -I../../../../components/esp_adc_cal/include -I../../../../components/esp_gdbstub/include -I../../../../components/esp_hid/include -I../../../../components/tcp_transport/include -I../../../../components/esp_http_client/include -I../../../../components/esp_http_server/include -I../../../../components/esp_https_ota/include -I../../../../components/protobuf-c/protobuf-c -I../../../../components/protocomm/include/common -I../../../../components/protocomm/include/security -I../../../../components/protocomm/include/transports -I../../../../components/mdns/include -I../../../../components/esp_local_ctrl/include -I../../../../components/sdmmc/include -I../../../../components/esp_serial_slave_link/include -I../../../../components/esp_websocket_client/include -I../../../../components/expat/expat/expat/lib -I../../../../components/expat/port/include -I../../../../components/wear_levelling/include -I../../../../components/fatfs/diskio -I../../../../components/fatfs/vfs -I../../../../components/fatfs/src -I../../../../components/freemodbus/common/include -I../../../../components/idf_test/include -I../../../../components/idf_test/include/esp32 -I../../../../components/jsmn/include -I../../../../components/json/cJSON -I../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../components/libsodium/port_include -I../../../../components/mqtt/esp-mqtt/include -I../../../../components/openssl/include -I../../../../components/spiffs/include -I../../../../components/ulp/include -I../../../../components/unity/include -I../../../../components/unity/unity/src -I../../../../components/wifi_provisioning/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1905-g625bd5eb1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/ [... cut ...]
../main/hello_world_main.c: In function 'app_main':
../main/hello_world_main.c:21:9: error: 'foo' undeclared (first use in this function); did you mean 'feof'?
  printf(foo "\n");
         ^~~
         feof
		 
../main/hello_world_main.c:21:9: note: each undeclared identifier is reported only once for each function it appears in
../main/hello_world_main.c:21:12: error: expected ')' before string constant
  printf(foo "\n");
            ^~~~~
            )
			
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

hassance3, 2020-07-16:
whats strange is when you add `#define foo "123456"` to `hello_world_main.c` basically it's compile because foo is defined, but the builder (idf.py) add this argument (EXTRA_CFLAGS="-Dfoo=123456") at linking stage!.

- 1: Add `#define foo "123456"` to `hello_world_main.c`
- 2: Build `idf.py build`

**Log**
```
Executing action: all (aliases: build)
Running ninja in directory e:\code\sdk\esp_idf_master\examples\get-started\hello_world\build
Executing "ninja all"...
[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[6/6] Generating binary image from built executable
esptool.py v3.0-dev
Generated E:/Code/SDK/ESP_IDF_Master/examples/get-started/hello_world/build/hello-world.bin
Executing action: EXTRA_CFLAGS=-Dfoo=123456
ninja: error: unknown target 'EXTRA_CFLAGS=-Dfoo=123456'
command "EXTRA_CFLAGS=-Dfoo=123456" is not known to idf.py and is not a Ninja target
```

hassance3, 2020-07-16:
**I don't report this as a bug**, but probably `idf.py` have a different way to add extras arguments for compiling stage.

igrr, 2020-07-16:
Taking environment variables from arguments is supported in GNU Make. With idf.py, you need to set the variable in the environment:
```
export EXTRA_CFLAGS="-Dfoo=123456"
idf.py build
```
Or `EXTRA_CFLAGS="-Dfoo=123456" idf.py build` which is a shell syntax for setting environment variables for a single command.

hassance3, 2020-07-16:
@igrr Thank you for your replay, right now I'm on a Windows PC., it's seem not working!

```
> Set "EXTRA_CFLAGS=-Dfoo=123456"
> Set EXTRA_CFLAGS=-Dfoo=123456
> echo %EXTRA_CFLAGS%
-Dfoo=123456
> idf.py build
...
../main/hello_world_main.c:21:9: error: 'foo' undeclared (first use in this function); did you mean 'feof'?
...
```

hassance3, 2020-07-16:
I don't see the defined foo in the log.
`.....................xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32/. -I../../../../components/soc/src/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/driver/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32/. -I../../../../components/soc/soc/esp32/include -I../../../../components/soc/soc/esp32/../include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/perfmon/include -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/includes -I../../../../components/esp-tls -I../../../../components/esp_adc_cal/include -I../../../../components/esp_gdbstub/include -I../../../../components/esp_hid/include -I../../../../components/tcp_transport/include -I../../../../components/esp_http_client/include -I../../../../components/esp_http_server/include -I../../../../components/esp_https_ota/include -I../../../../components/protobuf-c/protobuf-c -I../../../../components/protocomm/include/common -I../../../../components/protocomm/include/security -I../../../../components/protocomm/include/transports -I../../../../components/mdns/include -I../../../../components/esp_local_ctrl/include -I../../../../components/sdmmc/include -I../../../../components/esp_serial_slave_link/include -I../../../../components/esp_websocket_client/include -I../../../../components/expat/expat/expat/lib -I../../../../components/expat/port/include -I../../../../components/wear_levelling/include -I../../../../components/fatfs/diskio -I../../../../components/fatfs/vfs -I../../../../components/fatfs/src -I../../../../components/freemodbus/common/include -I../../../../components/idf_test/include -I../../../../components/idf_test/include/esp32 -I../../../../components/jsmn/include -I../../../../components/json/cJSON -I../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../components/libsodium/port_include -I../../../../components/mqtt/esp-mqtt/include -I../../../../components/openssl/include -I../../../../components/spiffs/include -I../../../../components/ulp/include -I../../../../components/unity/include -I../../../../components/unity/unity/src -I../../../../components/wifi_provisioning/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1905-g625bd5eb1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\hello_world_main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj   -c ../main/hello_world_main.c`

hassance3, 2020-07-16:
maybe something else than `EXTRA_CFLAGS`? or `idf.py` don't take the environment value correctly ?

igrr, 2020-07-16:
Can you try running `idf.py reconfigure` and then `idf.py build`, after setting the environment variable?

hassance3, 2020-07-17:
@igrr  Yes! that it, work fine.

```
> Set EXTRA_CFLAGS=-Dfoo=123456
> echo %EXTRA_CFLAGS%
-Dfoo=123456
> idf.py reconfigure
> idf.py build
```

Thank you.

closed 2020-07-17 by hassance3