espressif/esp-idf #5598: Build failed when CONFIG_SPIRAM_CACHE_WORKAROUND is set

ftoyon, 2020-07-15:
We migrate a esp-idf 3.3 project with C and C++ to esp-idf 4.0 on a EPS-WROVER with external SPI RAM enable.

if the CONFIG_SPIRAM_CACHE_WORKAROUND is ennable the compilation failed whit this errors:
```
Checking Python dependencies...
Python requirements from C:\Users\Frederic\esp32\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\projets\iozebre\esp32\gateway\zesp\build
Executing "ninja all"...
[1/15] Performing build step for 'ulp_ulpInputs'
ninja: no work to do.
[2/8] cmd.exe /C "cd /D C:\Projets\IOZebre\esp32\gateway\zesp\build\esp-idf\partition_table && C:\Users\Frederic\esp3...s\cmake\3.13.4\bin\cmake.exe -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1792K,
storage,data,spiffs,0x3a0000,384K,
*******************************************************************************
[3/8] Performing build step for 'bootloader'
ninja: no work to do.
[4/6] Linking CXX executable my-app.elf
FAILED: my-app.elf
cmd.exe /C "cd . && C:\Users\Frederic\esp32\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\my-app.elf.rsp  -o my-app.elf  && cd ."
c:/users/frederic/esp32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: my-app.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/users/frederic/esp32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/users/frederic/esp32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 6593 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

projectgus, 2020-08-06:
Hi @ftoyon ,

Thanks for being patient while someone got back to you.

The error you're seeing is because the [IRAM](https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-guides/general-notes.html#memory-layout) part of the app binary no longer fits in the available IRAM memory of the chip. This is because some bugfixes or performance optimizations in v4.x are requiring more IRAM.

Exactly what is consuming IRAM will depend on which features you are using in your project. If any of your code is using the IRAM_ATTR macro then removing this will reduce IRAM use.

Also, there are some configuration items which will reduce IRAM usage:

* https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/kconfig.html#config-esp32-wifi-iram-opt
* https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/kconfig.html#config-esp32-wifi-rx-iram-opt
* https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/kconfig.html#config-lwip-iram-optimization

If any of these are enabled, disabling one of them should reduce IRAM enough to link successfully. Some of these are new optimizations since v3.3, so you shouldn't notice any reduction in performance after upgrading.

After the project can link successfully, you can also use the `idf.py size-components` target to print a simmary of which components are contributing IRAM usage to the app. If there is any code here that you can disable or not call in your app, it may also reduce the IRAM size.

projectgus, 2020-08-27:
Hi @ftoyon ,

Were you able to reduce the IRAM footprint of your app successfully?

Angus

ftoyon, 2020-08-28:
Hi @projectgus ,

The version seems to have fixed the problem.

Frederic

closed 2020-09-08 by ftoyon