espressif/esp-idf #5599: coredump to ELF (or BIN) fails in master

cvonk, 2020-07-15:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.3-dev-472-gcf056a7d0
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: [ESP Command
- Using an IDE?: Yes, vscode
- Power Supply: USB

## Problem Description

The instruction `assert(0)` triggers the coredump to flash, but fails.

### Expected Behavior

Coredump completes sucessfully

### Actual Behavior

Fails with "Core dump write binary failed with error=-3". 

For some tasks in the function `esp_core_dump_get_stack()` the variable `task_snapshot->stack_start` is 0, 1 or 0xffffffff. 

Adding the debug trace at the beginning of `esp_core_dump_get_stack()`:
```c
ESP_COREDUMP_LOGE(" start = %x, end = %x", task_snapshot->stack_start, task_snapshot->stack_end);

```
reveals:
```
E (2398) esp_core_dump_port:  start = 3ffb6640, end = 3ffb68bc
E (2404) esp_core_dump_port:  start = 0, end = 3ffb778c
E (2410) esp_core_dump_port:  start = 20000000, end = 20000070
E (2417) esp_core_dump_port:  start = 3ffb7028, end = 3ffb6e80
E (2423) esp_core_dump_port:  start = 20000070, end = 200000e0
E (2430) esp_core_dump_port:  start = 3ffb8010, end = 3ffb81dc
E (2436) esp_core_dump_port:  start = ffffffff, end = 3ffafab4
E (2443) esp_core_dump_port:  start = 3ffb6640, end = 3ffb68bc
```

### Steps to reproduce

Use the specified board, and load the code as shown below.

### Code to reproduce this issue

Start with the [blink](https://github.com/espressif/esp-idf/blob/master/examples/get-started/blink/main/blink.c) example.

Create the file `partitions.csv`:
```
nvs,      data, nvs,      0x09000,  0x004000,
otadata,  data, ota,      0x0d000,  0x002000,
phy_init, data, phy,      0x0f000,  0x001000,
factory,  app,  factory,  0x010000, 0x150000,
ota_0,    app,  ota_0,    0x160000, 0x140000,
ota_1,    app,  ota_1,    0x2A0000, 0x140000,
coredump, data, coredump, 0x3E0000, 128k
```

Create the file `sdkconfig.defaults`:
```
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y
[... 126 lines cut ...]
ELF file SHA256: c51da26943ae128e

I (2387) esp_core_dump_flash: Save core dump to flash...
I (2393) esp_core_dump_elf: Found tasks: 7
E (2398) esp_core_dump_port:  start = 3ffb6640, end = 3ffb68bc
E (2404) esp_core_dump_port:  start = 0, end = 3ffb778c
E (2410) esp_core_dump_port:  start = 20000000, end = 20000070
E (2417) esp_core_dump_port:  start = 3ffb7028, end = 3ffb6e80
E (2423) esp_core_dump_port:  start = 20000070, end = 200000e0
E (2430) esp_core_dump_port:  start = 3ffb8010, end = 3ffb81dc
E (2436) esp_core_dump_port:  start = ffffffff, end = 3ffafab4
E (2443) esp_core_dump_port:  start = 3ffb6640, end = 3ffb68bc
E (2449) esp_core_dump_elf: elf_add_segment(163): Invalid data for segment.
E (2457) esp_core_dump_elf: Task (TCB:0) processing failure = -3
E (2464) esp_core_dump_elf: elf_write_tasks_data(501): Task #1, TCB write failed, return (-3).
E (2473) esp_core_dump_elf: esp_core_dump_do_write_elf_pass(563): ELF Size writing error, returned (-3).
E (2483) esp_core_dump_common: Core dump write binary failed with error=-3
I (2491) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jun  8 2016 00:22:57
```

## Other items if possible

see `sdkconfig.defaults` and `partitions.csv` above.

cvonk, 2020-07-16:
Marginally related: `espcoredump.py` on Windows needs `tempfile.NamedTemporaryFile()` changed to `tempfile.NamedTemporaryFile(delete=False)`.  Similar to `idf_monitor.py` that has the comment
```
# On Windows, the temporary file can't be read unless it is closed.
# Set delete=False and delete the file manually later.
```

Without the change, you get an error message such as `PermissionError: [Errno 13] Permission denied: 'C:\\Users\\YOURNAME\\AppData\\Local\\Temp\\tmphvoj_t8m'`.

dobairoland, 2020-07-17:
Hi @cvonk. Thank you for your report. Did I understood you correctly and your original issue got resolved and the only remaining issue is the temporary file access issue?

cvonk, 2020-07-21:
Correct.  The environment variables `IDF_PATH`/`IDF_TOOLS_PATH` cause havoc.  Once I removed them, the core dump worked fine in the master branch and I was able to forward the core dump from flash to a server (on the next boot).  If it helps anybody in the future, the code is on [github](https://github.com/cvonk/ESP32_coredump-to-server).

Along the way, I noticed
- temporary file access issue on Windows (see earlier)
- the "dump to UART" features is short a few bytes compared to the length noted in the the first 4 bytes.  It works OK though.  

Thanks @dobairoland  et al.  The Espressif code and documentation has come a long way!

an-dr, 2020-07-22:
@cvonk thanks for the report! An updated version is coming soon

an-dr, 2020-09-23:
@cvonk sorry for the late reply. The fix available in the master branch for a while already: https://github.com/espressif/esp-idf/commit/48c4378a70d51e98fd2a56668729adf65846587e

Please take a look, if it solves the problem for you.

Alvin1Zhang, 2021-08-25:
Thanks for reporting, the fix is available, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang