espressif/esp-idf #5600: Deep sleep current issue

waqasqammar, 2020-07-16:
`const uint64_t ext_wakeup_pin_1_mask = 1ULL << GPIO_NUM_0;
 esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask, ESP_EXT1_WAKEUP_ALL_LOW);
`
when using this code my deep sleep current can't gets lower than 77uA. All other hardware e.g WIFI, bluetooth, ADC are turned OFF.
'
 esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 0);
`
using this code, the sleep current goes down to 10uA but it won't wake up on button press.

I am using stable version4.0 and followed the example but the current is too high.

chegewara, 2020-07-16:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#external-wakeup-ext1

`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);`

waqasqammar, 2020-07-18:
Hi thanks for the reply.
I have used tihs
```c
esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 0);
 set_next_wakeup(wakeup_time);
 rtc_gpio_init(SW_BUTTON);
 rtc_gpio_set_direction(SW_BUTTON, RTC_GPIO_MODE_INPUT_ONLY);
 esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
 esp_deep_sleep_start();
}
```
```c
esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 0);
 set_next_wakeup(wakeup_time);
 rtc_gpio_init(SW_BUTTON);
 rtc_gpio_set_direction(SW_BUTTON, RTC_GPIO_MODE_INPUT_ONLY);
 esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
 esp_deep_sleep_start();
}
```
```c
esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 0);
 set_next_wakeup(wakeup_time);
 esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
 esp_deep_sleep_start();
}
```

```c
esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 0);
 set_next_wakeup(wakeup_time);
 esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
 esp_deep_sleep_start();
}
```

In all the above cases the system just restarted automatically. and don't stay in deep sleep.
p.S: the physical pullup is attached.

chegewara, 2020-07-18:
Try code with different pin, because im not sure how pin0 behaves on restart.

waqasqammar, 2020-07-18:
tried with another pin also. but currently it just restarted imidiately after sleep

igrr, 2020-07-21:
Hi @waqasqammar, could you please describe the hardware you are using? If this is not one of the standard development boards, please attach schematics, if possible.

I have checked the following with IDF v4.0.1 release:

* ESP32-WROOM-32D module; TXD, RXD, GND, 3V3, EN, GPIO0 pads soldered. TXD, RXD, GND connected to a USB-UART bridge. GPIO0 pulled high to 3V3 and connected via a button to GND. 3V3 and GND connected to a power supply.
* system/deep_sleep example. In menuconfig, disabled touch and ULP wakeup, kept GPIO wakeup enabled.
* modified the GPIO wakeup in the example as follows:
  ```c++
    const int ext_wakeup_pin_1 = 0;  // < changed to 0
    const uint64_t ext_wakeup_pin_1_mask = 1ULL << ext_wakeup_pin_1;
    const int ext_wakeup_pin_2 = 0;   // < changed to 0
    const uint64_t ext_wakeup_pin_2_mask = 1ULL << ext_wakeup_pin_2;

    printf("Enabling EXT1 wakeup on pins GPIO%d, GPIO%d\n", ext_wakeup_pin_1, ext_wakeup_pin_2);
    esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask | ext_wakeup_pin_2_mask, ESP_EXT1_WAKEUP_ALL_LOW); // < changed to ESP_EXT1_WAKEUP_ALL_LOW
  ```
  (essentially, use only GPIO0 for wake-up)
* *removed* the line which isolates GPIO12, since this pin is not pulled up in ESP32-WROOM32-D:
  ```c++
  rtc_gpio_isolate(GPIO_NUM_12);
  ```
* have measured the current using an ammeter, and got the expected 5uA value.

So my conclusion is that deep sleep current is at the expected level. If you are not getting the same result, the likely cause is leakage through some of the module GPIOs (which may be connected to external signals) or through some other components on the board.

Alvin1Zhang, 2021-01-12:
@waqasqammar Thanks for reporting, would you please help share if any updates for this issue? Thanks

waqasqammar, 2021-01-12:
Hi @Alvin1Zhang  Thanks for your reply. Yes, you were right there was some leakage for one of the GPIOs. I turned that off and I got the expected results.

Alvin1Zhang, 2021-08-25:
Thanks for reporting and updates, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang