espressif/esp-idf #5601: Embed binary data when using IDF-as-lib

galah92, 2020-07-16:
Using `v4.0` & `v4.1`:

When working with `idf.py` we can [embed binary data](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data) using:
```
idf_component_register(...
                       EMBED_FILES server_root_cert.der)
```

My project use IDF-as-lib, meaning when we `include($ENV{IDF_PATH}/tools/cmake/idf.cmake)` and use `idf_build_process(esp32)` while writing regular CMake files and compiling to the ESP with `idf_build_executable(...)`.

How can I embed files to CMake targets in my project?

Thanks

renzbagaporo, 2020-07-17:
You can use `target_add_binary_data`. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data. The first argument would be your own CMake targets.

galah92, 2020-07-28:
@renzbagaporo Thanks. This doesn't seems to work, though. I'm getting the following while compiling:
```
-- Configuring done
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
FAILED: build.ninja
```
Note I switched to v4.2.

closed 2020-10-04 by renzbagaporo

AndruePeters, 2022-05-18:
Can this be added to the documentation? @galah92's fix works for me on 5.0 as well, but this is not documented.