espressif/esp-idf #5602: javascript error message: undefined and close websocket connection on "httpd_ws_send_frame_async" function

baldhead69, 2020-07-16:
https://github.com/espressif/esp-idf/blob/cf056a7d0b90261923b8207f21dc270313b67456/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L51

Need to include this line instruction "ws_pkt.final = true;" because using javascript running in a browser the function "httpd_ws_send_frame_async" presents problem(javascript error message: undefined and close websocket connection).

"httpd_ws_recv_frame" and "httpd_ws_send_frame" not show this issue without "ws_pkt.final = true;", ie it works correctly. 

I need to use "ws_pkt.final = true;" in "httpd_ws_recv_frame" and "httpd_ws_send_frame" ?

Thank's.

Alvin1Zhang, 2020-07-17:
Thanks for reporting, we will look into.

baldhead69, 2020-07-17:
Hi again,

How to use "httpd_ws_send_frame_async" without "static esp_err_t trigger_async_send(httpd_handle_t handle, httpd_req_t *req)" ?

I need to use "httpd_queue_work(handle, ws_async_send, resp_arg);" ?

In the function "esp_err_t httpd_ws_send_frame_async(httpd_handle_t hd, int fd, httpd_ws_frame_t *frame)", is not very clear to me
which one httpd_handle_t hd and which one int fd to use.

Thank's.

baldhead69, 2020-07-17:
Hi,

How can i get "@param[in] fd Socket descriptor for sending data" to use in function
"esp_err_t httpd_ws_send_frame_async(httpd_handle_t hd, int fd, httpd_ws_frame_t *frame)" ????

I want to send data from http_server running in esp32 to the client asynchronously.

The example in
"https://github.com/espressif/esp-idf/bl ... o_server.c"
doesn't seem to be asynchronous communication to me since it depends on an "HTTP_GET" from the client.

Thank's.

baldhead69, 2020-07-20:
Hi,

I am new with sockets programming.

How i have access to all "websockets" clients connected to the server ?

Is there a list of connected clients to the server that i can access ?
These list return "@param[in] fd Socket descriptor for sending data" ?

How to know which clients are still connected to the server ?
How to know those clients who disconnected from server without using "websocket close handshake" ? 

I need to update all clients and the server with the same data message when any of these change your state.

Star topology with the server at the centre of star.

Thank's.

david-cermak, 2020-08-20:
Hi @baldhead69 

You might like to populate a list of your connected clients in user code, writing a handler which convert the `httpd_req_t` to a file descriptor using `httpd_req_to_sockfd()`

baldhead69, 2020-08-20:
Hi,

If client connect with server and dont send any message initially, the uri handler are not called.
If i manage the list inside uri handler, i need to verify the list all the time that some client send some message(new connected clients and old connected clients).
And on socket disconnect the uri handler are not called too.

I registered two callback functions on esp_http_server httpd_config_t structure.
"httpd_open_func_topen_fn" and
"httpd_close_func_tclose_fn".
I take the sockets in these two function to manage my list.
The problem are two:
First problem: not only sockets of type websockets open and close here.
Http sockets and possibly other sockets type came to here.
Second problem: when the open function are called, the socket yet not switch the protocol from http to websocket, ie: websocket handshake not yet completed.

A new callback open function would be interesting only for sockets of the type websockets, and this function will be called only after the socket has become a websocket (be ready for use).
A new callback close function for only websockets closed too would be much appreciated.

Thanks.

baldhead69, 2020-08-25:
Hi,

I would like to know if ESP32 Websocket Server accept blob(binary) data format ???

Thank's.

david-cermak, 2020-08-27:
@baldhead69 I would suggest adding a new API to check a supplied socket descriptor (invalid, active, active+ws) -- Would this solve your issue?

In that case you can collect the fd's of all connecting/disconnecting clients as mentioned above (using open/close callbacks) or even simply iterating over all sockets in the system, for example:
``` C
        for (int sock =LWIP_SOCKET_OFFSET; sock < LWIP_SOCKET_OFFSET+CONFIG_LWIP_MAX_SOCKETS; ++sock) {
            int type = httpd_ws_get_fd_info(h, sock);
            if (type >= 0) {
                ESP_LOGI(TAG, "active client: httpd_handle=%p, sock=%d, is_ws:%d", h, sock, type);
            }

```

Here's the related patch adding this API
[0001-esp_http_server-Add-httpd_ws_get_fd_info-API-to-chec.patch.txt](https://github.com/espressif/esp-idf/files/5137190/0001-esp_http_server-Add-httpd_ws_get_fd_info-API-to-chec.patch.txt)

---

> I would like to know if ESP32 Websocket Server accept blob(binary) data format ???

The implementation treats both types, TXT and BIN, the same way. So supports the binary format, but just by setting/reading the packet type as `HTTPD_WS_TYPE_BINARY` vs. `HTTPD_WS_TYPE_TEXT`.

baldhead69, 2020-08-28:
Hi david-cermak,

Thank's for the reply, i will test.

> The implementation treats both types, TXT and BIN, the same way. So supports the binary format, but just by setting/reading the packet type as `HTTPD_WS_TYPE_BINARY` vs. `HTTPD_WS_TYPE_TEXT`.
Last night i checked that the binary type works correctly.

baldhead69, 2020-08-28:
Hi david-cermak,

Did you take a look at this link ?
https://github.com/espressif/esp-idf/issues/5686

Thank's.

martinKupec, 2020-09-11:
> @baldhead69 I would suggest adding a new API to check a supplied socket descriptor (invalid, active, active+ws) -- Would this solve your issue?
>
> In that case you can collect the fd's of all [...]

Hi, this API is really useful.
It would be also nice to have handler called here:
https://github.com/espressif/esp-idf/blob/6c17e3a64c02eff3a4f726ce4b7248ce11810833/components/esp_http_server/src/httpd_uri.c#L326
That way there would be event on new websocket, so the code can register for data sending.

closed 2020-09-16 by espressif-bot

Maldus512, 2020-12-29:
> @baldhead69 I would suggest adding a new API to check a supplied socket descriptor (invalid, active, active+ws) -- Would this solve your issue?
> 
> In that case you can collect the fd's of all connecting/disconnecting clients as mentioned above (using open/close callbacks) or even simply iterating over all sockets in the system, for example:
> 
>         for (int sock =LWIP_SOCKET_OFFSET; sock < LWIP_SOCKET_OFFSET+CONFIG_LWIP_MAX_SOCKETS; ++sock) {
>             int type = httpd_ws_get_fd_info(h, sock);
>             if (type >= 0) {
>                 ESP_LOGI(TAG, "active client: httpd_handle=%p, sock=%d, is_ws:%d", h, sock, type);
>             }
> 
> Here's the related patch adding this API
> 0001-esp_http_server-Add-httpd_ws_get_fd_info-API-to-chec.patch.txt

This does not work for me. Using the latest master branch I'm trying to record all websocket connections to save them into a descriptor list and keep them updated. My web server should also answer to a few HTTP requests, thus I need to distinguish between HTTP sockets and WS sockets. I added two custom close and open callbacks, they seem to find no difference between websockets and HTTP requests:

```

void close_cb(httpd_handle_t hd, int sockfd) {
    if (httpd_ws_get_fd_info(hd, sockfd)) {
        ESP_LOGI(TAG, "websocket Session closed %i", sockfd);
    }
}

esp_err_t open_cb(httpd_handle_t hd, int sockfd) {
    if (httpd_ws_get_fd_info(hd, sockfd)) {
        ESP_LOGI(TAG, "websocket Session opened %i", sockfd);
    }

    return ESP_OK;
}
```

The `httpd_ws_get_fd_info` returns 1 in all cases. Is there something I don't understand about this API?

baldhead69, 2020-12-29:
Hi @Maldus512 ,

I dont use that api for now.

I created my "own protocol", where the first payload byte are a command and the others payload bytes are data, all this on top of the websocket protocol.

I created a simple list where i store the websockets open sockets.

Inside the URI callback handler function i analyze the first byte(command), if the byte are number 1 for example, i branch to a function that add the websocket socket on the list. I dont use open socket callback from api anymore, but i use close socket callback yet.
Inside the close socket callback, i scan the list for the closed socket and delete it from the list.
Notice that this list will only contain websockets sockets because i created my own open socket function "inside" the URI callback handler function.

I believe you can use the same idea for http sockets, then you create two lists, a list for http sockets and another list for websocket sockets.

Maldus512, 2020-12-30:
Hi @baldhead69, thanks for the clarification. I understand that the websocket server is a fairly new addition to the ESP-IDF.

As for the workaround I ended up implementing a similar solution, with the connected websockets signalling their presence to subscribe for future updates.

baldhead69, 2021-08-02:
Hi @david-cermak ,

"httpd_ws_client_info_t httpd_ws_get_fd_info(httpd_handle_t hd, int fd)"
In which version of esp-idf this function was added ?
In the version i'm using , v4.2.1, this function doesn't exist yet. 

```
httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT(); 
config.httpd.open_fn = opened_socket_handler;

// typedef esp_err_t (*httpd_open_func_t)(httpd_handle_t hd, int sockfd);
esp_err_t opened_socket_handler(httpd_handle_t hd, int sockfd)
{
    httpd_ws_client_info_t info = httpd_ws_get_fd_info( hd, sockfd);

    if( info != HTTPD_WS_CLIENT_WEBSOCKET )
        return ESP_FAIL;

    // add sockfd in websocket connected clients list.
     return ESP_OK;
}
```
I want to do this pseudo code algoritm:
```
// typedef esp_err_t (*httpd_open_func_t)(httpd_handle_t hd, int sockfd);
esp_err_t opened_socket_handler(httpd_handle_t hd, int sockfd)
{
    if(  ( sockfd == websocket socket ) && ( websocket handshake finished )  )
    {
         // add sockfd in websocket connected clients list.
         return ESP_OK;
    }
    else
    {
        return ESP_FAIL;
    }
}
```
I would like to know if my approach is correct.

Thank's.

baldhead69, 2021-08-03:
@david-cermak 

When a websocket client open a connection with the websocket server the callback function "opened_socket_handler" informs "HTTPD_WS_CLIENT_HTTP".

I am expecting for "HTTPD_WS_CLIENT_WEBSOCKET".

What this means ???
That the handshake is not finished yet ???

```
esp_err_t opened_socket_handler(httpd_handle_t hd, int sockfd)
{
    printf("Open Function:\n");
    printf("hd = %p\n", hd );
    printf("sockfd = %d \n\n", sockfd );

    httpd_ws_client_info_t info = httpd_ws_get_fd_info( hd, sockfd);
 
    printf("info: %d\n\n", info );

    if( info != HTTPD_WS_CLIENT_WEBSOCKET )
        return ESP_FAIL;
}
```
```
Open Function:
hd = 0x3ffed704
sockfd = 55

info: 1      // HTTPD_WS_CLIENT_HTTP
```

baldhead69, 2021-08-04:
From what i checked, i think that the server is not closing the socket when it returns ESP_FAIL.

// typedef esp_err_t (*httpd_open_func_t)(httpd_handle_t hd, int sockfd);
// - ESP_OK   : On success
// - Any value other than ESP_OK will signal the server to close the socket immediately

baldhead69, 2021-08-16:
????

david-cermak, 2021-08-16:
@baldhead69 

> When a websocket client open a connection with the websocket server the callback function "opened_socket_handler" informs "HTTPD_WS_CLIENT_HTTP".

Yes, this means that the client has just opened an http connection but the ws handshake is not finished yet

> From what i checked, i think that the server is not closing the socket when it returns ESP_FAIL.

Yes, correct. Thanks for pointing this out, this is probably a bug which could lead to socket exhaustion issue.

baldhead69, 2021-08-16:
> @baldhead69
> 
> > When a websocket client open a connection with the websocket server the callback function "opened_socket_handler" informs "HTTPD_WS_CLIENT_HTTP".
> 
> Yes, this means that the client has just opened an http connection but the ws handshake is not finished yet
How do i know if the handshake is finished  ?
I need to open a "feature request" or is there any other possible solution ???

Basically i want to do what this pseudo code algoritm do:
```
// typedef esp_err_t (*httpd_open_func_t)(httpd_handle_t hd, int sockfd);
esp_err_t opened_socket_handler(httpd_handle_t hd, int sockfd)
{
    if(  ( sockfd == "websocket socket" ) && ( "websocket socket" handshake finished )  )
    {
         Add sockfd in websocket connected clients list.
         return ESP_OK;
    }
    else
    {
        return ESP_FAIL;
    }
}
```

david-cermak, 2021-08-17:
> Basically i want to do what this pseudo code algoritm do:

When calling the `open_fd()`, the ws handshake is yet to be complete, so you cannot tell if it's ws or http protocol at this point.

>  is there any other possible solution ???

The easiest solution is to keep track of *all* the connected clients and check the connection type only when you have to communicate with the specific client. Assuming, you wanted to check if a WS client is alive, so the code would be approximately as follows:
```cpp
                for (int i=0; i<keep_alive_storage->max_clients; ++i) {
                    if (httpd_ws_get_fd_info(h, keep_alive_storage->clients[i].fd == HTTPD_WS_CLIENT_WEBSOCKET &&  keep_alive_storage->clients[i].type == CLIENT_ACTIVE) {
                        if (keep_alive_storage->clients[i].last_seen + keep_alive_storage->keep_alive_period_ms <= _tick_get_ms()) {
                            ESP_LOGD(TAG, "Haven't seen the client (fd=%d) for a while", keep_alive_storage->clients[i].fd);
                            if (keep_alive_storage->clients[i].last_seen + keep_alive_storage->not_alive_after_ms <= _tick_get_ms()) {
                                ESP_LOGE(TAG, "Client (fd=%d) not alive!",  keep_alive_storage->clients[i].fd);
                                keep_alive_storage->client_not_alive_cb(keep_alive_storage, keep_alive_storage->clients[i].fd);
                            } else {
                                keep_alive_storage->check_client_alive_cb(keep_alive_storage, keep_alive_storage->clients[i].fd);
                            }
                        }
                    }
                }
```

baldhead69, 2021-08-17:
Hi @david-cermak ,

I really think that is more interesting the "server" only call "open_fd( )" after handshake is finished for websocket socket.

You can use ".is_websocket = true" to identify if the client is a websocket or not.
You can use the beginning of uri to identify the protocol too( ws or wss ). 
```
static const httpd_uri_t get_clients_message =
{
    .uri       = "/",  
    .method    = HTTP_GET,
    .handler   = get_clients_message_handler,
    .user_ctx  = NULL,
    .is_websocket = true
};
```
I implemented something similar when the "server" call the uri handler( get_clients_message_handler ), because the handshake is already finished.
So i don't see why the "server" wouldn't be able to use this same solution to call "open_fd( )" instead of "get_clients_message_handler". In fact the server will call these two functions when the handshake is finished.  

I can open a feature request for this.

Thank's.

david-cermak, 2021-08-17:
Yes, it's possible to add a handler/callback once the ws handshake completes, but I think it's just a very specific use case that won't fit well to the generic http server, while there's a very simple way to check from the user space (as described https://github.com/espressif/esp-idf/issues/5602#issuecomment-900029946)

> I can open a feature request for this.

Please feel free to create one. We can reevaluate this and also see if other users find it useful.

baldhead69, 2021-08-17:
> Yes, it's possible to add a handler/callback once the ws handshake completes, but I think it's just a very specific use case that won't fit well to the generic http server, while there's a very simple way to check from the user space (as described [#5602 (comment)](https://github.com/espressif/esp-idf/issues/5602#issuecomment-900029946))

I think it might be the way you did it( httpd_ws_get_fd_info( ) ), but the callback function "open_fd( )" must be only called after handshake finished.
```
esp_err_t opened_socket_handler(httpd_handle_t hd, int sockfd)
{
    printf("Open Function:\n");
    printf("hd = %p\n", hd );
    printf("sockfd = %d \n\n", sockfd );

    httpd_ws_client_info_t info = httpd_ws_get_fd_info( hd, sockfd);
 
    printf("info: %d\n\n", info );

    if( info != HTTPD_WS_CLIENT_WEBSOCKET )
        return ESP_FAIL;

  // Add to websocket clients list.

    return ESP_OK;
}
```