espressif/esp-idf #5603: CAN (TWAI) driver broken by 97721d4

atanisoft, 2020-07-16:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): master (latest)
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): gcc 8.2.0r2
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:        USB

## Problem Description

Commit 97721d4 renamed CAN -> TWAI and introduced a very incomplete CAN API translation which will not compile:
```
In file included from ../components/OpenMRNLite/src/freertos_drivers/esp32/Esp32HardwareCanAdapter.hxx:42,
                 from ../main/esp32s2io.cpp:55:
.../components/driver/include/driver/can.h:21:2: warning: #warning driver/can.h is deprecated, please use driver/twai.h instead [-Wcpp]
 #warning driver/can.h is deprecated, please use driver/twai.h instead
  ^~~~~~~
.../components/driver/include/driver/can.h:23:10: fatal error: hal/can_types.h: No such file or directory
 #include "hal/can_types.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
After renaming `can_types.h` to `twai_types.h` compilation got further but failed again:
```
../components/OpenMRNLite/src/freertos_drivers/esp32/Esp32HardwareCanAdapter.hxx:77:9: note: suggested alternative: 'twai_timing_config_t'
         can_timing_config_t can_timing_config = CAN_TIMING_CONFIG_125KBITS();
         ^~~~~~~~~~~~~~~~~~~
         twai_timing_config_t
../components/OpenMRNLite/src/freertos_drivers/esp32/Esp32HardwareCanAdapter.hxx:79:9: error: 'can_filter_config_t' was not declared in this scope
         can_filter_config_t can_filter_config = CAN_FILTER_CONFIG_ACCEPT_ALL();
         ^~~~~~~~~~~~~~~~~~~
../components/OpenMRNLite/src/freertos_drivers/esp32/Esp32HardwareCanAdapter.hxx:82:60: error: 'CAN_MODE_NORMAL' was not declared in this scope
         can_general_config_t can_general_config = {.mode = CAN_MODE_NORMAL,
                                                            ^~~~~~~~~~~~~~~
```
Most of the older CAN types are not correctly defined to their replacements.

### Steps to reproduce

Attempt to compile CAN example (any) from IDF v4.0 or prior.

Dazza0, 2020-07-30:
@atanisoft I can't recreate this issue. Are you weren't building for `esp32s2` instead of `esp32`? (i.e., did you call `idf.py set-target esp32` before building?

atanisoft, 2020-07-30:
I've tested it with both esp32 and esp32s2. There are missing defines for compatibility between the old API and new API as well as missing includes for the new data type definitions.

Take the can examples from IDF v4 release branch as-is and compile them against master and they will fail.

Dazza0, 2020-07-30:
For esp32:
Tested again esp-idf latest (cf056a7d0b90261923b8207f21dc270313b67456) compiling against CAN Self Test example as-is from release/v4.0 (7abb8e7fc34085360cbae87c8e2a19c339228b12). Compiled with no issues, log output below:
```
user@user-pc:~/Desktop/can_self_test$ idf.py set-target esp32
user@user-pc:~/Desktop/can_self_test$ idf.py app
Executing action: app
Running ninja in directory /home/user/Desktop/can_self_test/build
Executing "ninja app"...
[917/925] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/can_self_test_example_main.c.obj
In file included from ../main/can_self_test_example_main.c:28:
/home/user/esp/esp-idf/components/driver/include/driver/can.h:21:2: warning: #warning driver/can.h is deprecated, please use driver/twai.h instead [-Wcpp]
 #warning driver/can.h is deprecated, please use driver/twai.h instead
  ^~~~~~~
In file included from /home/user/esp/esp-idf/components/driver/include/driver/can.h:23,
                 from ../main/can_self_test_example_main.c:28:
/home/user/esp/esp-idf/components/soc/src/esp32/include/hal/can_types.h:21:2: warning: #warning hal/can_types.h is deprecated, please use hal/twai_types.h instead [-Wcpp]
 #warning hal/can_types.h is deprecated, please use hal/twai_types.h instead
  ^~~~~~~
[925/925] Generating binary image from built executable
esptool.py v3.0-dev
Generated /home/user/Desktop/can_self_test/build/can_self_test_example.bin

App build complete. To flash, run this command:
/home/user/.espressif/python_env/idf4.3_py3.6_env/bin/python ../../esp/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash 0x10000 build/can_self_test_example.bin
or run 'idf.py -p (PORT) app-flash'
```

For esp32s2:
CAN API is not supported when building for esp32s2 because the esp32s2 contains a TWAI peripheral.

atanisoft, 2020-07-30:
I agree that CAN is not expected to work on the esp32s2 (even TWAI is problematic on master). But with can.h being nothing more than a series of `#define` macros to redirect to the TWAI APIs it should in theory allow easy portability of existing code that works on IDF v4.1 (and prior) to IDF v4.2 (master) and extend the support to the esp32s2.

Very likely the reference to `hal/can_types.h` should be replaced with `hal/twai_types.h` and that will solve most of it, but there are a number of other parts of the older API that are not mapped correctly (such as `CAN_TIMING_CONFIG_125KBITS`)

closed 2020-09-22 by atanisoft