espressif/esp-idf #5605: V4.1-Beta2: Blufi example reports "blufi_dh_negotiate_data_handler make public failed -12416"

LeoYan, 2020-07-17:
V4.1-Beta2 also has the issue #5411 

)## Environment
- Development Kit:  ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): V4.1-Beta2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:     USB

## Problem Description
   Blufi example reports "blufi_dh_negotiate_data_handler make public failed -12416(MBEDTLS_ERR_DHM_BAD_INPUT_DATA)" when click "encryption" 加密 button of Android EspBlueFi APP.   but it works if use IDF-V3.3.2.

### Actual Behavior
I (4237) BLUFI_EXAMPLE: BLUFI BLE is not connected yet

I (154067) BLUFI_EXAMPLE: BLUFI ble connect

E (157627) BLUFI_EXAMPLE: blufi_dh_negotiate_data_handler make public failed -12416

E (157627) BLUFI_EXAMPLE: BLUFI report error, error code 8

//Detailed problem description goes here.

- [*] sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4935446/sdkconfig.zip)

WCCWCC, 2020-07-17:
Hi, @LeoYan 

Turning off this configuration item(Component config -> mbedTLS -> Enable hardware MPI(bignum) acceleration), there will be no error.

LeoYan, 2020-07-23:
thanks
 yes, it works.  but is it just get around this issue?

Alvin1Zhang, 2020-08-06:
Thanks for reporting, and sorry for the late reply. We had a fix under internal reviewing. Will update once the issue is available on GitHub. Thanks.

Alvin1Zhang, 2021-08-25:
Thanks for reporting, and sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/c519d78e00ab437b8630e81bb132098c84185afc, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang