espressif/esp-idf #5606: CMake Problem with Backslash on Windows

zinke-ct-video, 2020-07-17:
## Environment

- IDF version: v4.3-dev-472-gcf056a7d0
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No

## Problem Description

I have a fresh new installation of the ESP IDF Toolchain. The ESP-IDF is on master-branch. I have run an idf.py fullclean.
When I want to recompile my project using idf.py build, I get the following output:

> idf.py build
> Executing action: all (aliases: build)
> Running cmake in directory c:\users\user258\esp\projects\esp32-mesh\universal-webinterface\build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\users\user258\esp\projects\esp32-mesh\universal-webinterface"...
> Warn about uninitialized values.
> -- Found Git: C:/Program Files/Git/bin/git.exe (found version "2.21.0.windows.1")
> -- ccache will be used for faster recompilation
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: C:/Users/user258/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: C:/Users/user258/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: C:/Users/user258/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: C:/Users/user258/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
> -- Check for working CXX compiler: C:/Users/user258/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Building ESP-IDF components for target esp32
> CMake Error at C:/Users/user258/esp/esp-idf/tools/cmake/component.cmake:224 (message):
>   CMake Error at build_properties.temp.cmake:2 (set):
> 
>     Syntax error in cmake code at
>       C:/Users/user258/esp/projects/ESP32-Mesh/universal-webinterface/build/build_properties.temp.cmake:2
>     when parsing string
>       C:\Users\user258\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe
>     Invalid character escape '\U'.
> 
>   Call Stack (most recent call first):
> 
>     C:/Users/user258/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)
> 
> Call Stack (most recent call first):
>   C:/Users/user258/esp/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
>   C:/Users/user258/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
>   CMakeLists.txt:7 (project)
> 
> -- Configuring incomplete, errors occurred!
> See also "C:/Users/user258/esp/projects/ESP32-Mesh/universal-webinterface/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> 

### Expected Behavior

I assume the paths in the .cmake files should use a slash instead of a backslash.

### Actual Behavior

The .cmake file contains a backslash in the path for the python.exe file, which is incorrecty interpreted as a special char.

Any ideas how to fix?

dobairoland, 2020-07-17:
Hi @zinke-ct-video. This seems to be the same issue as https://github.com/espressif/esp-idf/issues/5587 and https://github.com/espressif/esp-idf/issues/5512. The fix will be published soon. Until then you may try this [workaround](https://github.com/espressif/esp-idf/issues/5587#issuecomment-658306645).

zinke-ct-video, 2020-07-17:
Hi dobairoland, thank you for the workaround, and sorry for opening another issue. I overlooked #5587 and #5512.

igrr, 2020-07-17:
The fix has been merged as commit f5b1f43a. It is not on Github yet, but should be synced in the next few days.

Alvin1Zhang, 2020-08-05:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/f5b1f43a2b0d88493224ed3587bf67f8c52eaa5b, feel free to reopen if the issue still happens. Thanks.

closed 2020-08-05 by Alvin1Zhang