espressif/esp-idf #5608: [v3.3.2] Removal of `ip_addr_t dns_getserver()` in lwip breaks compatibility with previous 3.3.x versions

codefrey, 2020-07-17:
## Environment

- IDF version: v3.3.2
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Using an IDE: Visual Studio Code

## Problem Description

With commit `da2740fa8d56b9b9e8a10602f38df1ea4dbd9b74` of the lwip component, the definition and declaration
of the function `ip_addr_t dns_getserver()` was removed. The function in `dns.h` was available when the flag `ESP_DNS` was set.

This change causes two problems:
1) Compatibility to v3.3.1 is broken
2) Since `ESP_DNS` is still set to `1` in `lwipopts.h` there is no way to write software which is agnostic to minor releases. There is no standard way to determine whether `ip_addr_t* dns_getserver()` or `ip_addr_t dns_getserver()` are defined.

### Expected Behavior

When `ESP_DNS` is set, `ip_addr_t dns_getserver()` is defined in `dns.h`. When `ESP_DNS` is not set, `ip_addr_t* dns_getserver()` is defined in `dns.h`.

### Actual Behavior

In both cases, `ip_addr_t* dns_getserver()` is defined in `dns.h`.

### Steps to reproduce

`git diff 057f7f6a8ca3cd736009cc80d4f429f42b0db254 da2740fa8d56b9b9e8a10602f38df1ea4dbd9b74` in `esp-lwip` shows the offending commit.

david-cermak, 2020-08-25:
Hi @mkfrey 

I'm sorry, indeed this is a breaking change. 
Might I suggest a workaround to call `tcpip_adapter_get_dns_info()`? Using a `tcpip_adapter` API is a general approach to use lwIP functionality from esp-idf. we have to be more careful about such changes between releases.

david-cermak, 2021-01-07:
@mkfrey We are really sorry for this inconvenience. Unfortunately, there's no way to rectify this now (just to learn from the problem so it won't happen in the future) so we are closing this issue.

We do recommend using Espressif's abstraction (tcpip-adapter, esp-netif) layer when working with lwip, but all APIs in the public folders must maintain compatibility, especially in bugfix releases. We're currently evaluation options to test braking API changes in the CI. Thanks for raising this issue to point out this important problem.

closed 2021-01-07 by Alvin1Zhang