espressif/esp-idf #5610: ESP32 NVS problem

benas1989, 2020-07-17:
## Environment

- Development Kit: ESP32-Wrover
- Kit version DevKitC) v4
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:       idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    //xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description
I'm trying to save data to nvs partition. I saved one integer and one custom type. But when i try to read values i get error 0x110C.

### Expected Behavior
Always saves and reads data

### Actual Behavior
Does not always read the data. Sometimes an error is received

### Steps to reproduce

1. Clone repo "https://github.com/benas1989/esp32-nvs.git"
2. Set struct settings data size to 30 
3. Upload to esp32 
4. reboot esp.  In the console you will see reboot times and settings values

1. Clone repo "https://github.com/benas1989/esp32-nvs.git"
2. Set struct settings data size to 40
3. Upload to esp32 
4. reboot esp.  In the console you will see reboot times and error code 4364(0x110C)

1. Clone repo "https://github.com/benas1989/esp32-nvs.git"
2. Set struct settings data size to 50
3. Upload to esp32 
4. reboot esp.  In the console you will see reboot times and error code 4364(0x110C)

1. Clone repo "https://github.com/benas1989/esp32-nvs.git"
2. Set struct settings data size to 100 
3. Upload to esp32 
4. reboot esp.  In the console you will see reboot times and settings values

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
My code: https://github.com/benas1989/esp32-nvs/blob/master/main/main.c

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2020-07-18:
> When calling nvs_get_str and nvs_get_blob with non-zero out_value, length has to be non-zero and has to point to the length available in out_value.

benas1989, 2020-07-18:
Thank you. Length means how much space can be allocated. If I don't initialize it is any number :)

0xjakob, 2020-07-20:
@benas1989 error 0x110C means `ESP_ERR_NVS_INVALID_LENGTH` and is returned

> if length is not sufficient to store data.

In your code, this `length` parameter is uninitialized. How do you do that:

> Set struct settings data size to 30

? Can you give an example?

0xjakob, 2020-07-20:
> Thank you. Length means how much space can be allocated. If I don't initialize it is any number :)

Yes, exactly. If you `out_value` is initialized and you don't initialize the value pointed to by `length`, then the behavior is undefined (and we can't reproduce this because on our devices it would behave differently). So please show us what you mean with setting the length of the struct to 30, 40, etc.

benas1989, 2020-07-20:
now i know :)
I save settings struct in nvs. inside i have date array. But when i change size of date i actualy change where length variable  will be allocated. Becouse i not initialize lenght i got random number. Sometime 0 sometimes very big number.

typedef struct settings_t
{
  int version;
  char date[30];
  int speed;

} settings;

0xjakob, 2020-07-20:
@benas1989 Very good. Sounds like the issue is solved, is that correct?

benas1989, 2020-07-20:
Yes

0xjakob, 2020-07-20:
Excellent! Thanks for reporting back.

closed 2020-07-20 by 0xjakob

hancunchou, 2021-11-09:
define the struct variable on global