espressif/esp-idf #5611: Installing/configuring  IDF v3.2.2 or v3.2.3  with IDF Installer 2.3

mrutyunjay-patel, 2020-07-18:
## Environment
- IDF version : v3.2.3  
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt

## Problem Description
configuring IDF tools with IDF v3.2.2/v3.2.3  with **esp-idf-tools-setup-2.3.exe** fail with error !
as IDF3.2.2/3.2.3  do not have 
**idf_tools.py **
**tools.json **

and the installation fails !

### Expected Behavior
IDF tools should be able to install/configure or idf version 3.2.2 and 3.2.3 .

### Actual Behavior
IDF tools should be able to install/configure with older idf version 3.2.2 and 3.2.3 .

### Steps to reproduce
1. install idf repo to a folder with git command 
`git clone -bv3.2.3 --recursive https://github.com/espressif/esp-idf esp-idf-323 '

2. run  **esp-idf-tools-setup-2.3.exe** to use existing IDF folder at **esp-idf-323** . 

**Workaround to this issue :**
Manually copy the 2 missing files to idf  3.2.2/3.2.3 from 3.3.x to your idf folder
Copy idf_tools.py and tools.json from 
https://github.com/espressif/esp-idf/blob/release/v3.3/tools/idf_tools.py
https://github.com/espressif/esp-idf/blob/release/v3.3/tools/tools.json
\esp-idf-323\tools\idf_tools.py
\esp-idf-323\tools\tools.json 

Thanks and Regards
Mrutyunjay

igrr, 2020-07-18:
@mrutyunjay-patel, please make sure you are referring to the correct version of documentation. For 3.2 releases, please use https://docs.espressif.com/projects/esp-idf/en/v3.2.4/get-started/windows-setup.html. Note that the installation procedure for 3.x releases is different from 4.x. In particular, IDF Tools installer version 2.3 is not being used for IDF 3.x releases.

mrutyunjay-patel, 2020-07-18:
hi igrr,
ok 3.2.4 is released , but still the same issue with 2.3 installer. 

I came across this issue while testing idf vscode plugin for new development workstation.
as we wanted to use the vscode  for idf development .

we are using idf msys32 env for production as our product is based on IDF 3.2.2 , also the idf version 3. 2 is the LTS version,
we do not want to change over to 3.3.x or 4.x for existing projects .   

My workaround of copying idf-tool.py and tools.json allows to use idf version 3.2.x with 
newer idf tool in **windows environment** without depending on **msys32 environment**  and could use pure windows based environment, the newer **VSCODE IDE for development** .

The exact issue after trying 3.2.4 with idf tool installer 2.3 .
missing a space between parameters in idf tool installer 2.3 routines. 
**"--idf-path D:\esp-idf-324--tools "**
should have been
 **"--idf-path D:\esp-idf-324 --tools"**
`------------------------- Setup Log  ------------------------------------------------------
2020-07-18 16:52:31.512   PYTHONPATH=
2020-07-18 16:52:31.512   idf_tools.py does not exist in IDF directory, using a fallback version
2020-07-18 16:52:31.512   idf_tools.py command: C:\Python37-32\python.exe "C:\Users\mrutyunjay\.espressif\idf_tools_fallback.py" --idf-path D:\esp-idf-324--tools "C:\Users\mrutyunjay\.espressif\tools_fallback.json"
2020-07-18 16:52:31.512   Installing tools:C:\Python37-32\python.exe "C:\Users\mrutyunjay\.espressif\idf_tools_fallback.py" --idf-path D:\esp-idf-324--tools "C:\Users\mrutyunjay\.espressif\tools_fallback.json" install
2020-07-18 16:52:31.916   Done, exit code=2
2020-07-18 16:52:31.916   --------
2020-07-18 16:52:31.916   Running command: C:\Python37-32\python.exe **"C:\Users\mrutyunjay\.espressif\idf_tools_fallback.py" --idf-path D:\esp-idf-324--tools "C:\Users\mrutyunjay\.espressif\tools_fallback.json" install**
                          usage: idf_tools_fallback.py [-h] [--quiet] [--non-interactive]
                                                       [--tools-json TOOLS_JSON] [--idf-path IDF_PATH]
                                                       {list,check,export,install,download,install-python-env}
                                                       ...
                          idf_tools_fallback.py: error: argument action: invalid choice: 'C:\\Users\\mrutyunjay\\.espressif\\tools_fallback.json' (choose from 'list', 'check', 'export', 'install', 'download', 'install-python-env')
                          
2020-07-18 16:52:31.916   --------
2020-07-18 16:52:31.916   Message box (OK):
                          Installation has failed with exit code 2
2020-07-18 16:52:38.572   User chose OK.
2020-07-18 16:52:38.578   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-07-18 16:52:42.439   User chose Yes.
2020-07-18 16:52:42.461   CurStepChanged raised an exception.
-------------------------------------------------------------------------------`

igrr, 2020-07-18:
Hi @mrutyunjay-patel, as explained in my previous message, 3.x releases are not compatible with IDF Tools installer 2.3.

Unfortunately the only way to use Windows environment without Msys is with IDF 4.x and later.

IDF Vscode plugin works with CMake based build system, which is supported in IDF v4.0 and above.

closed 2020-07-24 by igrr