espressif/esp-idf #5613: Peek CAN/TWAI receive message queue or enable MailBox

keelung-yang, 2020-07-18:
It's helpfull to simplify designing middleware/app which take care paired nodes, in other words request/response communication, such as in ISO-14229 UDS(Unified Diagnostic Services).

In a diagnostic session, we only take care of 3 node addresses: request/response/functional, such 7C6/7E6/7DF. So if a received message address is not mine, not 7C6/7E6/7DF, I shouldn't receive it. But without peeking, I don't know which address is in receiving queue. But even if I know, then what about others? receive and drop them? If receive or drop, then it's hard to design middleware/app which support multi-session.

So the mailbox solution is better then only peeking:  there is a mailbox for each node address.

Actually, the mailbox solution is used in many productions from many manufactures, such as ST, NXP, Renesas, etc.

atanisoft, 2020-07-19:
you could possibly implement this using filters and routing to multiple RX queues based on the node id in your RX handler.

keelung-yang, 2020-07-19:
@atanisoft Yes, I can implement it by myself, which means in app level. But I think it should be implemented in driver level.

atanisoft, 2020-07-19:
As long as this is an opt-in feature and not forced on all consumers it can be implemented in the driver itself. Otherwise it would make more sense to use filters to drop the frames which do not have your expected addresses.

keelung-yang, 2020-07-20:
@atanisoft OK then how many filters can be configurated in one app?

atanisoft, 2020-07-20:
I haven't used filters in my usages. But from a quick read of the API it looks like you can configure at least one filter and a filter mask to accept multiple IDs.

collin80, 2020-07-21:
I agree, multiple mailboxes would be handy. And, yes, many MCUs have CAN hardware with multiple mailboxes. However, the CAN hardware in the ESP32 is well... not that great. Hardware based filtering is total garbage with this hardware. So, the realistic solution would be for the driver to emulate having a nice mailbox system. I don't believe the stock driver you get with ESP-IDF emulates mailboxes and such. As luck might have it, I don't use the built-in driver and wrote my own back before they even started including their official driver: https://github.com/collin80/esp32_can

Mine emulates 32 mailboxes, each with a filter/mask pair and each can issue a callback when that "mailbox" gets a message. So, it's a lot higher level than the built-in driver, lots more features. It was written to be used with the Arduino-ESP32 project to allow use of the ESP32 with Arduino. You could probably make it work directly with ESP-IDF too. If you can use it, great. If it helps you to steal some of it and write something of your own, great. If if gives someone some ideas and they add stuff to the official driver, great. If no one wants it, luckily you didn't pay a dime for it. ;)

keelung-yang, 2024-07-30:
> I agree, multiple mailboxes would be handy. And, yes, many MCUs have CAN hardware with multiple mailboxes. However, the CAN hardware in the ESP32 is well... not that great. Hardware based filtering [...]

General features should be implemented officially but individually.

keelung-yang, 2024-07-30:
> But IIUC, you are asking for independent filter for each ID? AFAIK, other solutions like ST do support independent mailbox with filters, but still don't have so many set of mailboxs.

No. No need for all IDs but expected IDs. 
This bug fired about 4 years ago, so I need time to recall the memory of mine but the board.
At the beginning, I just want to a feature for making a diagnostic "session" as described in ISO14229/ISO15765.
And then I'm thinking the potential implementations: Peek(check but not fetch) or session? And then who and where it should be implemented? On the other hand, it should be implemented by users or in official driver/library ?