espressif/esp-idf #5614: FATFS partition random empty page in the middle of file

ryan4volts, 2020-07-19:
Unfortunately I don't have much information on this. Perhaps this issue has been fixed one of the current versions, or there was a bug in the way I am writing data?

I have had several esp32-pico-d4 units deployed in field that used the following partition table:
# Name, 	Type, 	SubType,	Offset, 	  Size, Flags
nvs,		data,	nvs,		0x9000,	  24K,
phy_init,	data,	phy,		0xf000, 	  4K,
factory,	app,		factory,	0x10000,	  1216K,
fact_data,	data,	fat,	 	0x140000,  1M,

App version:      4fac774-dirty
ESP-IDF:          v4.0-dirty

After several weeks in field of writing data directly into the "fact_data" partition some of the units stopped functioning correctly, as they ran into a blank section within its memory that was unhandled. Within this FATFS section I have 2 files, the first one is a configuration file that would be written once every few months. The configuration file I have only written 32bytes into this section. The general data.log file contains a rolling memory method that writes continuously and appends to its last point of data. Once the end of the file is reached, it writes again to the start of the file overwriting the old data. If the old data was offloaded to a server or not, it is still overridden. I needed to do this instead of using the in-build wearlevelling because I needed maximum space for the data I was writing, that is 1MB or raw space. The remainder of the partition table above has since been expanded to allow for OTA functionality with the 1MB file size, so I never wanted to expand on that.

Nevertheless, reading this partition back and looking at a hex editor I have ACTUAL data start at 0x146000 and ending 0x16BFFF. A blank section exists between 16C000 to 16CFFF then data starts again from 16D000 to 1F0FFF.

Looking at the data that exists of both sides of the boundary of this blank section, its as if it tried to write into that section of memory and kept incrementing through it, despite not storing any actual data. When the data was finally read, my code was unable to handle this condition and got stuck.

My write function is below. I found that when I wrote data that went past the end of the file/partition bounds, it would still return that the bytes were written, so I had to triple check my file position to see if it was at the end of the file or not. I tried to make this function more generic instead of knowing the bounds of the partition or file that is was writing into. Everytime I write into this file, the FILE * pointer is reopened and closed after.

```
int Hal_Storage_FatFs::WriteFile(FILE * file, uint8_t * buf, int32_t len, int32_t position) {
	//write all bytes possible into file from a SPECIFIC address, return the ACTUAL amount of bytes written
	//FIRST seek len past file to ensure it can actually be written.
	int resp = fseek (file, position + len, SEEK_SET);
	long posAfterWrite = ftell(file);
	printf("ftell pos before write. tell %ld = %d + %d (start + len)\n", posAfterWrite, position, len);
	if (posAfterWrite != position + len) {
		//end of file
		puts("Hal_Storage_FatFs::WriteFile: found end of file");
		return 0;
	}

	//try to seek again, if we can't seek to correct location, memory was full
	if (fseek(file, -1*len, SEEK_CUR) != 0 || ftell(file) != position) {
		puts("Couldn't seek to correct location");
		return 0;
	}

	long current_pos = ftell(file);
	printf("WriteFile fseek resp: %d, tell %ld\n", resp, current_pos);

	int bytes_written = fwrite(buf, 1, len, file);

	if (this->debugLvl >= DEBUG_LVL_VERBOSE) {
		printf("Wrote %d bytes\n", bytes_written);
	}

	fflush(file);
	long posNow = ftell(file);
	if (posAfterWrite != posNow) {
		printf("WriteFile after write: estimated %lu != tell %ld\n", posAfterWrite, posNow);
		return 0;
	}
	return bytes_written;
}
```

I hope the above information is all you need to provide me with some direction on how this may have happened?

Cheers,
Ryan.

igrr, 2020-07-19:
Hi @ryan4volts, you mentioned having "to do this instead of using the in-build wearlevelling". Could you explain how you use FATFS without wear levelling? Have you written a custom diskio driver?

To clarify, is the behavior that there is a blank sector inserted into the steam of data, or is one sector of data skipped, with a blank sector in its place?

If it's the latter, I would suggest enabling menuconfig option to verify flash content after writing.

Is bytes_written value checked anywhere? If it is -1, will the code continue writing, or will it check errno?

ryan4volts, 2020-07-21:
Hi @igrr, the above code is the lowest level I go in writing to the filesystem. To say i wrote a library that replaces wear levelling is completly wrong. I would simply say that I chose to use a FATFS, created a file a file, and treated it as a circular buffer. In this way I had maximum space to hold a data log. I could store up to a month of data, and if it would over ride the old data.

Looking at the data stored, it doesn't appear to have skipped the page entirely being half a packet before and the other after afterwards. I can see that the function attached in the initial post has written to this section as if it worked correctly, all the way through. But nothing was ACTUALLY written.

I'll check out the menuconfig option as a possible solution.

Looking at the fwrite function, it seems to only be able to return 0 or greater. if it != len (in my case) the errno is populated. So I don't think -1 is possible? unless it was written into the esp library? Nonetheless the functions that handle this code handle any value that is 0 or greater. That is if only 5 bytes were written and 10 were intended, it attempts one more times to write the final 5 bytes. Else just generates a logging error. Which of course if I wasn't connected to it at the time, wouldn't show anything :(

Mliron, 2021-07-16:
Hi @ryan4volts, everything you have provided checks out (write function and data addresses seem fine) as you did not bypass the built-in wearlevleling system unless you wrote your own disk-io driver, which you stated you did not.

With the partition table and write function you have provided, we weren't able to reproduce the problem. However it is still possible that the bug is caused by our code, so would you kindly provide us with more information?
Code related to filesystem and data storage, for example how do you initialize the FAT filesystem, how do you read the data, etc.

Mliron, 2021-07-31:
As we weren't able to reproduce the problem we are closing the issue. Feel free to create a new issue thread or reopen this one if you encounter this problem again.

closed 2021-07-31 by Mliron